Quedale/OnvifDeviceManager #1: compilation

boc99, 2023-03-17:
meson.build:1:0: ERROR: Meson version is 0.61.2 but project requires >= 0.62.0
the version is 0.61 the 0.62 does not exist

Quedale, 2023-03-17:
What distro and version are you using?
There might be an alternative source to get meson like PIP instead of apt.

Let me know and I could look it up.

boc99, 2023-03-18:
Linux mint 21.1
pip 23.0.1
meson 0.61.2-1

boc99, 2023-03-18:
I managed to get it to work.
After the scan find the camera
No information is reported.
I tried with two different cameras

Quedale, 2023-03-18:
What are the camera models?

This tool should be compatible with any ONVIF camera that implements [WS-Discovery](https://www.google.com/url?sa=t&rct=j&q=&esrc=s&source=web&cd=&cad=rja&uact=8&ved=2ahUKEwj8xuGSp-b9AhUrlIkEHYjsC3IQFnoECBEQAQ&url=https%3A%2F%2Fwww.onvif.org%2Fwp-content%2Fuploads%2F2016%2F12%2FONVIF_Feature_Discovery_Specification_16.07.pdf&usg=AOvVaw1_SxX2uKThjbQgMHtyBYkM).

boc99, 2023-03-19:
Honwywell H4W4PER2
hIKVISION DS-2CD22512F

They are visible on Onvif Manager windows which, unlike OnvifDeviceManager,
asks for a user and password

Quedale, 2023-03-19:
Thank you for this feedback.
The fact that they are discoverable on the official ONVIF Manager on windows confirm they should be compatible. (This eliminates the devices)

Something else to look into is the network configuration. Can you ping the camera's IPs from your linux machine?
If you are running it from a VM configured with a NAT NIC, the discovery will be limited to the virtual NAT network.
Using a bridged adapter should resolve it in such case. 

Could you also provide the application logs?

If everything is configured correctly and nothing shows up in the application logs, capturing messages using wireshark might be helpful.

Looking forward to hear back.

Thanks!

Quedale, 2023-03-19:
Please note that I personally only worked off open source Onvif camera solution like [rpos](https://github.com/Quedale/rpos), since I'm too broke to afford an official Onvif Camera with backchannel. (Backchannel was my focus until I finished it recently)

I'm looking forward to support official devices if you are willing to help out.

boc99, 2023-03-19:
HI
I'm sure I'm willing to help you
My linux machine is real not virtual
Even the windows 10 machine is real
Ping is ok
The difference I noticed :
Onvifmanager doesn't ask me to authenticate to the camera
That window finds it and asks me to login

Sorry if my english is poor

boc99, 2023-03-21:
> Thank for the help!
>
> The authentication happens after the device has been discovered.
> The discovery is unprotected by design. The response message from the
> camera should contain the "DeviceService" url, which would triggers the
> authentication.
>
> ONVIF Device Manager on windows uses a different approach than I do. Since
> you provide the credentials before the discovery, it is capable to
> retrieving the camera snapshot immediately after the discovery. This
> approach requires all camera to use the same credentials, which I find is a
> security risk.
>
> This project uses a different approach where you provide credentials for
> each individual cameras after the discovery.
> The discovery will show limited camera details until credentials are
> provided.
> Unfortunately I didn't include the ability to save credentials yet, so it
> has to be entered every time.
>
> A protected device should show up like this after discovery:
> [image: image]
> <https://user-images.githubusercontent.com/13303174/226494625-4b59ade5-f3d5-4c27-812b-1cea4d4a9c4b.png>
>
> After selecting it it should ask for credentials:
> [image: image]
> <https://user-images.githubusercontent.com/13303174/226494689-5e497242-da2c-499c-9cf6-eb000e676666.png>
>
> After providing the credentials it should finally show the stream (The
> snapshot not updating is a bug I will fix soon):
> [image: image]
> <https://user-images.githubusercontent.com/13303174/226495106-706dc093-ce98-4afa-b11b-88df9a8336dd.png>
>
> That being said, you don't seem to get a response from the camera at all.
> I will build a Linux Mint 21 machine and test it to see if it blocks it
> for some reason.
>
> Until then, can you provide the application logs after running at least
> one scan please?
>
> Note #1 <https://github.com/Quedale/OnvifDeviceManager/issues/1>: The
> screenshot includes my local changes and may look slightly different than
> what you have.

HI.

This is what I see .

[image: Schermata del 2023-03-21 10-19-26.png]

But it doesn't go further even by clicking on the address

Quedale, 2023-03-22:
Just a small update that I did find issues in the credentials dialog callback.

I'll try to get it fixed in the next commit since this is essentially breaking the application.

Quedale, 2023-03-29:
I've added several changes in the last commit which most likely include a fix for your issue.
There was a core issue handling multi-threaded soap calls, which caused issues after authentication.

The changes are also included in the subprojects "[OnvifSoapLib](https://github.com/Quedale/OnvifSoapLib)" and "[OnvifDiscoveryLib](https://github.com/Quedale/OnvifDiscoveryLib)".
For the sake of simplicity you can run the following commands to remove them before rebuilding the project:
```
rm -rf subprojects/OnvifDiscoveryLib
rm -rf subprojects/OnvifSoapLib
git pull
./autogen.sh %YOUR_CONFIG_PARAMS%
```
autogen.sh will automatically download, configure and compile missing projects.

Note that OnvifSoapLib takes a while to build from scratch, so alternatively you can find and remove "onvifdisco.pc" and "onvifsoap.pc" instead of removing the entire projects. autogen.sh will pull the latest changes and make an incremental build.

Let me know how that goes!

boc99, 2023-03-29:
HI,
i removed everything
i recompiled
Here it stops

lib/libcrypto.a(dso_dlfcn.o): in function `dlfcn_unload':
dso_dlfcn.c:(.text+0x6b8): undefined reference to `dlclose'
collect2: error: ld returned 1 exit status
make[5]: *** [Makefile:435: wsdl2h] Errore 1
make[5]: uscita dalla directory
«/home/boc/OnvifDeviceManager/subprojects/gsoap-2.8/gsoap/wsdl»
make[4]: *** [Makefile:357: all] Errore 2
make[4]: uscita dalla directory
«/home/boc/OnvifDeviceManager/subprojects/gsoap-2.8/gsoap/wsdl»
make[3]: *** [Makefile:828: all-recursive] Errore 1
make[3]: uscita dalla directory
«/home/boc/OnvifDeviceManager/subprojects/gsoap-2.8/gsoap»
make[2]: *** [Makefile:456: all] Errore 2
make[2]: uscita dalla directory
«/home/boc/OnvifDeviceManager/subprojects/gsoap-2.8/gsoap»
make[1]: *** [Makefile:473: all-recursive] Errore 1
make[1]: uscita dalla directory
«/home/boc/OnvifDeviceManager/subprojects/gsoap-2.8»
make: *** [Makefile:381: all] Errore 2
*****************************
*** Make failed gsoap-2.8 ***
*****************************

boc99, 2023-03-29:
Hi , in the subproyects directory there are gsoap-2.8 and openssl.
 I would have expected OnviifSoapLib and OnvifDiscoveryLib but I don't see
them
SSDs also fail
I replaced it

Quedale, 2023-03-29:
RIP SSD! :headstone:

That had to potential to wast a lot of your time. 
At least you found it quickly.

boc99, 2023-03-29:
Hi
gsoap 2.8.91-2

g++    -I/usr/lib/ssl/include -DWITH_OPENSSL -DWITH_GZIP
-DWSDL2H_IMPORT_PATH="\"/home/boc/OnvifDeviceManager/subprojects/gsoap-2.8/build/dist/share/gsoap/WS\""
-g -O2 -L../../gsoap/wsdl -I../../gsoap -I../../gsoap/plugin  -o wsdl2h
wsdl2h-wsdl2h.o wsdl2h-wsdl.o wsdl2h-wadl.o wsdl2h-schema.o wsdl2h-types.o
wsdl2h-service.o wsdl2h-soap.o wsdl2h-mime.o wsdl2h-wsp.o wsdl2h-bpel.o
wsdl2h-wsdlC.o wsdl2h-httpda.o wsdl2h-smdevp.o ../../gsoap/libgsoapssl++.a
-L/usr/lib/ssl/lib  -lssl -lcrypto -lz -lpthread
/usr/bin/ld:
/home/boc/OnvifDeviceManager/subprojects/openssl/build/dist/lib/../lib/libcrypto.a(dso_dlfcn.o):
in function `dlfcn_globallookup':
dso_dlfcn.c:(.text+0x17): undefined reference to `dlopen'
/usr/bin/ld: dso_dlfcn.c:(.text+0x2a): undefined reference to `dlsym'
/usr/bin/ld: dso_dlfcn.c:(.text+0x35): undefined reference to `dlclose'
/usr/bin/ld:
/home/boc/OnvifDeviceManager/subprojects/openssl/build/dist/lib/../lib/libcrypto.a(dso_dlfcn.o):
in function `dlfcn_bind_func':
dso_dlfcn.c:(.text+0x1b7): undefined reference to `dlsym'
/usr/bin/ld: dso_dlfcn.c:(.text+0x282): undefined reference to `dlerror'
/usr/bin/ld:
/home/boc/OnvifDeviceManager/subprojects/openssl/build/dist/lib/../lib/libcrypto.a(dso_dlfcn.o):
in function `dlfcn_load':
dso_dlfcn.c:(.text+0x2f5): undefined reference to `dlopen'
/usr/bin/ld: dso_dlfcn.c:(.text+0x369): undefined reference to `dlclose'
/usr/bin/ld: dso_dlfcn.c:(.text+0x3a5): undefined reference to `dlerror'
/usr/bin/ld:
/home/boc/OnvifDeviceManager/subprojects/openssl/build/dist/lib/../lib/libcrypto.a(dso_dlfcn.o):
in function `dlfcn_pathbyaddr':
dso_dlfcn.c:(.text+0x466): undefined reference to `dladdr'
/usr/bin/ld: dso_dlfcn.c:(.text+0x4d7): undefined reference to `dlerror'
/usr/bin/ld:
/home/boc/OnvifDeviceManager/subprojects/openssl/build/dist/lib/../lib/libcrypto.a(dso_dlfcn.o):
in function `dlfcn_unload':
dso_dlfcn.c:(.text+0x6b8): undefined reference to `dlclose'
collect2: error: ld returned 1 exit status
make[5]: *** [Makefile:435: wsdl2h] Errore 1
make[5]: uscita dalla directory
«/home/boc/OnvifDeviceManager/subprojects/gsoap-2.8/gsoap/wsdl»
make[4]: *** [Makefile:357: all] Errore 2
make[4]: uscita dalla directory
«/home/boc/OnvifDeviceManager/subprojects/gsoap-2.8/gsoap/wsdl»
make[3]: *** [Makefile:828: all-recursive] Errore 1
make[3]: uscita dalla directory
«/home/boc/OnvifDeviceManager/subprojects/gsoap-2.8/gsoap»
make[2]: *** [Makefile:456: all] Errore 2
make[2]: uscita dalla directory
«/home/boc/OnvifDeviceManager/subprojects/gsoap-2.8/gsoap»
make[1]: *** [Makefile:473: all-recursive] Errore 1
make[1]: uscita dalla directory
«/home/boc/OnvifDeviceManager/subprojects/gsoap-2.8»
make: *** [Makefile:381: all] Errore 2
*****************************
*** Make failed gsoap-2.8 ***
*****************************

boc99, 2023-03-29:
maybe this is the problem : CDPATH="${ZSH_VERSION+.}:" && cd . && /bin/bash '/home/boc/OnvifDeviceManager/subprojects/gsoap-2.8/missing' autoheader)
aclocal.m4:17: warning: this file was generated for autoconf 2.71.

You have another version of autoconf.  It may work, but is not guaranteed to.
If you have problems, you may need to regenerate the build system entirely.
To do so, use the procedure documented by the package, typically 'autoreconf'.

My version2.69 autoconf

Too sensitive to versions

Quedale, 2023-03-29:
The subproject gsoap-2.8 wasn't written by me. It's a library I use.
I have no control over their dependencies.

That being said, I did build a Linux Mint 21 VM and I have autoconf 2.71 installed from apt.
Your distro ins't up-to-date?

That being said, last week you did successfully build gsoap in order to compile the project. I highly doubt this is the cause now.
Did you change the environment you are building it on?

The errors that I can see above are:
`undefined reference to 'dlerror'` 
`undefined reference to 'dlclose'`   
`undefined reference to 'dlopen'`   

These symbols should be part of `libdl` which should already be installed. 
Do you still have `libc6-dev` installed?

boc99, 2023-03-30:
I rebuilt virtual mint 21 because it was giving me problems with the
keyboard stopping working even though I replaced it.
using a bluethoot keyboard it didn't show the pin for pairing.
I recompiled everything and finally got the binary.
I launched it here is what I see. It seems nothing has changed

Quedale, 2023-03-30:
I don't see any screenshot in any of your posts. (It seems picture attachments don't make it through by emails)

Here's how it comes out:
[image]

Could you launch the application with the environment variable "GST_DEBUG=3" set and share the application logs?

At this point, I will really need the logs.

Quedale, 2023-03-30:
I just thought about the fact that this project currently only support H264 stream and AAC audio.

Honwywell H4W4PER2 - Supports H265 and H264
hIKVISION DS-2CD22512F - Supports H264 and MJPEG

If your cams are set to H265 and MJPEG. the player won't be able to render the stream yet. (I will eventually add support for those)

boc99, 2023-03-30:
Hi
***@***.***:~/OnvifDeviceManager/dist/bin$ ./onvifmgr
EventQueue__create...
EventQueue__start...
EventQueue__start...
EventQueue__start...
EventQueue__start...
Using Gstreamer Version : 1.22.0.0
ProbMatches__create...
ProbMatch__create...
GTK Thread Dispatch...
Found server ---------------- 1
Prob Match #0 ----------------
OnvifDevice__create
OnvifSoapClient__create [http://192.168.1.102/onvif/device_service http://
[fe80::4619:b6ff:fe68:e132]/onvif/device_service]
Created Device:
protocol -- http
ip : 192.168.1.102
port -- (null)
DEBUG List Record :[0] 192.168.1.102:(null)
NSLookup ... 192.168.1.102
ProbMatches__destroy...
ProbMatch__destroy...
GTK Thread Dispatch Done...
OnvifDevice__device_createMediaSoap
OnvifDevice__device_getCapabilities
OnvifSoapClient__create [http://192.168.1.102/onvif/Media]
OnvifDevice__media_getSnapshotUri
SOAP 1.2 fault SOAP-ENV:Sender ["http://www.onvif.org/ver10/error
":NotAuthorized]
"The action requested requires authorization and the sender is not
authorized"
Detail: [no detail]
ERROR : profile index out-of-bounds.
GTK Thread Dispatch...
GTK Thread Dispatch Done...
Failed to get hostname ...
Retrieved hostname : (null)
switch_ui_page 1
update_details_page 0
No device selected...
switch_ui_page 0
OnvifPlayer__stop
OnvifDevice__media_getStreamUri
Error: soap struct state not initialized
ERROR : profile index out-of-bounds.
set location : (null)

(onvifmgr:1842): GLib-CRITICAL **: 20:46:15.385: g_str_has_prefix:
assertion 'str != NULL' failed

(onvifmgr:1842): GStreamer-RTSP-CRITICAL **: 20:46:15.385:
gst_rtsp_url_parse: assertion 'urlstr != NULL' failed
OnvifPlayer__play
switch_ui_page 1
update_details_page 0
dev : 192.168.1.102
OnvifPlayer__stop
Errore di segmentazione (core dump creato)
***@***.***:~/OnvifDeviceManager/dist/bin$ ./onvifmgr
GST_DEBUG=3
EventQueue__create...
EventQueue__start...
EventQueue__start...
EventQueue__start...
EventQueue__start...
Using Gstreamer Version : 1.22.0.0
ProbMatches__create...
ProbMatch__create...
GTK Thread Dispatch...
Found server ---------------- 1
Prob Match #0 ----------------
OnvifDevice__create
OnvifSoapClient__create [http://192.168.1.102/onvif/device_service http://
[fe80::4619:b6ff:fe68:e132]/onvif/device_service]
Created Device:
protocol -- http
ip : 192.168.1.102
port -- (null)
DEBUG List Record :[0] 192.168.1.102:(null)
NSLookup ... 192.168.1.102
GTK Thread Dispatch Done...
ProbMatches__destroy...
ProbMatch__destroy...
OnvifDevice__device_createMediaSoap
OnvifDevice__device_getCapabilities
Errore di segmentazione (core dump creato)
***@***.***:~/OnvifDeviceManager/dist/bin$

boc99, 2023-03-30:
Honwywell H4W4PER2  H264
hIKVISION DS-2CD22512F H264

Quedale, 2023-03-31:
Unfortunately, at this point I will really need the application logs in order to help.
The screenshots also didn't make it, so I don't have much to work with.

Can you please share the application logs?

Quedale, 2023-03-31:
Okay, I think you get `SOAP 1.2 fault SOAP-ENV:Sender ["http://www.onvif.org/ver10/error
":NotAuthorized]` when calling `OnvifDevice_get_profiles` which leads to `ERROR : profile index out-of-bounds.`.

There is no fallback, in case GetProfiles fails. 
According to the specs, there is no mention that ProfileToken is a mandatory field in order to make calls like GetSnapshotUri or GetStreamUri.

Let me make some adjustments so that it can use the default profile without needing additional authorization.

Until then, do you know why you don't have the necessary access? (I would like to understand the use-case)
Does those camera support multiple profiles?

Quedale, 2023-03-31:
I also noticed that the port comes out null.
`port -- (null)`

I didn't handle this scenario and always expected a port. Let me put defaults 80 and 443 in place.

Thanks for this valuable feedback!

Quedale, 2023-03-31:
And another thing!!

I didn't realize that in the discovery there could be two address within one probe. (IPv4 and IPv6)
`OnvifSoapClient__create [http://192.168.1.102/onvif/device_service http://
[fe80::4619:b6ff:fe68:e132]/onvif/device_service]`

Thanks again, this is good stuff.

boc99, 2023-03-31:
Onvif manager windows 

Name HIKVISION DS-2CD2512F-I
Onvif version 2.40
Url http://192.168.1.102/onvif/device_service
http 80
rtsp 554
resolution 640x480
encoding h264
ONVIF discoverable

Quedale, 2023-03-31:
Once again I've made changes to [OnvifDeviceManager](https://github.com/Quedale/OnvifDeviceManager), [OnvifDiscoveryLib ](https://github.com/Quedale/OnvifDiscoveryLib)and [OnvifSoapLib](https://github.com/Quedale/OnvifSoapLib).

[OnvifDiscoveryLib ](https://github.com/Quedale/OnvifDiscoveryLib)will now properly handle cameras with multiple IP addresses.
[OnvifSoapLib ](https://github.com/Quedale/OnvifDiscoveryLib)will now attempt to use the default onvif profile if the soap call `GetProfiles` fails (This is the reason it didn't work for you)
[OnvifDeviceManager](https://github.com/Quedale/OnvifDeviceManager) will temporarely defaults to the first ProbeMatch XAddr until networking preferences are added.

Here's a better script to pull and compile the subprojects changes. Run the following from the project's top directory:
```
rm subprojects/OnvifDiscoveryLib/build/dist/lib/pkgconfig/onvifdisco.pc
rm subprojects/OnvifSoapLib/build/dist/lib/pkgconfig/onvifsoap.pc
git pull
./autogen.sh %YOUR_CONFIG_PARAMS%
```

Only removing  the ".pc" files will cause autogen.sh to perform an incremental build.

Let me know how that goes.

Thanks for the time you are putting on this, it is really helpful.

Quedale, 2023-03-31:
Okay I can see something else wrong.

Since ONVIF doesn't offer an authentication url, each url proceeds to authenticate. 
In order to create the [Media](https://www.onvif.org/ver10/media/wsdl/media.wsdl) soap, "GetCapabilities" has to be call to retrieve the various URLs.
For this reason, I picked "GetCapabilities" has a credentials checkpoint.

It seems that in your case, the camera doesn't actually protect it, so it is accessible and allows the creation of the media soap anonymously.
Therefor the application doesn't show you any credential dialog, because it thinks the camera is unprotected.

Let me add a check for "GetStreamURI", which is the most important URL to open the stream.

Quedale, 2023-04-01:
Hopefully this will do it.

This time, I've made a considerable update to [autogen.sh](https://github.com/Quedale/OnvifDeviceManager/blob/main/autogen.sh). It will now detect upstream change, pull them and force an incremental build.

So this time, you only need to do this: (No more removing stuff inside subprojects)
```
git pull
./autogen.sh %YOUR_CONFIG_PARAMS%
```

Once again, please let me know how that goes!

Quedale, 2023-04-01:
I managed to reproduce the EXACT same errors as you did, but only under very specific conditions:
1. Linux Mint VM inside Virtualbox under Windows 10
2. The error may only occur after a fresh boot. (After running it the second time it seems to work fine)

This seems to indicate a race condition that only occurs on lower specs since my VM is running very slow. (Some windows 10 issue)

I really can't seem to reproduce this issue on any native hosts I have.

I'll keep digging.

boc99, 2023-04-01:
***@***.***:~/OnvifDeviceManager$ ./autogen.sh
%YOUR_CONFIG_PARAMS%
openssl already found.
zlib already found.
gsoap already found.
glib already found.
Recupero di origin in corso
OnvifDiscoveryLib is already up-to-date. Do nothing...
onvifdisco already found.
Recupero di origin in corso
OnvifSoapLib is already up-to-date. Do nothing...
onvifsoaplib already found.
alsa-lib already found.
Gstreamer already installed.
configure: WARNING: you should use --build, --host, --target
configure: WARNING: invalid host type: %YOUR_CONFIG_PARAMS%
checking for %YOUR_CONFIG_PARAMS%-gcc... no
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables...
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether the compiler supports GNU C... yes
checking whether gcc accepts -g... yes
checking for gcc option to enable C11 features... none needed
checking whether gcc understands -c and -o together... yes
checking whether make supports nested variables... yes
checking for %YOUR_CONFIG_PARAMS%-pkg-config... no
checking for pkg-config... /usr/bin/pkg-config
checking pkg-config is at least version 0.9.0... yes
checking for
        gstreamer-1.0 >= 1.20.3
    ... no
configure: gstreamer >= 1.20.3 not found. - Using latest static libraries
configure: Setting static Gstreamer path
checking build system type... Invalid configuration `%YOUR_CONFIG_PARAMS%':
machine `%YOUR_CONFIG_PARAMS%-unknown' not recognized
configure: error: /bin/bash ./config.sub %YOUR_CONFIG_PARAMS% failed
***@***.***:~/OnvifDeviceManager$

I'm doing something wrong ?

boc99, 2023-04-01:
What linux do you use?
I could make the same PC again
I don't like windows i use linux mint .
A derivative of ubuntu derived from debian
It doesn't exist for Raspberry pity
I'm using mint 20.3 with vmware play 17
running mint 21.1

Quedale, 2023-04-01:
> I'm doing something wrong ?

`%YOUR_CONFIG_PARAMS%` was meant to be your preferred autoconf parameters.

Something along the line of this:
`./autogen.sh --prefix=$(pwd)/dist`

> What linux do you use?

If you want to test something closer to my setup, I'm currently using Kubuntu 22.04.
I know it works fine under my WSL VM running Ubuntu as well.

My current setup currently runs WSL and Virtualbox, leading to very poor VM performance. I'm suspecting this poor performance is the reason I can reproduce it.

At least I managed to reproduce the same issue you got, so I can work from that. 
I have my suspicions on a particular piece, but troubleshooting this will take a considerable amount of time since the issue occurs randomly for me.

It should be running just as well inside a VM running any popular distro, so I want to fix this.

Quedale, 2023-04-03:
Can you please try it once more?

I've added several fixed targeting race conditions that I didn't catch without adding artificial delays.
I can now run this on a lower than average spec VM running Linux Mint.

Let me know if you have any issues now.

Thanks!

boc99, 2023-04-03:
HI,
now after pressing scan a padlock with camera ip appears.
Pressing on the padlock a login window appears
the padlock disappears and the image appears.
The Details tab is all SomeName
Seems like a good way

Quedale, 2023-04-03:
Sounds like everything is working as expected.

Did the stream start successfully as well?

The details page is still a placeholder, but I can start working on this relatively soon.

boc99, 2023-04-03:
The image is on the right a bit small,
but it shows

Quedale, 2023-04-03:
The stream will render according to the camera's resolution or the space on available screen.
in short, it won't scale up, but it will scale down to fit on screen.

So if you are running a 4K monitor and your camera streams at 640x480, it will look very small.

I'll take that as a possible enhancement to add a scale up option.

Are we good to close this issue?

boc99, 2023-04-03:
Ok

Quedale, 2023-04-05:
Thanks for your help.

Feel free to report additional issues or submit feature requests.

Cheers!

closed 2023-04-05 by Quedale