Quedale/OnvifDeviceManager #2: Commpilation

mladenb, 2023-03-30:
Hi and thank you for sharing this code.

Just out of the curiosity, why did you build the openssl from scratch? Couldn't you use libopenssl from openssl-dev package?

Quedale, 2023-03-31:
[autogen.sh](https://github.com/Quedale/OnvifDeviceManager/blob/main/autogen.sh) will check if openssl exists in the system first. (More accurately, it will check for the libcrypto package)
As a fallback mechanism, it will download and compile it.

Thank you for bringing this up, the wrong optional package was listed in [README.md#Autogen, Configure, Download and build dependencies](https://github.com/Quedale/OnvifDeviceManager#autogen-configure-download-and-build-dependencies). 
The optional dependency "openssl" should be "libssl-dev". 

What distro/version are you using if openssl-dev is valid in your case?

mladenb, 2023-03-31:
Yes, libssl-dev should be a correct one. Thanks for updating the Readme.

I'm using Ubuntu 22.04.2 LTS

Quedale, 2023-03-31:
Okay I just wanted to confirm the distro if "openssl-dev" was accurate for you.

Did you encounter any issues building the project?

mladenb, 2023-04-01:
actually, yes, this one:

```
*****************************
*** Copy repo from cache ***
*** /home/user2/Downloads/OnvifDeviceManager/subprojects/.cache/gstreamer-1.22.0 ***
*****************************
*****************************
* Building Github Project ***
* Src dir : gstreamer ***
* Prefix : /home/user2/Downloads/OnvifDeviceManager/subprojects/gstreamer/build/dist ***
*****************************
*****************************
*** Download Subprojects gstreamer ***
*****************************
setup patch : 
*****************************
*** Meson Setup gstreamer ***
*****************************
The Meson build system
Version: 0.61.2
Source dir: /home/user2/Downloads/OnvifDeviceManager/subprojects/gstreamer
Build dir: /home/user2/Downloads/OnvifDeviceManager/subprojects/gstreamer/build
Build type: native build

meson.build:1:0: ERROR: Meson version is 0.61.2 but project requires >= 0.62.0

A full log can be found at /home/user2/Downloads/OnvifDeviceManager/subprojects/gstreamer/build/meson-logs/meson-log.txt
*****************************
*** Meson Setup failed gstreamer ***
*****************************

```

mladenb, 2023-04-01:
I guess the latest LTS version of Ubuntu is using meson v0.61.2, judging from this page: https://packages.ubuntu.com/search?keywords=meson

Quedale, 2023-04-01:
> I guess the latest LTS version of Ubuntu is using meson v0.61.2, judging from this page: https://packages.ubuntu.com/search?keywords=meson

That's right, GStreamer raised the minimum meson version in their latest release.

You can get the latest meson from PIP.

Start by removing the APT package:
```
sudo apt remove meson
```

Then install python pip and meson:
```
sudo apt install python3-pip
python3 -m pip install meson
```

Then you have to make sure that your PATH include python's bin directory.
```
export PATH=$HOME/.local/bin:$PATH
```

I want to add a fallback where it will build meson from source in this scenario. 
I just didn't get around to it yet.

Let me know how that goes.

mladenb, 2023-04-01:
that worked, thanks :) it's been built and working now :)

Quedale, 2023-04-01:
Good to hear.

My primary focus until now was performance and stability around the discovery and the player.
Almost no ONVIF feature were implemented yet.

Feel free to submit enhancement request if there is any feature you would like added.

closed 2023-04-01 by Quedale