Quedale/OnvifDeviceManager #3: Error while autogen

johnyfernandes, 2023-06-13:
Hello! First of all, thank you for your time for building this awesome app, I really want to try it! I'm just having some problems when running ./autogen.sh --prefix=$(pwd)/dist --enable-latest, it gives me this error after a while:

```
*****************************
*** compile OnvifSoapLib ***
*** Make Args :  ***
*****************************
  CC       /home/john/Downloads/OnvifDeviceManager/subprojects/gsoap-2.8/gsoap/plugin/libonvifsoap_la-wsaapi.lo
  CC       /home/john/Downloads/OnvifDeviceManager/subprojects/gsoap-2.8/gsoap/plugin/libonvifsoap_la-wsseapi.lo
  CC       /home/john/Downloads/OnvifDeviceManager/subprojects/gsoap-2.8/gsoap/plugin/libonvifsoap_la-wsddapi.lo
  CC       /home/john/Downloads/OnvifDeviceManager/subprojects/gsoap-2.8/gsoap/custom/libonvifsoap_la-struct_timeval.lo
  CC       src/libonvifsoap_la-onvif_device.lo
  CC       src/libonvifsoap_la-client.lo
  CC       src/generated/libonvifsoap_la-soapClient.lo
  CC       src/generated/libonvifsoap_la-soapC.lo
/home/john/Downloads/OnvifDeviceManager/subprojects/gsoap-2.8/gsoap/custom/struct_timeval.c: In function ‘soap_xsd__dateTime2s’:
/home/john/Downloads/OnvifDeviceManager/subprojects/gsoap-2.8/gsoap/custom/struct_timeval.c:82:72: warning: format ‘%d’ expects argument of type ‘int’, but argument 4 has type ‘__suseconds_t’ {aka ‘long int’} [-Wformat=]
   82 |   (SOAP_SNPRINTF(soap->tmpbuf + n, sizeof(soap->tmpbuf) - n, 10), ".%.6dZ", a.tv_usec);
      |                                                                     ~~~^    ~~~~~~~~~
      |                                                                        |     |
      |                                                                        int   __suseconds_t {aka long int}
      |                                                                     %.6ld
```
I am using PopOS, the latest version with Gnome 44. 
Sorry for bothering, and again, thank you! 😄

Quedale, 2023-06-13:
The log provided seems truncated, so I can't really tell the cause.

I never tested PopOS, so there must be a missing dependancy I was not aware of.
I'll setup a VM and fix this next time I got a minute to spare.

Quedale, 2023-06-14:
I managed to successfully run autogen.sh from a fresh PopOS install.
The only issue I encountered was the package meson from apt was too old, so I installed it from python-pip instead.

That being said, you may be running out of memory.
The file "soapC.c" is automatically generated from WSDL files, so it is considerably big and requires a fair amount of RAM. 
How much RAM do you have?

Fortunately this is only required during build time, so you can temporarily increase your SWAP partition to run it once.
The application itself is pretty light.

johnyfernandes, 2023-06-14:
Hello! First of all, thank you for your attention, and sorry for the delay answering!
Right now, I am not home and my computer is not with me, but in 2 hours I'll be there and it'll be easier, but I am running dual boot (Win11 and PopOS), and PopOS only has 90gb and 3gb to boot, I think, so that might be the problem, right?
I have total 16gb!

When I get home, I will update you, again, thank you for all your attention!

Quedale, 2023-06-14:
16GB of RAM is more than enough. My VM has only 4GB and was successful.

Disk space could also be the cause. I've experienced all kind of unexpected issues once my root drive gets full.  (e.g. no errors)
autogen.sh will download and build dependencies which requires additional space.

I'll make an enhancement note to check disk space available before running the script.  :thumbsup:

johnyfernandes, 2023-06-15:
Hello! It's resolved now, all my fault, there was some problems with the GPU drivers and the Pop OS drivers... Thank you very much for your app, everything is working perfectly, congrats and good job!

Quedale, 2023-06-19:
Thanks! I'm actually glad it wasn't my script!

Feel free to open enhancement request if there are feature you would like to see first.

closed 2023-06-19 by Quedale