Quedale/OnvifDeviceManager #5: error compilation - seems -ldl missing (Ubuntu 20.04)

ptamb3, 2023-06-21:
Following instructions received an error.
  
  ```
mv -f .deps/wsdl2h-httpda.Tpo .deps/wsdl2h-httpda.Po
  gcc -DHAVE_CONFIG_H -I. -I../..  -I../../gsoap -I../../gsoap/plugin  -DLINUX     -I/usr/lib/ssl/include -DWITH_OPENSSL -DWITH_GZIP -g -O2 -MT wsdl2h-smdevp.o -MD -MP -MF .deps/wsdl2h-smdevp.Tpo -c -o wsdl2h-smdevp.o `test -f '../../gsoap/plugin/smdevp.c' || echo './'`../../gsoap/plugin/smdevp.c
  mv -f .deps/wsdl2h-smdevp.Tpo .deps/wsdl2h-smdevp.Po
  mv -f .deps/wsdl2h-mime.Tpo .deps/wsdl2h-mime.Po
  mv -f .deps/wsdl2h-bpel.Tpo .deps/wsdl2h-bpel.Po
  mv -f .deps/wsdl2h-soap.Tpo .deps/wsdl2h-soap.Po
  mv -f .deps/wsdl2h-wsdl2h.Tpo .deps/wsdl2h-wsdl2h.Po
  mv -f .deps/wsdl2h-wsp.Tpo .deps/wsdl2h-wsp.Po
  mv -f .deps/wsdl2h-wadl.Tpo .deps/wsdl2h-wadl.Po
  mv -f .deps/wsdl2h-wsdl.Tpo .deps/wsdl2h-wsdl.Po
  mv -f .deps/wsdl2h-types.Tpo .deps/wsdl2h-types.Po
  mv -f .deps/wsdl2h-schema.Tpo .deps/wsdl2h-schema.Po
  mv -f .deps/wsdl2h-service.Tpo .deps/wsdl2h-service.Po
  mv -f .deps/wsdl2h-wsdlC.Tpo .deps/wsdl2h-wsdlC.Po
  g++    -I/usr/lib/ssl/include -DWITH_OPENSSL -DWITH_GZIP -DWSDL2H_IMPORT_PATH="\"/home/user/Downloads/OnvifDeviceManager/subprojects/gsoap-2.8/build/dist/share/gsoap/WS\"" -g -O2 -L../../gsoap/wsdl -I../../gsoap -I../../gsoap/plugin  -o wsdl2h wsdl2h-wsdl2h.o wsdl2h-wsdl.o wsdl2h-wadl.o wsdl2h-schema.o wsdl2h-types.o wsdl2h-service.o wsdl2h-soap.o wsdl2h-mime.o wsdl2h-wsp.o wsdl2h-bpel.o wsdl2h-wsdlC.o wsdl2h-httpda.o wsdl2h-smdevp.o ../../gsoap/libgsoapssl++.a -L/usr/lib/ssl/lib  -lssl -lcrypto -lz -lpthread 
  /usr/bin/ld: /home/user/Downloads/OnvifDeviceManager/subprojects/openssl/build/dist/lib/../lib/libcrypto.a(dso_dlfcn.o): in function `dlfcn_globallookup':
  dso_dlfcn.c:(.text+0x17): undefined reference to `dlopen'
  /usr/bin/ld: dso_dlfcn.c:(.text+0x2a): undefined reference to `dlsym'
  /usr/bin/ld: dso_dlfcn.c:(.text+0x35): undefined reference to `dlclose'
  /usr/bin/ld: /home/user/Downloads/OnvifDeviceManager/subprojects/openssl/build/dist/lib/../lib/libcrypto.a(dso_dlfcn.o): in function `dlfcn_bind_func':
  dso_dlfcn.c:(.text+0x1b7): undefined reference to `dlsym'
  /usr/bin/ld: dso_dlfcn.c:(.text+0x282): undefined reference to `dlerror'
  /usr/bin/ld: /home/user/Downloads/OnvifDeviceManager/subprojects/openssl/build/dist/lib/../lib/libcrypto.a(dso_dlfcn.o): in function `dlfcn_load':
  dso_dlfcn.c:(.text+0x2f5): undefined reference to `dlopen'
  /usr/bin/ld: dso_dlfcn.c:(.text+0x369): undefined reference to `dlclose'
  /usr/bin/ld: dso_dlfcn.c:(.text+0x3a5): undefined reference to `dlerror'
  /usr/bin/ld: /home/user/Downloads/OnvifDeviceManager/subprojects/openssl/build/dist/lib/../lib/libcrypto.a(dso_dlfcn.o): in function `dlfcn_pathbyaddr':
  dso_dlfcn.c:(.text+0x466): undefined reference to `dladdr'
  /usr/bin/ld: dso_dlfcn.c:(.text+0x4d7): undefined reference to `dlerror'
  /usr/bin/ld: /home/user/Downloads/OnvifDeviceManager/subprojects/openssl/build/dist/lib/../lib/libcrypto.a(dso_dlfcn.o): in function `dlfcn_unload':
  dso_dlfcn.c:(.text+0x6b8): undefined reference to `dlclose'
  collect2: error: ld returned 1 exit status
  make[5]: *** [Makefile:435: wsdl2h] Error 1
  make[5]: Leaving directory '/home/user/Downloads/OnvifDeviceManager/subprojects/gsoap-2.8/gsoap/wsdl'
  make[4]: *** [Makefile:357: all] Error 2
  make[4]: Leaving directory '/home/user/Downloads/OnvifDeviceManager/subprojects/gsoap-2.8/gsoap/wsdl'
  make[3]: *** [Makefile:828: all-recursive] Error 1
  make[3]: Leaving directory '/home/user/Downloads/OnvifDeviceManager/subprojects/gsoap-2.8/gsoap'
  make[2]: *** [Makefile:456: all] Error 2
  make[2]: Leaving directory '/home/user/Downloads/OnvifDeviceManager/subprojects/gsoap-2.8/gsoap'
  make[1]: *** [Makefile:473: all-recursive] Error 1
  make[1]: Leaving directory '/home/user/Downloads/OnvifDeviceManager/subprojects/gsoap-2.8'
  make: *** [Makefile:381: all] Error 2
```

Quedale, 2023-06-21:
These symbols should be part of libdl which should already be installed on ubuntu.
Do you have libc6-dev installed?

I started working on Ubuntu 22, so I never tested version 20 explicitly.
I'll setup a VM and give it a shot.

Quedale, 2023-06-22:
Okay, first, thanks for your feedback!

I found a solution and it should be available tomorrow.

Quedale, 2023-06-22:
You can pull the latest change and give it a shot!

You were right that the "-ldl" flag was missing within the gsoap-2.8 Makefile.
It's missing because this is no longer required in newer versions.
I don' t own this project, so I can't fix it directly, but I was able to  force the flag anyway .

I also caught a few additional issues specific for this version of Ubuntu.

I'm looking forward to hear back.

ptamb3, 2023-06-22:
Got a different error now with a warning prior `libtool: warning: relinking 'libatopology.la'`

```
  git switch -c <new-branch-name>

Or undo this operation with:

  git switch -

Turn off this advice by setting config variable advice.detachedHead to false

Updating files: 100% (10458/10458), done.
*****************************
*** Copy repo from cache ***
*** /home/user/Downloads/OnvifDeviceManager/subprojects/.cache/gstreamer-1.22.3 ***
*****************************
*****************************
* Building Github Project ***
* Src dir : gstreamer ***
* Prefix : /home/user/Downloads/OnvifDeviceManager/subprojects/gstreamer/build/dist ***
*****************************
*****************************
*** Download Subprojects gstreamer ***
*****************************
setup patch : 
*****************************
*** Meson Setup gstreamer ***
*****************************
./autogen.sh: line 479: /home/user/.local/bin/meson: No such file or directory
*****************************
*** Meson Setup failed gstreamer ***
*****************************
```

Quedale, 2023-06-22:
Okay, the issue was that autogen.sh was checking if meson was installed, but not the version.
The script would only fallback if the tool isn't installed at all.

I improved it so that it will get the latest version from pip if it doesn't meet the minimum requirement.

Give it a shot a let me know!

Thanks again for this feedback, it allowed to improve the autogen script.

ptamb3, 2023-06-22:
Okay so now seems like that process completed. 

I see openh264 warning 
`gst-plugins-bad| Run-time dependency openh264 found: NO (tried pkgconfig and cmake)` 

Other warnings were:

`Program dotnet-format found: NO`.
`libdrm| Run-time dependency pciaccess found: NO (tried pkgconfig and cmake)`
`gst-plugins-bad| Dependency libva from subproject libva found: NO (subproject failed to configure)`
`libdrm| subprojects/libdrm-2.4.111/meson.build:237:16: Exception: Dependency "pciaccess" not found, tried pkgconfig and cmake`
`gst-plugins-bad| Header "faac.h" has symbol "faacEncOpen" : NO`
`gst-plugins-bad| Run-time dependency fdk-aac found: NO (tried pkgconfig and cmake)`

Not sure if any of these are a blocker.

 I will run `make` and `make install`

gst-plugins-base 1.22.3

    Plugins: app, audioconvert, audioresample, audiotestsrc, overlaycomposition, pbtypes, playback, rawparse, tcp, typefindfunctions, videoconvertscale, videorate, videotestsrc, volume, alsa,
             ximagesink

gst-plugins-good 1.22.3

    Plugins: audioparsers, autodetect, navigationtest, debug, interleave, alaw, mulaw, level, rtp, rtpmanager, rtsp, udp, video4linux2, png, pulseaudio

gstreamer 1.22.3

    Plugins: coreelements

gstreamer-full 1.22.3

  Build options
    gstreamer-full library             : YES
    Tools                              :

  Subprojects
    fdk-aac                            : YES
    gst-devtools                       : NO Feature 'devtools' disabled
    gst-editing-services               : NO Feature 'ges' disabled
    gst-examples                       : NO Feature 'gst-examples' disabled
    gst-integration-testsuites         : NO Feature 'devtools' disabled
    gst-libav                          : NO Feature 'libav' disabled
    gst-omx                            : NO Feature 'omx' disabled
    gst-plugins-bad                    : YES 4 warnings
    gst-plugins-base                   : YES 1 warnings
    gst-plugins-good                   : YES
    gst-plugins-rs                     : NO Feature 'rs' disabled
    gst-plugins-ugly                   : NO Feature 'ugly' disabled
    gst-python                         : NO Feature 'python' disabled
    gst-rtsp-server                    : NO Feature 'rtsp_server' disabled
    gstreamer                          : YES 3 warnings
    gstreamer-sharp                    : NO Feature 'sharp' disabled
    gstreamer-vaapi                    : NO Feature 'vaapi' disabled
    libdrm                             : NO Dependency "pciaccess" not found, tried pkgconfig and cmake
    libnice                            : NO Feature 'libnice' disabled
    libva                              : NO Neither a subproject directory nor a libva.wrap file was found.
    openh264                           : YES

  User defined options
    auto_features                      : disabled
    buildtype                          : release
    default_library                    : static
    includedir                         : include
    libdir                             : lib
    prefix                             : /home/user/Downloads/OnvifDeviceManager/subprojects/gstreamer/build/dist
    strip                              : True
    bad                                : enabled
    base                               : enabled
    good                               : enabled
    gpl                                : enabled
    glib:tests                         : false
    gst-plugins-bad:fdkaac             : enabled
    gst-plugins-bad:nvcodec            : enabled
    gst-plugins-bad:onvif              : enabled
    gst-plugins-bad:openh264           : enabled
    gst-plugins-bad:v4l2codecs         : enabled
    gst-plugins-bad:videoparsers       : enabled
    gst-plugins-base:alsa              : enabled
    gst-plugins-base:app               : enabled
    gst-plugins-base:audioconvert      : enabled
    gst-plugins-base:audioresample     : enabled
    gst-plugins-base:audiotestsrc      : enabled
    gst-plugins-base:overlaycomposition: enabled
    gst-plugins-base:pbtypes           : enabled
    gst-plugins-base:playback          : enabled
    gst-plugins-base:rawparse          : enabled
    gst-plugins-base:tcp               : enabled
    gst-plugins-base:typefind          : enabled
    gst-plugins-base:videoconvertscale : enabled
    gst-plugins-base:videorate         : enabled
    gst-plugins-base:videotestsrc      : enabled
    gst-plugins-base:volume            : enabled
    gst-plugins-base:x11               : enabled
    gst-plugins-good:audioparsers      : enabled
    gst-plugins-good:autodetect        : enabled
    gst-plugins-good:debugutils        : enabled
    gst-plugins-good:interleave        : enabled
    gst-plugins-good:law               : enabled
    gst-plugins-good:level             : enabled
    gst-plugins-good:png               : enabled
    gst-plugins-good:pulse             : enabled
    gst-plugins-good:rtp               : enabled
    gst-plugins-good:rtpmanager        : enabled
    gst-plugins-good:rtsp              : enabled
    gst-plugins-good:udp               : enabled
    gst-plugins-good:v4l2              : enabled
    libdrm:cairo-tests                 : false
    x264:cli                           : false

Quedale, 2023-06-22:
The logs seems truncated, but everything looks fine.

autogen.sh builds dependency and may show several warnings outside my control.
If the script finished successfully, you should be good to compile the project now.

ptamb3, 2023-06-22:
[image]

```
user@user-master-2004:~/Downloads/OnvifDeviceManager (main)$ make install
make[1]: Entering directory '/home/user/Downloads/OnvifDeviceManager'
 /usr/bin/mkdir -p '/home/user/Downloads/OnvifDeviceManager/dist/bin'
  /bin/bash ./libtool   --mode=install /usr/bin/install -c onvifmgr '/home/user/Downloads/OnvifDeviceManager/dist/bin'
libtool: install: /usr/bin/install -c onvifmgr /home/user/Downloads/OnvifDeviceManager/dist/bin/onvifmgr
make[1]: Nothing to be done for 'install-data-am'.
make[1]: Leaving directory '/home/user/Downloads/OnvifDeviceManager'
user@user-master-2004:~/Downloads/OnvifDeviceManager (main)$ /home/user/Downloads/OnvifDeviceManager/dist/bin/onvifmgr
Initializing Gstreamer plugins...
Loading static plugins...
Gstreamer plugins initialized...
Using Gstreamer Version : 1.22.3.0
EventQueue__create...
Reading Settings file onvifmgr_settings.ini
WARNING no config file found. Using default configs. 1
Creating backchannel using source element pulsesrc
EventQueue__start...
EventQueue__start...
EventQueue__start...
EventQueue__start...
EventQueue__start...
EventQueue__start...
EventQueue__start...
EventQueue__start...
GTK Thread Dispatch...
```

Seems like that binary built fine now. Thanks.

closed 2023-06-22 by ptamb3

Quedale, 2023-06-22:
I'm glad you got it working.

Feel free to open more issues if you find some or enhancement request if you'd like to see features prioritized.

Thanks again!

ptamb3, 2023-06-22:
Sure. Will do. Thanks for fixing the issues! Appreciate the prompt response.