Quedale/OnvifDeviceManager #6: Doesn't find any cameras - suggestion: add "manually add camera" function

sgofferj, 2023-07-02:
Somehow in my environment, no camera is detected. I'm running it on my desktop under Ubuntu 22.04 which has several network interfaces, for docker networks, zerotier, etc. Maybe a "manually ad camera" function would help?

Quedale, 2023-07-03:
Thank for you for this feedback!

It is leveraging UDP broadcast, so it might be invalid network configuration or even a network appliance stopping the traffic.
e,g #1: On a virtual machine, a NAT interface won't allow the broadcast packet to leave the virtual network.
e.g #2: A firewall may block broadcast traffic to a camera DMZ. (enterprise scenario)

There's more potential causes, but more information about your environment would be helpful to narrow it down.

That being said, ONVIF devices aren't necessarily discoverable by default for obvious reasons. 
Therefor "manual add camera" would be a **great addition** and relatively easy to do.
I make sure to add it on the next commit.

Are you able to ping the camera from your environment?
If you can't ping it, manually adding the camera wouldn't work either.

sgofferj, 2023-07-03:
Great!

Yeah, that particular camera is in the same network as the PC in question. I can ping it, access it via browser, VLC and the Windows ONVIF Device Manager finds it. Also, Zoneminder and Shinobi were able to find it.

Quedale, 2023-07-03:
I already started working on the ability to add cameras manually and should be available relatively soon. 

I'm still curious to find out more about the failed discovery.
Are you using IPv6?
- The discovery is current set for IPv4 since I don't have an IPv6 network to test.

Is there anything virtualized?
- If you are compiling and running this project within a VM, it is possible the bridge adapter (e.g. KVM Macvtap) doesn't forward SSDP packets.

Is there anything showing in the logs during discovery?

sgofferj, 2023-07-04:
No v6, plain v4 with a /24 subnet. No virtualization.
Nothing stood out in the logs. I'll try again and paste logs.

sgofferj, 2023-07-04:
```
┌─[sgofferj@enterprise][~/source/OnvifDeviceManager]
└──▪ ./onvifmgr 
Initializing Gstreamer plugins...
Loading static plugins...
Gstreamer plugins initialized...
Using Gstreamer Version : 1.22.3.0
EventQueue__create...
Reading Settings file onvifmgr_settings.ini
WARNING no config file found. Using default configs. 1
Creating backchannel using source element pulsesrc
EventQueue__start...
EventQueue__start...
EventQueue__start...
EventQueue__start...
EventQueue__start...
EventQueue__start...
EventQueue__start...
EventQueue__start...
GTK Thread Dispatch...
GTK Thread Dispatch Done...
GTK Thread Dispatch...
GTK Thread Dispatch Done...
GTK Thread Dispatch...
GTK Thread Dispatch Done...
GTK Thread Dispatch...
GTK Thread Dispatch Done...
GTK Thread Dispatch...
GTK Thread Dispatch Done...
GTK Thread Dispatch...
GTK Thread Dispatch Done...
RtspPlayer__stop 
```

Quedale, 2023-07-04:
There's clearly nothing wrong showing in the logs.

I did look at gsoap's documentation and found that the default TTL value for UDP probes is set to 1.
This is for security reason to prevent potential discovery "leak" on miss-configured network.

I'll increase this value since I don't consider myself responsible for a weak network security.

I'll let you know when it's available. (I'm still working ability to add cameras manually)

Quedale, 2023-07-04:
I got some good news!

I ended up analyzing the windows client's network traffic and figured out the most likely cause.

The windows client will actually send 3 different probes with different Types (Device, NetworkVideoDisplay and NetworkVideoTransmitter).
It also does it TWICE and aggregate the results.  (Total of 9 probes for one discovery).

In my case, I submit only one probe without any type defined.
It seems that not all devices will respond to such probe like in your case.

Thanks again for this valuable feedback!

Out of curiosity, what is the brand/model of your camera?

sgofferj, 2023-07-05:
> Out of curiosity, what is the brand/model of your camera?
Some "Techage" camera I bought a couple of years ago. It's marketed as ONVIF compliant and so far, even the event stream in Shinobi works with it.
It's the only one not in my camera VLAN because I was so far too lazy to route a second cable to the room :rofl:

Quedale, 2023-07-07:
Some status update.

I got everything working on my local workspace. I just need to clean it up a bit before commit.
I should have some time tomorrow to finish it up.

This includes capability to add ONVIF cameras manually and a fix for the failed discovery.

P.S. Yea, cabling can be a major pain

Quedale, 2023-07-08:
The repo has been updated.
The commit ended up considerably larger than I expected.

After pulling the latest changes, you will need to run `autogen.sh` once more, in order to pull the latest and rebuild [OnvifDiscoveryLib](https://github.com/Quedale/OnvifDiscoveryLib) and [OnvifSoapLib](https://github.com/Quedale/OnvifSoapLib).

I still have some work to do in order to consider this completed (e.g. Saving manually added cameras), but you should have something functional now.

Let me know how that goes!

sgofferj, 2023-07-10:
Will test.

sgofferj, 2023-07-10:
Scan finds the camera now. Awesome! :)

Quedale, 2023-07-11:
I'm super glad its working now!

I'll close this issue although the feature isn't entirely completed, given that it's now working as intended.

Feel free to submit additional feature request if there's anything else you'd like to see added.

Thanks for giving my application a try!

closed 2023-07-11 by Quedale