Quedale/OnvifDeviceManager #7: click scan core dump

Mxk123456, 2023-07-17:
GTK Thread Dispatch...
Found server - Match count : 1
OnvifDevice__create
OnvifSoapClient__create [http://192.168.100.18:80/onvif/device_service]
Valid URL
Created Device:
	protocol -- http
	ip : 192.168.100.18
	hostname : (null)
	port -- 80
	endpoint : onvif/device_service
DEBUG List Record :[0] 192.168.100.18:80
GTK Thread Dispatch Done...
NSLookup ... 192.168.100.18
OnvifDevice_authenticate [http://192.168.100.18:80/onvif/device_service]
OnvifDevice__device_createMediaSoap
OnvifDevice__device_getCapabilities [http://192.168.100.18:80/onvif/device_service]
ProbMatches__destroy...
ProbMatch__destroy...
ProbMatches__create...
ProbMatch__create...
GTK Thread Dispatch...
Found server - Match count : 1
Record already part of list [http://192.168.100.18:80/onvif/device_service]
GTK Thread Dispatch Done...
ProbMatches__destroy...
ProbMatch__destroy...
ProbMatches__create...
ProbMatch__create...
段错误

Quedale, 2023-07-17:
First, thank you for trying out my project and providing feedback!

My last commit also included small changes to [OnvifSoapLib](https://github.com/Quedale/OnvifSoapLib) sub-project and it would crash there if it didn't get updated.

Can you try running `autogen.sh` again?
It will automatically pull and recompile its dependencies.

I  need to add a note to README.md to indicate that running autogen.sh is required after pulling changes from git.

Quedale, 2023-09-19:
I will close this issue since I didn't get a response and also that everything went under massive refactoring since.

Several critical bug were also fixed as part of Issue #11.

closed 2023-09-19 by Quedale