Quedale/OnvifDeviceManager #8: Any potential to use with a DVR

PeteC1773, 2023-07-19:
Hi, I installed this thinking I might be able to use it with a DVR (a chinese, floureon 8 channel cctv recorder), but scan does not pick the machine up and using the IP address does not work either. Maybe I'm confused about the protocol. With this particular box sofiactl.pl does work, to query the settings and the XMeye android app works too. Will these apps be using onvif and therefore potentially compatible with your project? Or is the sofia stuff completely different? I want primarily to be able to play recorded video and configure the box remotely. Your project seems more ready to use than the previously mentioned ones. Thx!

Quedale, 2023-07-20:
I don't see why not. This project does implement implement the ONVIF client specification (partially for now).

If the device you are using is [this one](https://www.aliexpress.com/item/32575470576.html), it may work. Although it doesn't say anything about what specification it implements.

I need to dig a little to understand it's capabilities. Does it show support for onvif as a client or NVT device? Which onvif profile is it certified for? (It's not exactly clear)
At first glance, it looks like this NVR is capable of multiple specification including ONVIF and SofiaCTL which are completely different but not exclusive.

Something for sure is that all ONVIF  devices require same same entry point to manage it: https://CameraIP:443/onvif/device_service.  This is the URL used to manage a ONVIF capable device from a client.

This is taken directly from the [ONVIF Core Specification](https://www.onvif.org/specs/core/ONVIF-Core-Specification.pdf) section 5.1.1

What URL are you trying to enter? 
(I don't actually need the IP)

I am more than willing to support any ONVIF capable device.
I could support non-ONVIF streams like RTSP, but it would lack any ONVIF capabilities, which is the primary goal of this tool.

PeteC1773, 2023-07-20:
That is just about exactly the box I am using - but its a good few years old, maybe 5 to 7. The only visible difference on the back panel is the video out BNC connector is missing - but its possible I actually removed that myself a few years ago when I added a hardware mod.

However hitting it on 443 gets a connection refused. A tcp port scan gives this:
80/tcp    open  http
554/tcp   open  rtsp
34567/tcp open  dhanalakshmi

So no 443? No Onvif? The software version for what its worth is 'V4.02.R11.00031149.10001.131900.00000'.

I can connect with ffplay to the rtsp port, and play live video. And port 80 hosts a very basic (broken) web page, I think this usually embeds the NETsurveillance ActiveX plugin (it shows the video - the rtsp stream I assume - i've never had it working because I have no windows machines). The web page offers no other control apart from camera panning.

And then I guess the 34567 port is for 'sofia', which is what XMEye connects to? IDK for sure. But I downloaded the android xmeye app and it does connect using the local connection option, once the login details are given and the ip address of the box. So XMeye can configure the device (although some things are missing configuration wise - its a bit basic and badly laid out).

I tried the above URL you posted but on http instead of https and I got a 404 :-(

I guess this all means no joy?

Thanks, Pete

Quedale, 2023-07-20:
Based on your port listening, the ONVIF Discovery service isn't running since it would listen on port 3702.
Note that the discovery feature can be turned off for security reasons as well.

It is also possible that the ONVIF capabilities might be turned off entirely by default and available as an option.
That would make sense for a DVR IMO, given that this isn't necessarily its primary use.

Note that the ONVIF specification itself is based on HTTP(S). It could technically run on any given port, but the discovery is defined on port 3702.

Do you have a model number so that I can dig up some documentation?
I did notice a few variant of "floureon 8 DVR" and some don't seem to support ONVIF.

I don't mind looking further into this to figure it out or even consider additional specifications if  it is documented.
It would obviously be hard to develop a specification without any device to test it with tho.

You are right that port 34567 is for sofia.
I never heard of it before this, so I'll go read a bit on it.

Supporting RAW rtsp stream would be relatively easy to do. (Lacking any management feature)

Quedale, 2023-07-21:
Looking forward to here back!
If it is truly ONVIF capable, it should work.  (Assuming it is enabled)

I checked for sofiactl, unfortunately I couldn't find the documentation.
On the sofiactl repo, it shows `vendor specifications: http://wiki.xm030.com:81/`, but the link is dead.
I couldn't it even find the offical specification name.

I don't mind attempting to support proprietary protocols, but I don't want to reverse engineer undocumented code while not having the device on hand.
This would be almost impossible.

PeteC1773, 2023-07-23:
Sorry - schools have just broken up so its a busy couple of days! Unfortunately it seems bad news. I have found time to mess around and dig a bit deeper over the last few days. There is no onvif options in the menus, its really bland there, very few options (in contrast to some ip cameras that I have that have a similar interface). This machine does not implement onvif it seems.

OTOH what is interesting, although you may not (currently) want to go down this route but for when you do get some hardware that supports it, the sofia protocol seems fairly basic and with a bit of wiresharking together with examining some other tools I think I see how it works. The sofiactl perl script is useful but does not implement the correct method to watch recorded streams - it can download them but only in whole chunks. The XMEye app uses a different method, 'PlayMode' is 'ByTime' instead of 'ByName' which is what the sofiactl script uses. For your future use, if you ever decide to check this out, what seems to happen is that the script (or xmeye app) opens a connection to 34567 and sends a password and such, searches for some file names by time (in the app, you drag a bar to view a certain time, the app sends this time to the server in one such query), gets a bit of info back and then uses this to open a second connection to the same port with 'Action' being 'Claim'. And then the DVR or camera streams back the recording starting from the time specified in the 'Start' query (the stream is sent on this second connection though - the original search connection stays open ready to accept [from the client] a 'Stop' and a new 'Start' at a different time when you scrub to view a different time).

If I get a working perl script / python / whatever I'll send it to you in case you want to ever incorporate this sofia stuff into your own tool. 

Thx for the help btw!

Quedale, 2023-08-13:
Sorry for the slow reply, I also have loads to deal with and this slipped my mind.

The lack of ONVIF support is unfortunate.

I would honestly welcome the challenge, since I'm doing all this out of pure interest.
Unfortunately I have little budget for my hobbies, hence why I'm doing a lot of the source myself with cheap hardware.
I chose ONVIF since the specs and some demo sample were readily available.

I highly suspect the "sofia spec" changed license a few years back and they pulled it from the public.
Allowing only trusted partners to access the actual documentation.
According to the internet archive, the last time it got a copy was in [2017](https://web.archive.org/web/20171014104017/http://wiki.xm030.com:81/)
Unfortunately the archive didn't actually pick up the files. So nothing really interesting there.

That being said, should anyone be interested in hardware donations, I could start from that! :P
Not that I need the hardware in any way, it would be purely academic.

Or even if someone wants to make contributions to support it in a non-invasive way. 
The player and ONVIF libraries are not really tied to each other, so it wouldn't be that hard to refactor the code to support additional protocols. (The player still has some kinks to fix and ONVIF is will supported just not fully implemented)
¯\\_(ツ)_/¯

I'm going to close this one, but feel free to open another one if you do find relevant documentation.

Cheers!

closed 2023-08-13 by Quedale