Quedale/OnvifDeviceManager #10: Add a web UI

edurenye, 2023-08-30:
Would be nice to have a web UI to use remotely.

Quedale, 2023-09-10:
Good day and thanks for this feedback!

I would be lying if I said I didn't consider this before starting this project.
The truth is that offering a web UI involves a lot more overhead. (Web Server + Web Frontend)

Codec compatibility is also an issue since browsers may not natively support the camera's stream and would require transcoding.
Although more accessible, it would result with less camera compatibility.
Since the goal is to manage and configure ONVIF cameras, it has to offer the best compatibility as possible.

This repository doesn't house any ONVIF specific functionality, but rather a GTK frontend which leverage [OnvifSoapLib ](https://github.com/Quedale/OnvifSoapLib) and [OnvifDiscoveryLib](https://github.com/Quedale/OnvifDiscoveryLib).
I designed both of these libraries so that they can be reused on other projects.
For that reason, if I start working on a Web version, it would have it's own repository.

That being said, the goal of this tool is to offer ONVIF management capabilities which are relatively scarce.
There are a few excellent Web based NVR for monitoring purpose, minus ONVIF features. (e.g. [MotionEye ](https://github.com/motioneye-project/motioneye) or [Frigate](https://github.com/blakeblackshear/frigate))
That doesn't mean that I won't ever attempt it. I fully consider working on it once this project and related libraries gained more maturity. (I'm also considering an Android client)

Thanks again for this feedback, it is truly appreciated regardless.

closed 2023-09-10 by Quedale

edurenye, 2023-09-10:
Hi, thanks for considering it.
For it I didn't mean to turn this tool into a NVR, I just wannted the same features but being able to use it with docker so I do not have to deal with local dependencies or python envs. That is why I asked for the web UI.

edurenye, 2023-09-10:
Also with a docker image I could add it to home assistant then maybe.

Quedale, 2023-09-10:
I will dive into this in the future. 

Leveraging a server instance instead of a native application changes the mindset around the flows.

WebRTC could be leveraged to open streams directly to individual cameras using P2P, which would behave closer to what this project does, although camera codec support on browsers would be a shot in the dark.

WebRTC/HLS using the web server as central service. This would concentrate the heavy lifting on the web server and could open up to additional capabilities. (e.g. Computer Vision, Motion sensor, AI Inference, Transcoding, etc)
This would be a greater challenge, but would offer greater versatility and compatibility.

In either case docker seems like the best course for deployment. 

I'm not a docker expert and I just learned that it can run GUI application, I originally thought it was exclusively for headless services.
The ultimate selfish goal of this project was simply to learn new things and this is actually interesting. 
This looks like a great opportunity to learn more about Docker, so I'll look into the feasibility to make this a docker UI app.

edurenye, 2023-09-10:
Well, I didn't know that Docker could run GUI applications, sounds interesting, will look into it too. Thanks!