Quedale/OnvifDeviceManager #11: GUI client closes when entering credentials

g33kenn, 2023-09-06:
Hi! Thanks for your work!
I'm not sure if I'm doing something wrong but I have the following behavior:

- When hitting the Scan button, it's able to find my discoverable ONVIF devices.
- Clicking on a tile asks me for creds
- When entering the creds, the GUI app just closes

g33kenn, 2023-09-06:
I'm on Debian 12, sorry should have mentioned !

Quedale, 2023-09-10:
Hi! Sorry for the long wait. I've been busy away from the computer.
Thank you for this feedback.

You are definitely not doing anything wrong.

I never tested debian, so I need to setup a VM and give is a shot before I can properly answer this.
That being said, all kind of things happens after entering the credentials, including starting the stream.

Can you provide the application logs to understand where the crash occurs?

Quedale, 2023-09-10:
I had time to setup a VM last night and give it a shot.
It worked fine out of the box on a fresh debian 12.1 build.

There are multiple factor that can make it result in different behavior on your end.
1. Your decoding hardware. What CPU/GPU are you using and driver versions?
2. Camera model and stream configuration. (codec and parameters)

I unfortunately only have a few Mali and Nvidia GPUs and no AMD card to test with.
In regards of camera configuration, I can mimic virtually anything since I built a custom one.

Application logs would also be a great help.

I'm very interested in knowing what causes this failure and resolve it.

I did noticed a regression in my last commit where the thumbnail won't load properly on a secure camera, but didn't cause a crash. 
I'll look into it.

g33kenn, 2023-09-11:
Good Morning!

I really appreciate you answering and even installing your own Debian 12 to
double check. Sorry I was out of town in a cottage and just came back
yesterday.
I'm remoting on the machine right now

I attach the config.log (please let me know if you meant another file) also
the device info
I feel like something went wrong during the installation. I'm too noob in
the linux multiverse to be able to assess

Please let me know if you need anything else!

*Elvin RAHAMEFY*

Quedale, 2023-09-11:
Unfortunately files don't attach when replying using emails.

You'll either have to create a link to that file or past the content in clear text.

You can easily attach files using the web interface by simply drag and drop over the comment box. 
It will upload it to github and generate a URL for it.

g33kenn, 2023-09-11:
hey! try this instead

https://u.pcloud.link/publink/show?code=XZrac3VZuveGc4jjcwmQmGbEyIBLdbLWlLGV

Quedale, 2023-09-11:
The log files you sent aren't application log, but rather configuration/compile logs.

There are no app log files automatically generated yet. It only prints to the console.
So either you copy the console content our pipe the log output to a log file.
To achieve the latter,  simply run the application `./onvifmgr > app.log`.

The section `> app.log` indicate to send all console log generated by `./onvifmgr` to app.log.

Sorry for the confusion!

g33kenn, 2023-09-11:
1. I run  ./onvifmgr > app.log.
   2. the GUI shows
   3. I Click on SCAN
   4. Click on the IP that was found
   5. Enter my creds and click Login
   6. GUI disappears and below message appears in the console

free(): double free detected in tcache 2
Aborted

*Elvin RAHAMEFY*

Quedale, 2023-09-11:
Can you share the resulting app.log?

After clicking on login, it dispatches several parallel events and this will give me insight on the progress of each events.

Thanks!

g33kenn, 2023-09-11:
See below
https://u.pcloud.link/publink/show?code=XZOxo3VZhKpAYhwMO87Yo6gRIdlsiSeX4k2V

Reading the log i can see that it's looking for a file named
onvifmgr_settings.ini
But I don't see any in the OnvifDeviceManager folder

*Elvin RAHAMEFY*

Quedale, 2023-09-14:
Unfortunately, the log file you sent didn't show anything wrong.
The logs shows a graceful shutdown, just like if you clicked on the "Quit" button.
Maybe you ran the application one more time, which overwrote the previous log file.

That being said, I did spend some time testing it and managed to make it crash once.
The cause seems to be the "gtk_entry_get_text" called from a thread.
It seems to be caused by a race condition that may occur more or less often depending on the hosts specifications. (CPU threads, GPU acceleration, etc.)

I'll try to fix this later this evening and see if that resolves your problem.

Thanks for bringing this to my attention!

g33kenn, 2023-09-14:
weirdly enough, the commands creates the log file but doesn't write
anything on it when it crashes. I even tried to change to export to another
file.same behavior, file gets created but nothing written when i open the
app or make it crash. The one I sent probably comes from 1 time where I
just opened and closed.that would be the reason.

Anyway, really appreciate you troubleshooting, please let me know When The
new version is out and I'll test and keep you posted! But no rush at all!

Elvin.

Quedale, 2023-09-14:
What I would except from a log file "showing" (but not really showing) is an abrupt stop of log output.
This would give me an indicator of the progress of each thread when it crashed.
The log file you sent doesn't even show that the "Scan" or "Add" button were used, but gracefully quit.
Basically : Launch --> Quit

The file you sent shows a graceful shutdown generated by a user event. (Either clicking on "Quit" or closing the window)

Regardless, I hope that the fault I found will resolve your issue.

g33kenn, 2023-09-15:
Hi!

I'll try the new version!

But I was actually able to generate logs by just running onvifmgr from the
console without redirecting the logs. I just had to copy/paste the console
content into a logfile after. See below.
https://u.pcloud.link/publink/show?code=XZOxo3VZhKpAYhwMO87Yo6gRIdlsiSeX4k2V

*Elvin RAHAMEFY*

Quedale, 2023-09-16:
Thank you so much!

That helped me determine that the issue is within [OnvifSoapLib](https://github.com/Quedale/OnvifSoapLib).

I found several point of concerns to improve, but I think I found possible root cause.

Inside the file [onvif_device.c](https://github.com/Quedale/OnvifSoapLib/blob/main/src/onvif_device.c):
 - The first is that I don't properly clean up previously cached onvif profile data. Inside function `OnvifProfiles__destroy`, on `line 498` and `line 499`, I forgot to specify the index to clean with `[i]`

- The second is that I don't handle one thread lock properly. Inside the function `OnvifDevice__device_getCapabilities`, I meant to lock the thread on `line 78`, but mistakenly used "unlock" instead.

- I noticed the existence of empty struct inside the gsoap library which can lead to unexpected behavior depending on the compiler used. I will add the necessary gsoap built-in compile flag to prevent this. Although I doubt this had an impact on you.

I was extremely novice when I first started to write this library so will also take this opportunity to clean/improve it.

I will let you know once it's ready for testing, it shouldn't take too long.

Thanks for your patience!

g33kenn, 2023-09-16:
awesome, thanks for the detailed answer!

Quedale, 2023-09-17:
New update.

I found quite several critical issues, from race conditions to invalid memory allocation.
Once again, I was very novice when I wrote this, so I'm in the process or refactoring the entire library.
I still consider myself novice, but at least I properly understand memory allocation now.

The changes should be available soon. 
I just have to clean my code and test it a little bit more considering the size of the change.

Thanks again for reporting this issue. That made me review old code and catch a large amount of issues.

Cheers!

Quedale, 2023-09-18:
Alright, I just committed the first part of the OnvifSoapLib project refactoring.
This should address several possible cause for your crash.

To test the change you will need to run `git pull` in order to get the latest from `OnvifDeviceManager`.

After you pulled the latest changes you will need to run ./autogen.sh` %YOUR_ARGUMENTS_HERE%` once again.
This will pull the latest change and rebuild OnvifSoapLib.

Once `autogen.sh` is complete, simple run `make` and/or `make install`.

Let me know how that goes!

g33kenn, 2023-09-18:
Hi!

Appreciate the quick release time! I've just installed the new release. I
confirm it does not close anymore after entering my creds. I don't still it
works tho so I pulled the log again.

See the logs
https://u.pcloud.link/publink/show?code=XZ0peGVZkBCQhLpM8b5SP0eXBXWcU5WpI6ay

But I really think I do something wrong when running the autogen command.
what kind of argument an I supposed to have? I just use the exact same line
as the Readme page, I don't change anything...

./autogen.sh --prefix=$(pwd)/dist --enable-latest

*Elvin RAHAMEFY*

Quedale, 2023-09-18:
Actually this one is simple.
I simply forgot to pass the credentials to open the rtsp stream.

The error:
`good/gst/rtsp/gstrtspsrc.c(6906): gst_rtspsrc_send (): /GstPipeline:onvif-pipeline/GstRTSPSrc:rtspsrc:
Unauthorized (401)`

This one should be a lot simpler to fix.

Funny enough this highlighted an issue on another open source project I'm using [rpos](https://github.com/Quedale/rpos). The project support protecting the camera with credentials, but it apparently doesn't secure the rtsp stream.

Thank so much for providing this feedback.

Quedale, 2023-09-19:
Good day!

I made the necessary change to support RTSP authentication.

Repeat the same steps as last time since I finished the refactoring of [OnvifSoapLib](https://github.com/Quedale/OnvifSoapLib).

From the root directory
```
git pull
./autogen.sh %YOUR_ARGUMENTS_HERE%
make
```

Thanks for your patience testing my project!

g33kenn, 2023-09-22:
Hi!

Just tried it!
Description of the new behavior:
No more crash after entering the creds but it still doesnt work

   1. Opening the App
   2. Clicking on Scan
   3. Clicking on the found IP
   4. Entering Creds
   5. Cliking on Login

Result

   - The cred prompt stays
   - Nothing Else happens (I don't access the camera feed)
   - The count of the Task Manager Tab incrase by 1 (1/8)
      - When clicking on the Task Manager Tab, it's a blank screen saying
      "Work In Progress"

link to the log
https://u.pcloud.link/publink/show?code=XZ7p3GVZhzvOq6jYDVuxuME1kMIFDByMm8NV

*Elvin RAHAMEFY*

Quedale, 2023-09-23:
From what I understands from the log, some event dispatched hanged and caused a lock up.
A wild guess is the GetSnapshot call on the camera that takes longer than mine here.
I'll dig deeper and try to simulate such behavior.

That being said, I was kinda expecting such thing, which is why I started working on a task manager.
Unfortunately it isn't complete yet and only shows "Work In Progress".
The only indicator implemented so far is the PendingEvents/NumThread label.
Adding a Logging framework showing the thread ID and time would also be very useful.

I just finished rewriting OnvifSoapLib to support cross-platform, so I will be able to dive further into this. 

Just to clarify, it looks like you pressed "Scan..." a few times before selecting the camera. (hence multiple slow GetSnapshot for the same camera were invoked)
Is that accurate?

g33kenn, 2023-09-25:
I don't think I pressed several time. Could be wrong but I always do a
dedicated "simple" run through from a fresh Terminal before copy/pasting
the content. Do you want me to do another one again?

*Elvin RAHAMEFY*

Quedale, 2023-09-27:
You can sure try it again, but I'm actually working on a logging library to help the troubleshooting process.

I have my suspicions on what could be the cause, but over-using some mutex locks inside the player.
Although I am unable to reproduce this lock up on any of my system and I can't picture the faulty flow in my head.

Once the logging library is complete, I will jump back on the Task Manager GUI.

Quedale, 2024-02-03:
Almost everything was reviewed and updated since this issue was reported.
Closing this issue since it is most likely out-dated by now.

Feel free to open a new issue if you are still experiencing it.

closed 2024-02-03 by Quedale