Quedale/OnvifDeviceManager #15: cannot make installation

kmorber, 2024-02-03:
Hello,

i have an error during "make":

root@tvheadend:/isos/OnvifDeviceManager# make
  CC       src/onvifmgr-onvif-mgr.o
Package gstreamer-video-1.0 was not found in the pkg-config search path.
Perhaps you should add the directory containing `gstreamer-video-1.0.pc'
to the PKG_CONFIG_PATH environment variable
No package 'gstreamer-video-1.0' found
Package alsa was not found in the pkg-config search path.
Perhaps you should add the directory containing `alsa.pc'
to the PKG_CONFIG_PATH environment variable
No package 'alsa' found
Package gstde265 was not found in the pkg-config search path.
Perhaps you should add the directory containing `gstde265.pc'
to the PKG_CONFIG_PATH environment variable
No package 'gstde265' found
Package gstjpeg was not found in the pkg-config search path.
Perhaps you should add the directory containing `gstjpeg.pc'
to the PKG_CONFIG_PATH environment variable
No package 'gstjpeg' found
Package gstjpegformat was not found in the pkg-config search path.
Perhaps you should add the directory containing `gstjpegformat.pc'
to the PKG_CONFIG_PATH environment variable
No package 'gstjpegformat' found
Package gstlevel was not found in the pkg-config search path.
Perhaps you should add the directory containing `gstlevel.pc'
to the PKG_CONFIG_PATH environment variable
No package 'gstlevel' found
Package gstximagesink was not found in the pkg-config search path.
Perhaps you should add the directory containing `gstximagesink.pc'
to the PKG_CONFIG_PATH environment variable
No package 'gstximagesink' found
Package gstoverlaycomposition was not found in the pkg-config search path.
Perhaps you should add the directory containing `gstoverlaycomposition.pc'
to the PKG_CONFIG_PATH environment variable
No package 'gstoverlaycomposition' found
Package gstpulseaudio was not found in the pkg-config search path.
Perhaps you should add the directory containing `gstpulseaudio.pc'
to the PKG_CONFIG_PATH environment variable
No package 'gstpulseaudio' found
Package gstopenh264 was not found in the pkg-config search path.
Perhaps you should add the directory containing `gstopenh264.pc'
to the PKG_CONFIG_PATH environment variable
No package 'gstopenh264' found
Package gstfdkaac was not found in the pkg-config search path.
Perhaps you should add the directory containing `gstfdkaac.pc'
to the PKG_CONFIG_PATH environment variable
No package 'gstfdkaac' found
Package gstv4l2codecs was not found in the pkg-config search path.
Perhaps you should add the directory containing `gstv4l2codecs.pc'
to the PKG_CONFIG_PATH environment variable
No package 'gstv4l2codecs' found
Package gstvideo4linux2 was not found in the pkg-config search path.
Perhaps you should add the directory containing `gstvideo4linux2.pc'
to the PKG_CONFIG_PATH environment variable
No package 'gstvideo4linux2' found
Package gstvideoparsersbad was not found in the pkg-config search path.
Perhaps you should add the directory containing `gstvideoparsersbad.pc'
to the PKG_CONFIG_PATH environment variable
No package 'gstvideoparsersbad' found
Package gstvideoconvertscale was not found in the pkg-config search path.
Perhaps you should add the directory containing `gstvideoconvertscale.pc'
to the PKG_CONFIG_PATH environment variable
No package 'gstvideoconvertscale' found
Package gstrtponvif was not found in the pkg-config search path.
Perhaps you should add the directory containing `gstrtponvif.pc'
to the PKG_CONFIG_PATH environment variable
No package 'gstrtponvif' found
Package gstudp was not found in the pkg-config search path.
Perhaps you should add the directory containing `gstudp.pc'
to the PKG_CONFIG_PATH environment variable
No package 'gstudp' found
Package gstcoreelements was not found in the pkg-config search path.
Perhaps you should add the directory containing `gstcoreelements.pc'
to the PKG_CONFIG_PATH environment variable
No package 'gstcoreelements' found
Package gstrtsp was not found in the pkg-config search path.
Perhaps you should add the directory containing `gstrtsp.pc'
to the PKG_CONFIG_PATH environment variable
No package 'gstrtsp' found
Package gstrtpmanager was not found in the pkg-config search path.
Perhaps you should add the directory containing `gstrtpmanager.pc'
to the PKG_CONFIG_PATH environment variable
No package 'gstrtpmanager' found
Package gstrtp was not found in the pkg-config search path.
Perhaps you should add the directory containing `gstrtp.pc'
to the PKG_CONFIG_PATH environment variable
No package 'gstrtp' found
Package gstmulaw was not found in the pkg-config search path.
Perhaps you should add the directory containing `gstmulaw.pc'
to the PKG_CONFIG_PATH environment variable
No package 'gstmulaw' found
Package gstinterleave was not found in the pkg-config search path.
Perhaps you should add the directory containing `gstinterleave.pc'
to the PKG_CONFIG_PATH environment variable
No package 'gstinterleave' found
Package gstautodetect was not found in the pkg-config search path.
Perhaps you should add the directory containing `gstautodetect.pc'
to the PKG_CONFIG_PATH environment variable
No package 'gstautodetect' found
Package gstaudioparsers was not found in the pkg-config search path.
Perhaps you should add the directory containing `gstaudioparsers.pc'
to the PKG_CONFIG_PATH environment variable
No package 'gstaudioparsers' found
Package gstalsa was not found in the pkg-config search path.
Perhaps you should add the directory containing `gstalsa.pc'
to the PKG_CONFIG_PATH environment variable
No package 'gstalsa' found
Package gstvolume was not found in the pkg-config search path.
Perhaps you should add the directory containing `gstvolume.pc'
to the PKG_CONFIG_PATH environment variable
No package 'gstvolume' found
Package gstvideorate was not found in the pkg-config search path.
Perhaps you should add the directory containing `gstvideorate.pc'
to the PKG_CONFIG_PATH environment variable
No package 'gstvideorate' found
Package gsttypefindfunctions was not found in the pkg-config search path.
Perhaps you should add the directory containing `gsttypefindfunctions.pc'
to the PKG_CONFIG_PATH environment variable
No package 'gsttypefindfunctions' found
Package gsttcp was not found in the pkg-config search path.
Perhaps you should add the directory containing `gsttcp.pc'
to the PKG_CONFIG_PATH environment variable
No package 'gsttcp' found
Package gstrawparse was not found in the pkg-config search path.
Perhaps you should add the directory containing `gstrawparse.pc'
to the PKG_CONFIG_PATH environment variable
No package 'gstrawparse' found
Package gstplayback was not found in the pkg-config search path.
Perhaps you should add the directory containing `gstplayback.pc'
to the PKG_CONFIG_PATH environment variable
No package 'gstplayback' found
Package gstpbtypes was not found in the pkg-config search path.
Perhaps you should add the directory containing `gstpbtypes.pc'
to the PKG_CONFIG_PATH environment variable
No package 'gstpbtypes' found
Package gstaudiotestsrc was not found in the pkg-config search path.
Perhaps you should add the directory containing `gstaudiotestsrc.pc'
to the PKG_CONFIG_PATH environment variable
No package 'gstaudiotestsrc' found
Package gstaudioresample was not found in the pkg-config search path.
Perhaps you should add the directory containing `gstaudioresample.pc'
to the PKG_CONFIG_PATH environment variable
No package 'gstaudioresample' found
Package gstaudioconvert was not found in the pkg-config search path.
Perhaps you should add the directory containing `gstaudioconvert.pc'
to the PKG_CONFIG_PATH environment variable
No package 'gstaudioconvert' found
Package gstapp was not found in the pkg-config search path.
Perhaps you should add the directory containing `gstapp.pc'
to the PKG_CONFIG_PATH environment variable
No package 'gstapp' found
Package onvifsoap was not found in the pkg-config search path.
Perhaps you should add the directory containing `onvifsoap.pc'
to the PKG_CONFIG_PATH environment variable
No package 'onvifsoap' found
In file included from ./src/app/onvif_app.h:5,
                 from ./src/onvif-mgr.c:3:
./src/app/../gst/player.h:4:10: fatal error: gtk/gtk.h: Datei oder Verzeichnis nicht gefunden
    4 | #include <gtk/gtk.h>
      |          ^~~~~~~~~~~
compilation terminated.
make: *** [Makefile:1283: src/onvifmgr-onvif-mgr.o] Fehler 1

Quedale, 2024-02-03:
Did you run `autogen.sh --enable-latest` before executing `make`?
The script takes care of compiling missing dependencies and it does add it to the project's path.

If you did run it, can you run it again and provide the log output?

Thanks!

kmorber, 2024-02-03:
Now i have this problem during make:

./src/app/onvif_app.c: In function ‘_onvif_device_add’:
./src/app/onvif_app.c:535:5: error: a label can only be part of a statement and a declaration is not a statement
  535 |     MsgDialog * msgdialog = OnvifApp__get_msg_dialog((OnvifApp *) event->user_data);
      |     ^~~~~~~~~
make: *** [Makefile:1381: src/app/onvifmgr-onvif_app.o] Fehler 1

Quedale, 2024-02-04:
This error seems out of context and I suspect there are other error prior to that one.

Can you run the following and provide the entire log output? (Not just the trailing part)
`make clean && make > build_debug.log 2>&1`

Can you please attach the resulting `build_debug.log` file please?

Thanks!

kmorber, 2024-02-04:
[build_debug.log](https://github.com/Quedale/OnvifDeviceManager/files/14154651/build_debug.log)

Quedale, 2024-02-04:
Okay I understand the problem, although I'm not able to reproduce it.
It seems that your version of gcc doesn't allow a label to appear immediately before a declaration, which is the case here.
I made the necessary change for this one and should work now. (Just pull the latest changes)

Considering this is related to a C syntax issue, I would like to reproduce it locally and address possibly other similar issue across the project.

Would you mind sharing the operation system and version that you are using?

Quedale, 2024-02-04:
I figured that I was able to see the error you are getting as a warning by adding the compile option -Wpedantic.
I made a global clean up of remaining warnings and it should be good to go now.

Let me know how that goes!

Quedale, 2024-03-07:
I see that you managed to build the application on Issue #17 .

Closing this issue.

closed 2024-03-07 by Quedale