Quedale/OnvifDeviceManager #16: missing plugin

kmorber, 2024-02-04:
Hi,

i have the following problem, when connecting to a camera.

WARN  24-02-04 16:59:15.504864 139786151249664 onvif_media_service.c:246: [http://192.168.1.68:2020/onvif/service] No snapshot URI found.
ERROR 24-02-04 16:59:15.504880 139786151249664 device.c:152: _priv_Device__load_thumbnail- Error retrieve snapshot.
TRACE 24-02-04 16:59:15.507802 139786151249664 device.c:236: _priv_Device__load_thumbnail done.
TRACE 24-02-04 16:59:15.507845 139786151249664 event_queue.c:173: notify dispatched event...
DEBUG 24-02-04 16:59:16.420985 139786098767616 player.c:491: Received new pad 'recv_rtp_src_0_1111122025_96' from 'rtspsrc':
TRACE 24-02-04 16:59:16.424089 139786206595776 player.c:186: State set to READY for video_bin
TRACE 24-02-04 16:59:16.424149 139786206595776 player.c:186: State set to PAUSED for onvif-pipeline
DEBUG 24-02-04 16:59:16.479164 139786206595776 player.c:308: msg : GST_MESSAGE_STREAM_COLLECTION
DEBUG 24-02-04 16:59:16.487153 139786206595776 player.c:311: msg : GST_MESSAGE_STREAMS_SELECTED
DEBUG 24-02-04 16:59:16.527606 139784796256000 player.c:491: Received new pad 'recv_rtp_src_1_328291751_8' from 'rtspsrc':
DEBUG 24-02-04 16:59:16.566874 139785034123008 overlay.c:33: Preparing level indicator overlay
TRACE 24-02-04 16:59:16.591869 139786206595776 player.c:186: State set to PAUSED for video_bin
TRACE 24-02-04 16:59:16.591903 139786206595776 player.c:186: State set to PAUSED for onvif-pipeline
TRACE 24-02-04 16:59:16.606301 139786206595776 player.c:186: State set to READY for audiobin
TRACE 24-02-04 16:59:16.606594 139786206595776 player.c:186: State set to PLAYING for video_bin
ERROR 24-02-04 16:59:16.606985 139786206595776 player.c:253: Unhandled element msg name : missing-plugin//32768
DEBUG 24-02-04 16:59:16.607115 139786206595776 player.c:308: msg : GST_MESSAGE_STREAM_COLLECTION
ERROR 24-02-04 16:59:16.607460 139786206595776 player.c:253: Unhandled element msg name : missing-plugin//32768
DEBUG 24-02-04 16:59:16.607495 139786206595776 player.c:311: msg : GST_MESSAGE_STREAMS_SELECTED

Quedale, 2024-02-04:
Good day!

This message implies that gstreamer's decodebin wasn't able to find the appropriate decoder on the system.
There's a series of reason why this may happen, like missing gstreamer package, missing dependencies, old gstreamer pluging cache, etc...

From the logs, you can see the gstreamer decoders available at the start of the application.
Here's what I have:
```
DEBUG 24-02-04 09:31:22.277850 139842943007360 onvif-mgr.c:49: *** video/x-h264 decoders ***
DEBUG 24-02-04 09:31:22.277951 139842943007360 onvif-mgr.c:57: *    openh264dec[65]
DEBUG 24-02-04 09:31:22.277960 139842943007360 onvif-mgr.c:60: ****************************
DEBUG 24-02-04 09:31:22.277966 139842943007360 onvif-mgr.c:49: *** video/x-h265 decoders ***
DEBUG 24-02-04 09:31:22.278029 139842943007360 onvif-mgr.c:57: *    libde265dec[128]
DEBUG 24-02-04 09:31:22.278037 139842943007360 onvif-mgr.c:60: ****************************
DEBUG 24-02-04 09:31:22.278042 139842943007360 onvif-mgr.c:49: *** image/jpeg decoders ***
DEBUG 24-02-04 09:31:22.278097 139842943007360 onvif-mgr.c:57: *    jpegdec[256]
DEBUG 24-02-04 09:31:22.278104 139842943007360 onvif-mgr.c:60: ****************************
DEBUG 24-02-04 09:31:22.278112 139842943007360 onvif-mgr.c:49: *** video/x-av1 decoders ***
DEBUG 24-02-04 09:31:22.278184 139842943007360 onvif-mgr.c:60: ****************************
```
</br>

Until I start distributing this software with the official distro repo, using system libraries isn't recommended.

Try adding `--enable-latest` when invoking `autogen.sh`.
This will make it download the latest gstreamer supported and compile all dependencies.

Quedale, 2024-03-07:
I see that you managed to build the application with the necessary plugins on Issue #17 .

Closing this issue.

closed 2024-03-07 by Quedale