Quedale/OnvifDeviceManager #18: I want to scan with my Onvif username and password set

nyxrobotics, 2024-02-16:
Nice to meet you, this is a great project!
Please let me request a feature about this software

I have multiple onvif cameras connected in one network, each with a different username and password.
Now when I run a scan I get ```: NOT AUTHORIZED HTTP Error code [401]``` and ```No stream uri returned...``` errors and the application terminates. 

I would like the following two features
- Set a username and password when scanning
- Devices with ```NOT AUTHORIZED HTTP Error code [401]``` errors will not request images

I am new to onvif programming, so if there is a way to implement these features, I would appreciate it if you could show me how to do it!

nyxrobotics, 2024-02-16:
Error messages may change to the following
```bash
TRACE 24-02-16 15:31:18.323976 139896516142848 probmatch.c:15: ProbMatches__create...
onvifmgr: tpp.c:82: __pthread_tpp_change_priority: Assertion `new_prio == -1 || (new_prio >= fifo_min_prio && new_prio <= fifo_max_prio)' failed.
Aborted (core dumped)
```

nyxrobotics, 2024-02-16:
After a few tries, a window came up where I had to enter my Username and Password.
When I typed them in, I got the following error
```bash
INFO  24-02-16 15:34:38.852396 140114827134720 player.c:709: set location : rtsp://192.168.128.106:554/cam/realmonitor?channel=1&subtype=0&unicast=true&proto=Onvif
DEBUG 24-02-16 15:34:38.852425 140114801956608 onvif_media_service.c:147: [http://192.168.128.106/onvif/media_service] Invoking GetSnapshotUri
TRACE 24-02-16 15:34:38.852471 140114801956608 onvif_base_service.c:100: [http://192.168.128.106/onvif/media_service] Time adjustment [2024-02-16 06:34:38] to [1999-12-31 17:24:57] offset -761404181
Error: signal 11:
./onvifmgr(+0x596a0)[0x55b40766e6a0]
/lib/x86_64-linux-gnu/libc.so.6(+0x43090)[0x7f6f0f43f090]
/lib/x86_64-linux-gnu/libc.so.6(+0x188915)[0x7f6f0f584915]
./onvifmgr(+0x6a385)[0x55b40767f385]
./onvifmgr(+0x6042e)[0x55b40767542e]
./onvifmgr(+0x60605)[0x55b407675605]
./onvifmgr(+0x606d0)[0x55b4076756d0]
./onvifmgr(+0x6f665)[0x55b407684665]
/lib/x86_64-linux-gnu/libpthread.so.0(+0x8609)[0x7f6f0f5f6609]
/lib/x86_64-linux-gnu/libc.so.6(clone+0x43)[0x7f6f0f51b353]
```

Quedale, 2024-02-16:
Good day and thank you for testing this project.

This application is designed to show secure camera without prompts, so that you get a populated device list without interruption. If 12 cameras are detected, I don't want to bother the user with 12 different credentials prompt.
From the device list, selecting a protected device will prompt for credentials.

I can see that you do have a number of camera discovered.
I have a hard time simulating such environment since I simply don't have the hardware and mocked cameras don't behave the same.

Would you mind rebuilding the application with debugging enabled so that I can have more relevant details?

Clearing existing build and configuration:
```
make distclean
```
Clearing dependencies to rebuild
```
rm -rf ./subprojects/OnvifDiscoveryLib
rm -rf ./subprojects/OnvifSoapLib
```

Reconfigure and rebuild with debug symbols enabled
```
./autogen.sh --enable-latest --full-debug
make
```

Once the new build is ready, run it with valgrind. This will extract additional details about the crash.
```
valgrind ./onvifmgr > onvifmgr_debug.log 2>&1
```

This will generate a log output `onvifmgr_debug.log `.
If you could share that file that would be extremely helpful.

There's shouldn't be any sensitive information printed, but feel free to review its content.

Sorry for the trouble and thank you very much for helping out!

nyxrobotics, 2024-03-01:
Sorry for the late reply, I've been busy at work.
I used d2d599598bee4ee30412bd85fb396e19e726a9fa and configure it with
`./autogen.sh --prefix=$(pwd)/dist --enable-latest  --full-debug`, and ran onvifmgr generated by make command.
The login screen appears, username and password are entered, and then it drops out
I'll look into this some more.
```bash
INFO  24-03-01 11:43:40.138004 140327917539136 onvif_app.c:379: OnvifAuthentication attempt...
TRACE 24-03-01 11:43:40.138077 140327746774784 event_queue.c:145: pop
TRACE 24-03-01 11:43:40.138105 140327746774784 event_queue.c:165: notify dispatching event...
DEBUG 24-03-01 11:43:40.138139 140327746774784 onvif_app.c:364: _onvif_authentication_reload
INFO  24-03-01 11:43:40.138148 140327746774784 onvif_device.c:73: [http://192.168.128.106/onvif/device_service] OnvifDevice__authenticate
INFO  24-03-01 11:43:40.138164 140327746774784 onvif_device.c:93: [http://192.168.128.106/onvif/device_service] Camera SystemDateAndTime : '2024-03-01 11:43:42'
TRACE 24-03-01 11:43:40.138171 140327746774784 onvif_device.c:46: [http://192.168.128.106/onvif/device_service] OnvifDevice__createMediaService
DEBUG 24-03-01 11:43:40.138178 140327746774784 onvif_device_service.c:99: [http://192.168.128.106/onvif/device_service] Invoking GetCapabilities
TRACE 24-03-01 11:43:40.138259 140327746774784 onvif_base_service.c:100: [http://192.168.128.106/onvif/device_service] Time adjustment [2024-03-01 02:43:40] to [2024-03-01 02:43:54] offset 14
DEBUG 24-03-01 11:43:40.152873 140327746774784 onvif_device.c:104: [http://192.168.128.106/onvif/media_service] Successfully created Media soap 
TRACE 24-03-01 11:43:40.152889 140327746774784 onvif_media_service.c:267: [http://192.168.128.106/onvif/media_service] OnvifMediaService__get_profile_token 0
DEBUG 24-03-01 11:43:40.152896 140327746774784 onvif_media_service.c:79: [http://192.168.128.106/onvif/media_service] Invoking GetProfiles
TRACE 24-03-01 11:43:40.152936 140327746774784 onvif_base_service.c:100: [http://192.168.128.106/onvif/media_service] Time adjustment [2024-03-01 02:43:40] to [2024-03-01 02:43:54] offset 14
TRACE 24-03-01 11:43:40.196624 140327746774784 onvif_media_service.c:280: [http://192.168.128.106/onvif/media_service] OnvifMediaService__get_profile_token : Found profile [MediaProfile00000]
DEBUG 24-03-01 11:43:40.196641 140327746774784 onvif_media_service.c:118: [http://192.168.128.106/onvif/media_service] Invoking GetStreamUri
TRACE 24-03-01 11:43:40.196682 140327746774784 onvif_base_service.c:100: [http://192.168.128.106/onvif/media_service] Time adjustment [2024-03-01 02:43:40] to [2024-03-01 02:43:54] offset 14
DEBUG 24-03-01 11:43:40.218395 140327746774784 onvif_device.c:112: [http://192.168.128.106/onvif/media_service] StreamURI : rtsp://192.168.128.106:554/cam/realmonitor?channel=1&subtype=0&unicast=true&proto=Onvif
TRACE 24-03-01 11:43:40.218462 140327746774784 onvif_app.c:261: _play_onvif_stream http://192.168.128.106/onvif/device_service
INFO  24-03-01 11:43:40.218470 140327746774784 onvif_device.c:73: [http://192.168.128.106/onvif/device_service] OnvifDevice__authenticate
DEBUG 24-03-01 11:43:40.218477 140327746774784 omgr_device_row.c:404: OnvifMgrDeviceRow__get_profile
WARN  24-03-01 11:43:40.218482 140327746774784 onvif_media_profile.c:100: Failed to get profile index. NULL pointer. Defaulting to 0.
TRACE 24-03-01 11:43:40.218490 140327746774784 onvif_media_service.c:267: [http://192.168.128.106/onvif/media_service] OnvifMediaService__get_profile_token 0
TRACE 24-03-01 11:43:40.218498 140327746774784 onvif_media_service.c:280: [http://192.168.128.106/onvif/media_service] OnvifMediaService__get_profile_token : Found profile [MediaProfile00000]
DEBUG 24-03-01 11:43:40.218507 140327746774784 onvif_media_service.c:118: [http://192.168.128.106/onvif/media_service] Invoking GetStreamUri
TRACE 24-03-01 11:43:40.218522 140327755167488 event_queue.c:145: pop
TRACE 24-03-01 11:43:40.218550 140327746774784 onvif_base_service.c:100: [http://192.168.128.106/onvif/media_service] Time adjustment [2024-03-01 02:43:40] to [2024-03-01 02:43:54] offset 14
TRACE 24-03-01 11:43:40.218565 140327755167488 event_queue.c:165: notify dispatching event...
TRACE 24-03-01 11:43:40.218587 140327755167488 onvif_app.c:216: _display_onvif_device
INFO  24-03-01 11:43:40.218596 140327755167488 onvif_device.c:73: [http://192.168.128.106/onvif/device_service] OnvifDevice__authenticate
DEBUG 24-03-01 11:43:40.218604 140327755167488 omgr_device_row.c:404: OnvifMgrDeviceRow__get_profile
TRACE 24-03-01 11:43:40.218624 140327755167488 omgr_device_row.c:418: OnvifMgrDeviceRow__load_thumbnail
DEBUG 24-03-01 11:43:40.218630 140327755167488 onvif_media_service.c:242: [http://192.168.128.106/onvif/media_service] OnvifMediaService__getSnapshot
TRACE 24-03-01 11:43:40.218637 140327755167488 onvif_media_service.c:267: [http://192.168.128.106/onvif/media_service] OnvifMediaService__get_profile_token 0
TRACE 24-03-01 11:43:40.218644 140327755167488 onvif_media_service.c:280: [http://192.168.128.106/onvif/media_service] OnvifMediaService__get_profile_token : Found profile [MediaProfile00000]
INFO  24-03-01 11:43:40.233612 140327746774784 gstrtspplayer.c:851: set location : rtsp://192.168.128.106:554/cam/realmonitor?channel=1&subtype=0&unicast=true&proto=Onvif
DEBUG 24-03-01 11:43:40.233639 140327755167488 onvif_media_service.c:147: [http://192.168.128.106/onvif/media_service] Invoking GetSnapshotUri
Error: signal 11:
./onvifmgr(+0x5cdb0)[0x55c6c4e82db0]
[... cut ...] ime adjustment [2024-03-01 02:43:40] to [2024-03-01 02:43:54] offset 14
/lib/x86_64-linux-gnu/libc.so.6(+0x43090)[0x7fa0a4d6b090]
/lib/x86_64-linux-gnu/libc.so.6(+0x188915)[0x7fa0a4eb0915]
./onvifmgr(+0x733bd)[0x55c6c4e993bd]
./onvifmgr(+0x63bce)[0x55c6c4e89bce]
./onvifmgr(+0x63da5)[0x55c6c4e89da5]
./onvifmgr(+0x63e70)[0x55c6c4e89e70]
./onvifmgr(+0x73ad5)[0x55c6c4e99ad5]
/lib/x86_64-linux-gnu/libpthread.so.0(+0x8609)[0x7fa0a4f22609]
/lib/x86_64-linux-gnu/libc.so.6(clone+0x43)[0x7fa0a4e47353]
```

nyxrobotics, 2024-03-01:
I ran it after synchronizing the time and it was almost the same.
```
WARN  24-03-01 12:03:40.576377 140154211645184 onvif_media_profile.c:100: Failed to get profile index. NULL pointer. Defaulting to 0.
TRACE 24-03-01 12:03:40.576384 140154228430592 event_queue.c:145: pop
TRACE 24-03-01 12:03:40.576404 140154228430592 event_queue.c:165: notify dispatching event...
TRACE 24-03-01 12:03:40.576416 140154211645184 onvif_media_service.c:267: [http://192.168.128.106/onvif/media_service] OnvifMediaService__get_profile_token 0
TRACE 24-03-01 12:03:40.576426 140154211645184 onvif_media_service.c:280: [http://192.168.128.106/onvif/media_service] OnvifMediaService__get_profile_token : Found profile [MediaProfile00000]
DEBUG 24-03-01 12:03:40.576432 140154211645184 onvif_media_service.c:118: [http://192.168.128.106/onvif/media_service] Invoking GetStreamUri
TRACE 24-03-01 12:03:40.576457 140154228430592 onvif_app.c:216: _display_onvif_device
INFO  24-03-01 12:03:40.576477 140154228430592 onvif_device.c:73: [http://192.168.128.106/onvif/device_service] OnvifDevice__authenticate
DEBUG 24-03-01 12:03:40.576486 140154228430592 omgr_device_row.c:404: OnvifMgrDeviceRow__get_profile
TRACE 24-03-01 12:03:40.576506 140154228430592 omgr_device_row.c:418: OnvifMgrDeviceRow__load_thumbnail
DEBUG 24-03-01 12:03:40.576514 140154228430592 onvif_media_service.c:242: [http://192.168.128.106/onvif/media_service] OnvifMediaService__getSnapshot
TRACE 24-03-01 12:03:40.576521 140154228430592 onvif_media_service.c:267: [http://192.168.128.106/onvif/media_service] OnvifMediaService__get_profile_token 0
TRACE 24-03-01 12:03:40.576529 140154228430592 onvif_media_service.c:280: [http://192.168.128.106/onvif/media_service] OnvifMediaService__get_profile_token : Found profile [MediaProfile00000]
INFO  24-03-01 12:03:40.597694 140154211645184 gstrtspplayer.c:851: set location : rtsp://192.168.128.106:554/cam/realmonitor?channel=1&subtype=0&unicast=true&proto=Onvif
DEBUG 24-03-01 12:03:40.597726 140154228430592 onvif_media_service.c:147: [http://192.168.128.106/onvif/media_service] Invoking GetSnapshotUri
Error: signal 11:
./onvifmgr(+0x5cdb0)[0x56109d988db0]
/lib/x86_64-linux-gnu/libc.so.6(+0x43090)[0x7f783cf93090]
/lib/x86_64-linux-gnu/libc.so.6(+0x188915)[0x7f783d0d8915]
./onvifmgr(+0x733bd)[0x56109d99f3bd]
./onvifmgr(+0x63bce)[0x56109d98fbce]
./onvifmgr(+0x63da5)[0x56109d98fda5]
./onvifmgr(+0x63e70)[0x56109d98fe70]
./onvifmgr(+0x73ad5)[0x56109d99fad5]
/lib/x86_64-linux-gnu/libpthread.so.0(+0x8609)[0x7f783d14a609]
/lib/x86_64-linux-gnu/libc.so.6(clone+0x43)[0x7f783d06f353]
```

nyxrobotics, 2024-03-01:
https://github.com/Quedale/OnvifDeviceManager/blob/d2d599598bee4ee30412bd85fb396e19e726a9fa/src/app/omgr_device_row.c#L405

I looked for more information about warning.
```bash
WARN  24-03-01 12:03:40.576377 140154211645184 onvif_media_profile.c:100: Failed to get profile index. NULL pointer. Defaulting to 0.
```
The pointer to the profile in this part seems to be null.

nyxrobotics, 2024-03-01:
Sorry for confusing the issue.
I turned off the authenticator but still had the same problem.
The problem may be dependent on the Dahua camera.
Please give me some more time to investigate.

Quedale, 2024-03-01:
Thank you for diving into it!

I may have made a mistake in the authentication sequence. 
Compared to the official ONVIF Device Manager, it actually attempts anonymous authentication to figure out if authentication is needed at all.
This could potentially lead to unexpected result.

This gave me the idea to add an application setting to prevent this behaviour.
A simple checkbox that will enforce credential capture for each camera.
This setting would be especially useful when the camera has a lockout mechanism.

That being said, it doesn't explain the crashes you are getting.

Can you try to run  the application using valgrind?
```
valgrind ./onvifmgr
```

Combined with the configure flag "--full-debug", it should give relevant details about the crash.

Thank you very much for your help!

Quedale, 2024-03-08:
I fixed a crash happening while snapshot were loaded.

It hopefully fixed your issue.