Quedale/OnvifDeviceManager #19: ip camera Not logging in OnvifDeviceManager

onyx82, 2024-03-01:
I have linux mint 21.3 and imou bullet lite 4mp camera.
I run the command ./onvifmgr to open the software,i press the search button C...I try to log in, but it doesn't log in, here's what it says in the terminal after I try to log in.
I don't know where the error log is

```

TRAIL 24-03-01 03:43:37.192632 140193375029888 onvif_details.c:33: update_details_priv - invalid device.
INFO  24-03-01 03:44:10.335724 140193375029888 onvif_app.c:144: Starting ONVIF Devices Network Discovery...
TRACE 24-03-01 03:44:10.337934 140193313453632 event_queue.c:145: pop
TRACE 24-03-01 03:44:10.337984 140193313453632 event_queue.c:165: notify dispatching event...
TRACE 24-03-01 03:44:10.338078 140193313453632 onvif_app.c:134: _start_onvif_discovery
DEBUG 24-03-01 03:44:10.342386 140193313453632 onvif_discovery.c:208: Sending NVT probe...
DEBUG 24-03-01 03:44:10.458029 140193313453632 onvif_discovery.c:64: ProbeMatches found...
TRACE 24-03-01 03:44:10.458089 140193313453632 probmatch.c:15: ProbMatches__create...
TRACE 24-03-01 03:44:10.458116 140193313453632 probmatch.c:84: ProbMatch__create...
TRACE 24-03-01 03:44:10.458184 140193313453632 onvif_app.c:118: _found_server
TRACE 24-03-01 03:44:10.458326 140193375029888 onvif_app.c:84: found_server
INFO  24-03-01 03:44:10.458355 140193375029888 onvif_app.c:93: Found server - Match count : 1
DEBUG 24-03-01 03:44:10.458388 140193375029888 onvif_device.c:184: OnvifDevice__create
INFO  24-03-01 03:44:10.458431 140193375029888 onvif_device.c:176: Created Device:
INFO  24-03-01 03:44:10.458450 140193375029888 onvif_device.c:177: 	protocol -- http
INFO  24-03-01 03:44:10.458468 140193375029888 onvif_device.c:178: 	host : 192.168.100.94
INFO  24-03-01 03:44:10.458485 140193375029888 onvif_device.c:179: 	port -- (null)
INFO  24-03-01 03:44:10.458506 140193375029888 onvif_device.c:180: 	endpoint : onvif/device_service
TRACE 24-03-01 03:44:10.482011 140193375029888 probmatch.c:22: ProbMatches__destroy...
TRACE 24-03-01 03:44:10.482078 140193375029888 probmatch.c:48: ProbMatch__destroy...
TRACE 24-03-01 03:44:10.482108 140193375029888 probmatch.c:68: ProbMatch__destroy... done
TRACE 24-03-01 03:44:10.482196 140193218537024 event_queue.c:145: pop
TRACE 24-03-01 03:44:10.482228 140193218537024 event_queue.c:165: notify dispatching event...
TRACE 24-03-01 03:44:10.482313 140193218537024 onvif_app.c:211: _display_onvif_device
INFO  24-03-01 03:44:10.482347 140193218537024 onvif_device.c:73: [http://192.168.100.94/onvif/device_service] OnvifDevice__authenticate
DEBUG 24-03-01 03:44:10.482386 140193218537024 onvif_device_service.c:82: [http://192.168.100.94/onvif/device_service] Invoking GetSystemDateAndTime
INFO  24-03-01 03:44:10.506492 140193218537024 onvif_device.c:93: [http://192.168.100.94/onvif/device_service] Camera SystemDateAndTime : '2024-03-01 03:44:10'
TRACE 24-03-01 03:44:10.506527 140193218537024 onvif_device.c:46: [http://192.168.100.94/onvif/device_service] OnvifDevice__createMediaService
DEBUG 24-03-01 03:44:10.506544 140193218537024 onvif_device_service.c:99: [http://192.168.100.94/onvif/device_service] Invoking GetCapabilities
DEBUG 24-03-01 03:44:10.543594 140193218537024 onvif_device.c:104: [http://192.168.100.94/onvif/media_service] Successfully created Media soap 
TRACE 24-03-01 03:44:10.543633 140193218537024 onvif_media_service.c:267: [http://192.168.100.94/onvif/media_service] OnvifMediaService__get_profile_token 0
DEBUG 24-03-01 03:44:10.543652 140193218537024 onvif_media_service.c:79: [http://192.168.100.94/onvif/media_service] Invoking GetProfiles
WARN  24-03-01 03:44:10.563702 140193218537024 onvif_base_service.c:200: Warning : NOT AUTHORIZED HTTP Error code [401]
DEBUG 24-03-01 03:44:10.563779 140193218537024 onvif_media_service.c:118: [http://192.168.100.94/onvif/media_service] Invoking GetStreamUri
WARN  24-03-01 03:44:10.582640 140193218537024 onvif_base_service.c:200: Warning : NOT AUTHORIZED HTTP Error code [401]
ERROR 24-03-01 03:44:10.582903 140193218537024 onvif_device.c:110: [http://192.168.100.94/onvif/media_service] No stream uri returned...
TRACE 24-03-01 03:44:10.582940 140193218537024 omgr_device_row.c:418: OnvifMgrDeviceRow__load_thumbnail
TRACE 24-03-01 03:44:10.587808 140193218537024 omgr_device_row.c:490: OnvifMgrDeviceRow__load_thumbnail done.
TRACE 24-03-01 03:44:10.587860 140193218537024 event_queue.c:173: notify dispatched event...
DEBUG 24-03-01 03:44:12.460929 140193313453632 onvif_discovery.c:208: Sending NVT probe...
DEBUG 24-03-01 03:44:12.747459 140193313453632 onvif_discovery.c:64: ProbeMatches found...
TRACE 24-03-01 03:44:12.747525 140193313453632 probmatch.c:15: ProbMatches__create...
TRACE 24-03-01 03:44:12.747559 140193313453632 probmatch.c:84: ProbMatch__create...
TRACE 24-03-01 03:44:12.747609 140193313453632 onvif_app.c:118: _found_server
TRACE 24-03-01 03:44:12.747762 140193375029888 onvif_app.c:84: found_server
INFO  24-03-01 03:44:12.747804 140193375029888 onvif_app.c:93: Found server - Match count : 1
DEBUG 24-03-01 03:44:12.747857 140193375029888 onvif_app.c:73: Record already part of list [http://192.168.100.94/onvif/device_service]
TRACE 24-03-01 03:44:12.747901 140193375029888 probmatch.c:22: ProbMatches__destroy...
TRACE 24-03-01 03:44:12.747933 140193375029888 probmatch.c:48: ProbMatch__destroy...
TRACE 24-03-01 03:44:12.747970 140193375029888 probmatch.c:68: ProbMatch__destroy... done
TRACE 24-03-01 03:44:14.750163 140193313453632 event_queue.c:173: notify dispatched event...
TRACE 24-03-01 03:44:14.750244 140193375029888 onvif_app.c:123: finished_discovery
DEBUG 24-03-01 03:44:31.917811 140193375029888 onvif_app.c:392: OnvifApp__select_device
TRACE 24-03-01 03:44:31.917854 140193375029888 onvif_details.c:108: clear_details
DEBUG 24-03-01 03:44:31.917895 140193375029888 onvif_app.c:902: OnvifApp__set_device - valid device
TRACE 24-03-01 03:44:31.917923 140193201751616 event_queue.c:145: pop
TRACE 24-03-01 03:44:31.917938 140193201751616 event_queue.c:165: notify dispatching event...
TRACE 24-03-01 03:44:31.917975 140193201751616 onvif_app.c:200: _stop_onvif_stream
DEBUG 24-03-01 03:44:31.917990 140193201751616 gstrtspplayer.c:526: RtspPlayer__stop
INFO  24-03-01 03:44:31.918296 140193201751616 onvif_app.c:165: Stream stopped
TRACE 24-03-01 03:44:31.918310 140193201751616 event_queue.c:173: notify dispatched event...
DEBUG 24-03-01 03:44:31.922940 140193375029888 app_dialog.c:98: show
INFO  24-03-01 03:50:01.141573 140193375029888 onvif_app.c:374: OnvifAuthentication attempt...
TRACE 24-03-01 03:50:01.141732 140192967030336 event_queue.c:145: pop
TRACE 24-03-01 03:50:01.141979 140192967030336 event_queue.c:165: notify dispatching event...
DEBUG 24-03-01 03:50:01.142097 140192967030336 onvif_app.c:359: _onvif_authentication_reload
INFO  24-03-01 03:50:01.142142 140192967030336 onvif_device.c:73: [http://192.168.100.94/onvif/device_service] OnvifDevice__authenticate
INFO  24-03-01 03:50:01.142226 140192967030336 onvif_device.c:93: [http://192.168.100.94/onvif/device_service] Camera SystemDateAndTime : '2024-03-01 03:44:10'
TRACE 24-03-01 03:50:01.142268 140192967030336 onvif_device.c:46: [http://192.168.100.94/onvif/device_service] OnvifDevice__createMediaService
DEBUG 24-03-01 03:50:01.142307 140192967030336 onvif_device_service.c:99: [http://192.168.100.94/onvif/device_service] Invoking GetCapabilities
DEBUG 24-03-01 03:50:01.183025 140192967030336 onvif_device.c:104: [http://192.168.100.94/onvif/media_service] Successfully created Media soap 
TRACE 24-03-01 03:50:01.183093 140192967030336 onvif_media_service.c:267: [http://192.168.100.94/onvif/media_service] OnvifMediaService__get_profile_token 0
DEBUG 24-03-01 03:50:01.183139 140192967030336 onvif_media_service.c:79: [http://192.168.100.94/onvif/media_service] Invoking GetProfiles
WARN  24-03-01 03:50:01.205261 140192967030336 onvif_base_service.c:200: Warning : NOT AUTHORIZED HTTP Error code [12]
DEBUG 24-03-01 03:50:01.205410 140192967030336 onvif_media_service.c:118: [http://192.168.100.94/onvif/media_service] Invoking GetStreamUri
WARN  24-03-01 03:50:01.227830 140192967030336 onvif_base_service.c:200: Warning : NOT AUTHORIZED HTTP Error code [12]
ERROR 24-03-01 03:50:01.227950 140192967030336 onvif_device.c:110: [http://192.168.100.94/onvif/media_service] No stream uri returned...
TRACE 24-03-01 03:50:01.227993 140192967030336 event_queue.c:173: notify dispatched event...

```

Quedale, 2024-03-01:
Good day!

Thank you for taking the time to try out my tool and provide feedback!

According to the logs, it seems like you aren't authorized to retrieve the device media profiles or retrieve the stream URI.
```
DEBUG 24-03-01 03:50:01.183139 140192967030336 onvif_media_service.c:79: [http://192.168.100.94/onvif/media_service] Invoking GetProfiles
WARN  24-03-01 03:50:01.205261 140192967030336 onvif_base_service.c:200: Warning : NOT AUTHORIZED HTTP Error code [12]
DEBUG 24-03-01 03:50:01.205410 140192967030336 onvif_media_service.c:118: [http://192.168.100.94/onvif/media_service] Invoking GetStreamUri
WARN  24-03-01 03:50:01.227830 140192967030336 onvif_base_service.c:200: Warning : NOT AUTHORIZED HTTP Error code [12]
ERROR 24-03-01 03:50:01.227950 140192967030336 onvif_device.c:110: [http://192.168.100.94/onvif/media_service] No stream uri returned...
```

Can you run the application with the environment ONVIF_DEBUG set to 1?
```
ONVIF_DEBUG=1 ./onvifmgr
```

This will cause the gsoap library to dump soap messages so that we can look at the raw response from the camera.

onyx82, 2024-03-01:
alexandru@alexandru-desktop:~/Documente/onvif/OnvifDeviceManager$ ONVIF_DEBUG=1 ./onvifmgr
INFO  24-03-01 22:49:03.108220 140443452058240 onvif-mgr.c:103: Using Gstreamer Version : 1.22.9.0
INFO  24-03-01 22:49:03.108319 140443452058240 onvifinitstaticplugins.c:223: Initializing Gstreamer plugins...
INFO  24-03-01 22:49:03.110165 140443452058240 onvifinitstaticplugins.c:228: Loading static plugins...
DEBUG 24-03-01 22:49:03.161023 140443452058240 onvifinitstaticplugins.c:435: Gstreamer plugins initialized...
WARN  24-03-01 22:49:03.161057 140443452058240 onvif-mgr.c:73: Element 'avdec_h264' not found.
WARN  24-03-01 22:49:03.161069 140443452058240 onvif-mgr.c:73: Element 'avdec_h265' not found.
WARN  24-03-01 22:49:03.161083 140443452058240 onvif-mgr.c:81: Found element 'openh264dec'. Increasing priority from 64 to 65...
INFO  24-03-01 22:49:03.161099 140443452058240 onvif-mgr.c:48: *** video/x-h264 decoders ***
INFO  24-03-01 22:49:03.161270 140443452058240 onvif-mgr.c:56: *    openh264dec[65]
INFO  24-03-01 22:49:03.161285 140443452058240 onvif-mgr.c:61: ****************************
INFO  24-03-01 22:49:03.161296 140443452058240 onvif-mgr.c:48: *** video/x-h265 decoders ***
INFO  24-03-01 22:49:03.161354 140443452058240 onvif-mgr.c:56: *    libde265dec[128]
INFO  24-03-01 22:49:03.161368 140443452058240 onvif-mgr.c:61: ****************************
INFO  24-03-01 22:49:03.161378 140443452058240 onvif-mgr.c:48: *** image/jpeg decoders ***
INFO  24-03-01 22:49:03.161436 140443452058240 onvif-mgr.c:56: *    jpegdec[256]
INFO  24-03-01 22:49:03.161449 140443452058240 onvif-mgr.c:61: ****************************
INFO  24-03-01 22:49:03.161460 140443452058240 onvif-mgr.c:48: *** video/x-av1 decoders ***
INFO  24-03-01 22:49:03.161518 140443452058240 onvif-mgr.c:61: ****************************
TRACE 24-03-01 22:49:03.161531 140443452058240 event_queue.c:83: create...
INFO  24-03-01 22:49:03.203333 140443452058240 app_settings.c:188: Reading Settings file onvifmgr_settings.ini
WARN  24-03-01 22:49:03.203416 140443452058240 app_settings.c:245: No config file found. Using default configs. 1
DEBUG 24-03-01 22:49:03.206684 140443452058240 src_retriever.c:12: Test Audio Src [pulsesrc] device [(null)]
INFO  24-03-01 22:49:03.404585 140443452058240 backchannel.c:111: Creating backchannel using source element pulsesrc
DEBUG 24-03-01 22:49:03.405195 140443390338624 queue_thread.c:25: Started...
DEBUG 24-03-01 22:49:03.405246 140443372115520 queue_thread.c:25: Started...
DEBUG 24-03-01 22:49:03.405304 140443363722816 queue_thread.c:25: Started...
DEBUG 24-03-01 22:49:03.405376 140443148875328 queue_thread.c:25: Started...
DEBUG 24-03-01 22:49:03.405435 140443140482624 queue_thread.c:25: Started...
DEBUG 24-03-01 22:49:03.405484 140443132089920 queue_thread.c:25: Started...
DEBUG 24-03-01 22:49:03.405557 140443123697216 queue_thread.c:25: Started...
TRACE 24-03-01 22:49:03.405593 140443452058240 profiles_dialog.c:119: create
TRACE 24-03-01 22:49:03.405607 140443452058240 app_dialog.c:69: init
DEBUG 24-03-01 22:49:03.405621 140443115304512 queue_thread.c:25: Started...
TRACE 24-03-01 22:49:03.406337 140443452058240 profiles_dialog.c:25: priv_ProfilesDialog__create_ui
TRACE 24-03-01 22:49:03.406651 140443452058240 app_dialog.c:69: init
TRACE 24-03-01 22:49:03.409274 140443452058240 credentials_input.c:28: create
TRACE 24-03-01 22:49:03.409394 140443452058240 app_dialog.c:69: init
TRACE 24-03-01 22:49:03.410998 140443452058240 msg_dialog.c:60: create
TRACE 24-03-01 22:49:03.411114 140443452058240 app_dialog.c:69: init
TRACE 24-03-01 22:49:03.425957 140443452058240 onvif_details.c:108: clear_details
TRAIL 24-03-01 22:49:03.426033 140443452058240 onvif_details.c:33: update_details_priv - invalid device.
INFO  24-03-01 22:49:27.689026 140443452058240 onvif_app.c:144: Starting ONVIF Devices Network Discovery...
TRACE 24-03-01 22:49:27.690090 140443390338624 event_queue.c:145: pop
TRACE 24-03-01 22:49:27.690107 140443390338624 event_queue.c:165: notify dispatching event...
TRACE 24-03-01 22:49:27.690149 140443390338624 onvif_app.c:134: _start_onvif_discovery
DEBUG 24-03-01 22:49:27.696813 140443390338624 onvif_discovery.c:208: Sending NVT probe...
DEBUG 24-03-01 22:49:29.700689 140443390338624 onvif_discovery.c:208: Sending NVT probe...
TRACE 24-03-01 22:49:31.703064 140443390338624 event_queue.c:173: notify dispatched event...
TRACE 24-03-01 22:49:31.703110 140443452058240 onvif_app.c:123: finished_discovery
INFO  24-03-01 22:49:36.048980 140443452058240 onvif_app.c:144: Starting ONVIF Devices Network Discovery...
TRACE 24-03-01 22:49:36.050094 140443372115520 event_queue.c:145: pop
TRACE 24-03-01 22:49:36.050111 140443372115520 event_queue.c:165: notify dispatching event...
TRACE 24-03-01 22:49:36.050151 140443372115520 onvif_app.c:134: _start_onvif_discovery
DEBUG 24-03-01 22:49:36.050480 140443372115520 onvif_discovery.c:208: Sending NVT probe...
DEBUG 24-03-01 22:49:38.052617 140443372115520 onvif_discovery.c:208: Sending NVT probe...
DEBUG 24-03-01 22:49:38.369128 140443372115520 onvif_discovery.c:64: ProbeMatches found...
TRACE 24-03-01 22:49:38.369166 140443372115520 probmatch.c:15: ProbMatches__create...
TRACE 24-03-01 22:49:38.369177 140443372115520 probmatch.c:84: ProbMatch__create...
TRACE 24-03-01 22:49:38.369194 140443372115520 onvif_app.c:118: _found_server
TRACE 24-03-01 22:49:38.369253 140443452058240 onvif_app.c:84: found_server
[... cut ...]
INFO  24-03-01 22:50:26.077963 140443132089920 onvif_base_service.c:158: ONVIF_DEBUG variable set. '1'
ERROR 24-03-01 22:50:26.078006 140443132089920 onvif_device.c:110: [http://192.168.100.94/onvif/media_service] No stream uri returned...
TRACE 24-03-01 22:50:26.078024 140443132089920 event_queue.c:173: notify dispatched event...
INFO  24-03-01 22:50:28.480545 140443452058240 onvif_app.c:381: OnvifAuthentication cancelled...
DEBUG 24-03-01 22:50:28.480583 140443452058240 app_dialog.c:131: hide
^C
alexandru@alexandru-desktop:~/Documente/onvif/OnvifDeviceManager$

Quedale, 2024-03-02:
Good news,

It seems to be a relatively obvious issue.
According to the following log you provided, your camera is expecting Digest authentication.
```
HTTP/1.1 401 Unauthorized
Content-Type: application/soap+xml; charset=utf-8
Connection: close
WWW-Authenticate: Digest realm="Login to 0ccf58c7410332c7caa363406561b351",qop="auth",nonce="b252aWYtZGlnZXN0OjQzNTY3Mzg5NDgw", opaque="", stale="false"
Content-Length: 0
```

It's simply not supported at the moment, but easily within reach.
I original focused on ONVIF specs only, which recommends WS-Security.

That being said, it shouldn't be hard to support alternative authentication schemes like Digest or NTLM.

I do want to support as many camera as possible.

I'll start working on it soon.

Quedale, 2024-03-04:
I just added support for HTTP authentication challenge.
It should simultaneously support Basic, Digest and NTLM authentication scheme.

Simply pull the latest changes and run `autogen.sh` once again to pull and build the latest changes from [OnvifSoapLib](https://github.com/Quedale/OnvifSoapLib) as well.

I was unfortunately only able to test Basic authentication without given the effort required to setup an appropriate test environment.

Let me know how that goes.

Thank for your help!

Quedale, 2024-10-30:
I think you this issue is now resolved considering you comment on #30.
I'm going to close this one and continue with the new issue.

Feel free to reopen it if I'm mistaken.

Cheers!

closed 2024-10-30 by Quedale