Quedale/OnvifDeviceManager #20: Received errno error when MAKEing...

mrinvader, 2024-03-07:
Hi All...

I received 
```

./src/app/settings/app_settings.c: In function ‘AppSettings__get_config_path’:
./src/app/settings/app_settings.c:77:17: error: ‘errno’ undeclared (first use in this function)
   77 |                 errno = s;
      |                 ^~~~~
./src/app/settings/app_settings.c:5:1: note: ‘errno’ is defined in header ‘<errno.h>’; did you forget to ‘#include <errno.h>’?
    4 | #include <pwd.h>
  +++ |+#include <errno.h>
    5 | 
./src/app/settings/app_settings.c:77:17: note: each undeclared identifier is reported only once for each function it appears in
   77 |                 errno = s;
      |                 ^~~~~
make: *** [Makefile:1892: src/app/settings/onvifmgr-app_settings.o] Error 1
make: *** Waiting for unfinished jobs....

```
during the final make

(https://techoverflow.net/2019/06/20/how-to-fix-c-error-errno-undeclared/)
gave me the answer...

#when you try to compile it you see an error message like

fix-c-error-errno-undeclared.txt📋 Copy to clipboard⇓ Download
main.c: In function ‘main’:
main.c:4:5: error: ‘errno’ undeclared (first use in this function)
     errno = EFAULT;
     ^~~~~
main.c:4:5: note: each undeclared identifier is reported only once for each function it appears in
main.c:4:13: error: ‘EFAULT’ undeclared (first use in this function)
     errno = EFAULT;
Solution:
Add

fix-c-error-errno-undeclared.c📋 Copy to clipboard⇓ Download
#include <errno.h>
at the top of the source file where the error occured. This will include both the errno variable and specific error codes like EFAULT.

### solution
 vi ./src/app/settings/app_settings.c
add 
#include <errno.h>
to top of file in the includes section

rerun make and it built successfully!

Quedale, 2024-03-07:
Good day and thank you for this feedback.

I'm not sure why this include is optional on all my environment, but you are correct.
I just added the missing include.

Feel free to submit a pull request if you find additional faults.

Thanks for this contribution!

mrinvader, 2024-03-07:
it ended up erroring out after I ran it. Ill file a new report.

Quedale, 2024-03-07:
Thanks!

Sorry, I assumed that would be it since you provided the solution.

Looking forward to hear back!