Quedale/OnvifDeviceManager #22: binary ./onvifmgr crashed after build was apparently successful

mrinvader, 2024-03-07:
Running the app crashes... crash output at bottom

** distro info from lsb-release 

DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=20.04
DISTRIB_CODENAME=focal
DISTRIB_DESCRIPTION="Ubuntu 20.04.2 LTS"

**configure output exceeded my scrollback...

see config.log attached
[config.log](https://github.com/Quedale/OnvifDeviceManager/files/14529783/config.log)

the last few lines as follows
```

configure: creating ./config.status
config.status: creating Makefile
config.status: executing libtool commands
config.status: executing depfiles commands
bridgeadmin@moodjacksonbridge:~/onvifdevicemanager/OnvifDeviceManager$

```

**Make output after errno fix
```
bridgeadmin@moodjacksonbridge:~/onvifdevicemanager/OnvifDeviceManager$ make -j$(nproc)
  CC       src/app/settings/onvifmgr-app_settings.o
  CC       src/gst/onvifmgr-onvifinitstaticplugins.o
  CC       src/gst/onvifmgr-overlay.o
  CC       src/gst/onvifmgr-src_retriever.o
  CC       src/gst/gtk/onvifmgr-gstplugin.o
  CC       src/gst/gtk/onvifmgr-gstgtkbasesink.o
  CC       src/gst/gtk/onvifmgr-gstgtksink.o
  CC       src/gst/gtk/onvifmgr-gstgtkutils.o
  CC       src/gst/gtk/onvifmgr-gtkgstbasewidget.o
  CC       src/gst/gtk/onvifmgr-gtkgstwidget.o
  CC       src/gst/onvifmgr-gstrtspplayer.o
  CCLD     locked-icon.o
  CCLD     microphone.o
  CCLD     warning.o
  CCLD     save.o
  CCLD     tower.o
  CC       src/queue/onvifmgr-event_queue.o
  CC       src/queue/onvifmgr-queue_event.o
  CC       src/queue/onvifmgr-queue_thread.o
  CCLD     onvifmgr
```

**Error output when adding an rtsp feed

```
bridgeadmin@moodjacksonbridge:~/onvifdevicemanager/OnvifDeviceManager$ ./onvifmgr 
INFO  24-03-06 23:51:39.923323 140177204042368 onvif-mgr.c:103: Onvif Device Manager Version : 0.0
INFO  24-03-06 23:51:39.923401 140177204042368 onvif-mgr.c:104: Using Gstreamer Version : 1.22.9.0
INFO  24-03-06 23:51:39.923419 140177204042368 onvifinitstaticplugins.c:223: Initializing Gstreamer plugins...
INFO  24-03-06 23:51:39.923665 140177204042368 onvifinitstaticplugins.c:228: Loading static plugins...
DEBUG 24-03-06 23:51:39.941358 140177204042368 onvifinitstaticplugins.c:435: Gstreamer plugins initialized...
WARN  24-03-06 23:51:39.941378 140177204042368 onvif-mgr.c:73: Element 'avdec_h264' not found.
WARN  24-03-06 23:51:39.941382 140177204042368 onvif-mgr.c:73: Element 'avdec_h265' not found.
WARN  24-03-06 23:51:39.941388 140177204042368 onvif-mgr.c:81: Found element 'openh264dec'. Increasing priority from 64 to 65...
INFO  24-03-06 23:51:39.941398 140177204042368 onvif-mgr.c:48: *** video/x-h264 decoders ***
INFO  24-03-06 23:51:39.941552 140177204042368 onvif-mgr.c:56: *    openh264dec[65]
INFO  24-03-06 23:51:39.941567 140177204042368 onvif-mgr.c:61: ****************************
INFO  24-03-06 23:51:39.941572 140177204042368 onvif-mgr.c:48: *** video/x-h265 decoders ***
INFO  24-03-06 23:51:39.941687 140177204042368 onvif-mgr.c:56: *    libde265dec[128]
INFO  24-03-06 23:51:39.941696 140177204042368 onvif-mgr.c:61: ****************************
INFO  24-03-06 23:51:39.941705 140177204042368 onvif-mgr.c:48: *** image/jpeg decoders ***
INFO  24-03-06 23:51:39.941770 140177204042368 onvif-mgr.c:56: *    jpegdec[256]
[... 162 lines cut ...]
TRACE 24-03-06 23:52:13.984158 140177062229760 onvif_media_service.c:270: [http://10.1.10.26:80/onvif/media_service] OnvifMediaService__get_profile_token 0
TRACE 24-03-06 23:52:13.984168 140177062229760 onvif_media_service.c:283: [http://10.1.10.26:80/onvif/media_service] OnvifMediaService__get_profile_token : Found profile [ProfileToken017]
TRACE 24-03-06 23:52:13.984185 140177053837056 onvif_base_service.c:117: [http://10.1.10.26:80/onvif/media_service] Time adjustment [2024-03-07 04:52:13] to [2024-03-07 04:52:12] offset -1
INFO  24-03-06 23:52:13.988147 140177053837056 gstrtspplayer.c:851: set location : rtsp://10.1.10.26:8554/Streaming/Unicast/channels/1701
DEBUG 24-03-06 23:52:13.988192 140177062229760 onvif_media_service.c:150: [http://10.1.10.26:80/onvif/media_service] Invoking GetSnapshotUri
DEBUG 24-03-06 23:52:13.988222 140177053837056 gstrtspplayer.c:190: RtspPlayer__play retry[0] - playing[0]
TRACE 24-03-06 23:52:13.988313 140177062229760 onvif_base_service.c:117: [http://10.1.10.26:80/onvif/media_service] Time adjustment [2024-03-07 04:52:13] to [2024-03-07 04:52:12] offset -1
TRACE 24-03-06 23:52:13.988387 140177053837056 onvif_app.c:302: _play_onvif_stream - done
TRACE 24-03-06 23:52:13.988462 140177053837056 event_queue.c:173: notify dispatched event...
DEBUG 24-03-06 23:52:13.988614 140177204042368 app_dialog.c:131: hide
INFO  24-03-06 23:52:13.991016 140177062229760 onvif_media_service.c:253: [http://10.1.10.26:80/onvif/media_service] Snapshot URI : http://10.1.10.26:80/onvif/snapshot/1701

(onvifmgr:188730): Gtk-WARNING **: 23:52:13.996: A floating object was finalized. This means that someone
called g_object_unref() on an object that had only a floating
reference; the initial floating reference is not owned by anyone
and must be removed with g_object_ref_sink().
Error: signal 11:
./onvifmgr(+0x5cdb0)[0x562ff4e61db0]
/lib/x86_64-linux-gnu/libc.so.6(+0x43090)[0x7f7d8d9d5090]
./onvifmgr(+0x624f8)[0x562ff4e674f8]
./onvifmgr(+0x636db)[0x562ff4e686db]
./onvifmgr(+0x73ad5)[0x562ff4e78ad5]
/lib/x86_64-linux-gnu/libpthread.so.0(+0x8609)[0x7f7d8db8c609]
/lib/x86_64-linux-gnu/libc.so.6(clone+0x43)[0x7f7d8dab1353]
```

Quedale, 2024-03-07:
Thanks for reporting this issue!

It seems to be similar to Issue #18.
Although he/she didn't respond with additional details yet.
Something bad seems to happen either during snapshot loading or while the stream starts.

Can you reconfigure the project with debugging enabled and then run it using valgrind?
```
rm -rf ./subprojects/OnvifSoapLib/build/dist/*
./autogen.sh --enable-latest --full-debug
make clean
make -j$(nproc)
valgrind ./onvifmgr
```
Doing so will add relevant details to valgrind's output.
</br>

Thanks again for your help improving this project!

Quedale, 2024-03-07:
I figured out the cause of the crash and at least one way to reproduce it.

The last commit should fix the crash, while I figure out why the snapshot won't load.

Let me know how that goes!

Quedale, 2024-03-08:
I actually managed to fix the thumbnail not loading properly as well.

Looking forward to hear back!