Quedale/OnvifDeviceManager #24: Linux mint 21.3

onyx82, 2024-03-30:
Hello.
I tried to run the command ./autogen.sh --prefix=$(pwd)/dist --enable-latest
make -j$(nproc), but various errors appear.
I understand that this command tries to install those packages by itself, but it fails.
I downloaded the gsoap 2.8.123 package on my own, installed it, but other errors appeared, something related to "CUtils is already up-to-date. Do nothing...
Package cutils was not found in the pkg-config search path.
Perhaps you should add the directory containing `cutils.pc'
to the PKG_CONFIG_PATH environment variable
No package 'cutils' found" the new gsoap version was not visible in synaptic and I deleted it, but I reinstalled the old gsoap version.
what could i do to solve it?

```
~/Documente/OnvifDeviceManager/OnvifDeviceManager-main$ ./autogen.sh --prefix=$(pwd)/dist --enable-latest
make -j$(nproc)
aclocal: warning: couldn't open directory 'm4': No such file or directory
configure.ac:12: installing './compile'
configure.ac:193: installing './config.guess'
configure.ac:193: installing './config.sub'
configure.ac:194: installing './install-sh'
configure.ac:193: error: required file './ltmain.sh' not found
configure.ac:194: installing './missing'
Makefile.am: installing './depcomp'
aclocal: warning: couldn't open directory 'm4': No such file or directory
libtoolize: putting auxiliary files in '.'.
libtoolize: copying file './ltmain.sh'
libtoolize: putting macros in AC_CONFIG_MACRO_DIRS, 'm4'.
libtoolize: copying file 'm4/libtool.m4'
libtoolize: copying file 'm4/ltoptions.m4'
libtoolize: copying file 'm4/ltsugar.m4'
libtoolize: copying file 'm4/ltversion.m4'
libtoolize: copying file 'm4/lt~obsolete.m4'
--2024-03-30 20:31:38--  https://bootstrap.pypa.io/virtualenv.pyz
It is resolved bootstrap.pypa.io (bootstrap.pypa.io)... 151.101.36.175, 2a04:4e42:9::175
Connects to bootstrap.pypa.io (bootstrap.pypa.io)|151.101.36.175|:443... conectat.
HTTP request sent, waiting for response... 200 OK
Dimensiune: 5705128 (5,4M) [application/octet-stream]
It is saved to: ‘virtualenv.pyz’

virtualenv.pyz      100%[===================>]   5,44M  15,6MB/s    în 0,3s    

2024-03-30 20:31:39 (15,6 MB/s) - ‘virtualenv.pyz’ salvat [5705128/5705128]

created virtual environment CPython3.10.12.final.0-64 in 598ms
  creator CPython3Posix(dest=/home/alexandru/Documente/OnvifDeviceManager/OnvifDeviceManager-main/subprojects/venvfolder, clear=False, no_vcs_ignore=False, global=False)
  seeder FromAppData(download=False, pip=bundle, setuptools=bundle, wheel=bundle, via=copy, app_data_dir=/home/alexandru/.local/share/virtualenv)
    added seed packages: pip==24.0, setuptools==69.1.0, wheel==0.42.0
  activators BashActivator,CShellActivator,FishActivator,NushellActivator,PowerShellActivator,PythonActivator
Meson 1.4.0 already installed.
Ninja 1.11.1.git.kitware.jobserver-1 already installed.
openssl already found.
zlib already found.
Requested 'gsoap >= 2.8.123' but version of GSoap is 2.8.117
-- Building gsoap libgsoap-dev --
*****************************
*** Downloading : https://sourceforge.net/projects/gsoap2/files/gsoap_2.8.129.zip/download ***
*** Destination : /home/alexandru/Documente/OnvifDeviceManager/OnvifDeviceManager-main/subprojects/.cache/gsoap.zip ***
*****************************
--2024-03-30 20:31:42--  https://sourceforge.net/projects/gsoap2/files/gsoap_2.8.129.zip/download
It is resolved sourceforge.net (sourceforge.net)... 104.18.37.111, 172.64.150.145, 2606:4700:4400::ac40:9691, ...
It connects to sourceforge.net (sourceforge.net)|104.18.37.111|:443... logged in.
Cerere HTTP trimisă, se așteaptă răspuns... 404 Not Found
2024-03-30 20:31:42 EROARE 404: Not Found.

*****************************
*** Extracting : gsoap.zip ***
*****************************
Archive:  /home/alexandru/Documente/OnvifDeviceManager/OnvifDeviceManager-main/subprojects/.cache/gsoap.zip
  End-of-central-directory signature not found.  Either this file is not
  a zipfile, or it constitutes one disk of a multi-part archive.  In the
  latter case the central directory and zipfile comment will be found on
  the last disk(s) of this archive.
unzip:  cannot find zipfile directory in one of /home/alexandru/Documente/OnvifDeviceManager/OnvifDeviceManager-main/subprojects/.cache/gsoap.zip or
        /home/alexandru/Documente/OnvifDeviceManager/OnvifDeviceManager-main/subprojects/.cache/gsoap.zip.zip, and cannot find /home/alexandru/Documente/OnvifDeviceManager/OnvifDeviceManager-main/subprojects/.cache/gsoap.zip.ZIP, period.
*****************************
* Building Project ***
* Src dir : gsoap-2.8 ***
* Prefix : /home/alexandru/Documente/OnvifDeviceManager/OnvifDeviceManager-main/subprojects/gsoap-2.8/build/dist ***
*****************************
./autogen.sh: linia 247: cd: gsoap-2.8: Nu există un astfel de fișier sau director
*****************************
*** no configuration available gsoap-2.8 ***
*****************************
*****************************
*** compile gsoap-2.8 ***
*** Make Args :  ***
*****************************
make: *** No targets specified and no makefile found.  Stop.
*****************************
*** Make failed gsoap-2.8 ***
*****************************

```

Quedale, 2024-04-03:
It seems gsoap simply retired the version gsoap_2.8.129 for gsoap_2.8.133 on sourceforge.

Unfortunately, I build this project using sources found in the zip file, so installing the library from the package manager isn't enough.

I updated the faulty URL inside autogen.sh. 
You should be able to setup the environment with it now.

onyx82, 2024-04-21:
> It seems gsoap simply retired the version gsoap_2.8.129 for gsoap_2.8.133 on sourceforge.
>
> Unfortunately, I build this project using sources found in the zip file, so installing the library from the [...]

Hello, (my camera is imou bulet lite 4mp), your software was successfully installed, but it does not log in and in the terminal it says:
```

INFO  24-04-22 01:35:19.499587 139708396040064 onvif_app.c:412: OnvifAuthentication attempt...
TRACE 24-04-22 01:35:19.502886 139708334343744 event_queue.c:221: event notify...
TRACE 24-04-22 01:35:19.502910 139708334343744 onvif_app.c:468: EventQueue Dispatching : 1/0/8
DEBUG 24-04-22 01:35:19.502955 139708334343744 onvif_app.c:284: _onvif_authentication_reload
INFO  24-04-22 01:35:19.502970 139708334343744 onvif_device.c:72: [http://192.168.100.94/onvif/device_service] OnvifDevice__authenticate
INFO  24-04-22 01:35:19.503007 139708334343744 onvif_device.c:93: [http://192.168.100.94/onvif/device_service] Camera SystemDateAndTime : '2024-04-22 01:32:12'
TRACE 24-04-22 01:35:19.503020 139708334343744 onvif_device.c:46: [http://192.168.100.94/onvif/device_service] OnvifDevice__createMediaService
DEBUG 24-04-22 01:35:19.503035 139708334343744 onvif_device_service.c:99: [http://192.168.100.94/onvif/device_service] Invoking GetCapabilities
DEBUG 24-04-22 01:35:19.529460 139708334343744 onvif_device.c:102: [http://192.168.100.94/onvif/media_service] Successfully created Media soap
TRACE 24-04-22 01:35:19.529498 139708334343744 onvif_media_service.c:380: [http://192.168.100.94/onvif/media_service] OnvifMediaService__get_profile_token 0
DEBUG 24-04-22 01:35:19.529512 139708334343744 onvif_media_service.c:79: [http://192.168.100.94/onvif/media_service] Invoking GetProfiles
WARN  24-04-22 01:35:19.556182 139708334343744 onvif_base_service.c:247: Warning : NOT AUTHORIZED HTTP Error code [12]
WARN  24-04-22 01:35:19.556245 139708334343744 onvif_media_service.c:101: Profile token not found.
ERROR 24-04-22 01:35:19.556259 139708334343744 onvif_device.c:106: [http://192.168.100.94/onvif/media_service] No stream uri returned...
TRACE 24-04-22 01:35:19.556436 139708334343744 event_queue.c:221: event notify...
TRACE 24-04-22 01:35:19.556450 139708334343744 onvif_app.c:468: EventQueue Dispatched : 0/0/8

```

Quedale, 2024-05-08:
Good day!

Sorry for the slow reply. I just got my hands on the Imou Ranger 2C and it worked without issues.
I highly doubt that your device has a different authentication mechanism than mine.

I did encounter a specific issue if the camera gets rebooted, which I will look into.
However, this shouldn't prevent normal usage.

Can you make sure that the right credentials are being entered without typo?

If you are certain that the credentials are correct, can you try the following to extract soap debug details:
```
ONVIF_DEBUG=1 ./onvifmgr
```

Thanks!

onyx82, 2024-05-18:
Hello,i copied the login and password from the chrome browser from here: chrome://password-manager/passwords and did a copy paste. In chrome, at the web interface of the camera, I did so and it logged in. It is excluded that the login and password were wrong.
in the code I added 2 comments like this:<--!-->

alexandru@alexandru-desktop:~/Documente/onvif/OnvifDeviceManager$ ONVIF_DEBUG=1 ./onvifmgr
INFO  24-05-18 05:30:42.299978 139703319850880 onvif-mgr.c:39: Onvif Device Manager Version : 0.0
INFO  24-05-18 05:30:42.300056 139703319850880 onvif-mgr.c:40: Using Gstreamer Version : 1.24.1.0
INFO  24-05-18 05:30:42.300077 139703319850880 gst_plugin_utils.c:224: Initializing Gstreamer plugins...
INFO  24-05-18 05:30:42.300088 139703319850880 gst_plugin_utils.c:229: Loading static plugins...
DEBUG 24-05-18 05:30:42.330871 139703319850880 gst_plugin_utils.c:436: Gstreamer plugins initialized...
INFO  24-05-18 05:30:42.330904 139703319850880 onvif-mgr.c:45: **** Video decoders *******
INFO  24-05-18 05:30:42.330915 139703319850880 gst_plugin_utils.c:485: * video/x-h264 decoders ***
INFO  24-05-18 05:30:42.331112 139703319850880 gst_plugin_utils.c:494: *    openh264dec[64]
INFO  24-05-18 05:30:42.331128 139703319850880 gst_plugin_utils.c:485: * video/x-h265 decoders ***
INFO  24-05-18 05:30:42.331191 139703319850880 gst_plugin_utils.c:494: *    libde265dec[128]
INFO  24-05-18 05:30:42.331205 139703319850880 gst_plugin_utils.c:485: * image/jpeg decoders ***
INFO  24-05-18 05:30:42.331270 139703319850880 gst_plugin_utils.c:494: *    jpegdec[256]
INFO  24-05-18 05:30:42.331284 139703319850880 gst_plugin_utils.c:485: * video/x-av1 decoders ***
INFO  24-05-18 05:30:42.331362 139703319850880 gst_plugin_utils.c:499: *    None
INFO  24-05-18 05:30:42.331375 139703319850880 onvif-mgr.c:50: ****************************
INFO  24-05-18 05:30:42.331386 139703319850880 onvif-mgr.c:52: **** Audio decoders *******
INFO  24-05-18 05:30:42.331396 139703319850880 gst_plugin_utils.c:485: * audio/x-mulaw decoders ***
INFO  24-05-18 05:30:42.331459 139703319850880 gst_plugin_utils.c:494: *    mulawdec[256]
INFO  24-05-18 05:30:42.331472 139703319850880 gst_plugin_utils.c:485: * audio/x-alaw decoders ***
INFO  24-05-18 05:30:42.331536 139703319850880 gst_plugin_utils.c:494: *    alawdec[256]
INFO  24-05-18 05:30:42.331550 139703319850880 gst_plugin_utils.c:485: * audio/mpeg decoders ***
INFO  24-05-18 05:30:42.331613 139703319850880 gst_plugin_utils.c:494: *    fdkaacdec[64]
INFO  24-05-18 05:30:42.331627 139703319850880 onvif-mgr.c:56: ****************************
TRACE 24-05-18 05:30:42.331725 139703319850880 event_queue.c:88: create...
TRACE 24-05-18 05:30:42.373128 139703319850880 app_settings.c:86: Generating default config path from HOME directory : /home/alexandru
INFO  24-05-18 05:30:42.373183 139703319850880 app_settings.c:249: Reading Settings file /home/alexandru/.config/onvifmgr_settings.ini
WARN  24-05-18 05:30:42.373273 139703319850880 app_settings.c:306: No config file found. Using default configs. 1
DEBUG 24-05-18 05:30:42.376476 139703319850880 src_retriever.c:12: Test Audio Src [pulsesrc] device [(null)]
INFO  24-05-18 05:30:42.573689 139703319850880 backchannel.c:111: Creating backchannel using source element pulsesrc
DEBUG 24-05-18 05:30:42.574352 139703240291904 queue_thread.c:37: Started...
DEBUG 24-05-18 05:30:42.574401 139703249397312 queue_thread.c:37: Started...
DEBUG 24-05-18 05:30:42.574470 139703231514176 queue_thread.c:37: Started...
DEBUG 24-05-18 05:30:42.574552 139703223121472 queue_thread.c:37: Started...
DEBUG 24-05-18 05:30:42.574666 139702862603840 queue_thread.c:37: Started...
TRACE 24-05-18 05:30:42.574735 139703319850880 profiles_dialog.c:119: create
TRACE 24-05-18 05:30:42.574755 139703319850880 app_dialog.c:75: init
DEBUG 24-05-18 05:30:42.574784 139702854211136 queue_thread.c:37: Started...
DEBUG 24-05-18 05:30:42.574892 139702845818432 queue_thread.c:37: Started...
DEBUG 24-05-18 05:30:42.575332 139702870996544 queue_thread.c:37: Started...
TRACE 24-05-18 05:30:42.575497 139703319850880 profiles_dialog.c:25: priv_ProfilesDialog__create_ui
TRACE 24-05-18 05:30:42.575861 139703319850880 app_dialog.c:75: init
TRACE 24-05-18 05:30:42.578393 139703319850880 credentials_input.c:28: create
TRACE 24-05-18 05:30:42.578409 139703319850880 app_dialog.c:75: init
TRACE 24-05-18 05:30:42.579922 139703319850880 msg_dialog.c:60: create
TRACE 24-05-18 05:30:42.579936 139703319850880 app_dialog.c:75: init

<--!My cooment   "Here I pressed the search button"-->

INFO  24-05-18 05:30:53.773890 139703319850880 onvif_app.c:499: Starting ONVIF Devices Network Discovery...
TRACE 24-05-18 05:30:53.775747 139703240291904 event_queue.c:221: event notify...
TRACE 24-05-18 05:30:53.775958 139703240291904 onvif_app.c:468: EventQueue Dispatching : 1/0/8
TRACE 24-05-18 05:30:53.776099 139703240291904 onvif_app.c:130: _start_onvif_discovery
DEBUG 24-05-18 05:30:53.780286 139703240291904 onvif_discovery.c:208: Sending NVT probe...
DEBUG 24-05-18 05:30:53.953309 139703240291904 onvif_discovery.c:64: ProbeMatches found...
TRACE 24-05-18 05:30:53.953343 139703240291904 probmatch.c:15: ProbMatches__create...
TRACE 24-05-18 05:30:53.953355 139703240291904 probmatch.c:84: ProbMatch__create...
TRACE 24-05-18 05:30:53.953394 139703240291904 onvif_app.c:120: _dicovery_found_server_cb
TRACE 24-05-18 05:30:53.953479 139703319850880 onvif_app.c:376: OnvifApp__found_server_cb
[... cut ...]
INFO  24-05-18 05:31:49.704786 139703223121472 onvif_base_service.c:205: ONVIF_DEBUG variable set. '1'
WARN  24-05-18 05:31:49.704833 139703223121472 onvif_media_service.c:101: Profile token not found.
ERROR 24-05-18 05:31:49.704852 139703223121472 onvif_device.c:106: [http://192.168.100.94/onvif/media_service] No stream uri returned...
TRACE 24-05-18 05:31:49.705058 139703223121472 event_queue.c:221: event notify...
TRACE 24-05-18 05:31:49.705072 139703223121472 onvif_app.c:468: EventQueue Dispatched : 0/0/8

Quedale, 2024-05-18:
Okay, we can clearly see now that it's not an authentication issue.
`HTTP/1.1 400 Bad Request`.

It seems that the soap request `GetProfiles` got corrupted somehow :
```
....</trt:GetProfiles>undefined</SOAP-ENV:Body>undefined
```

This corruption seems to be caused by an invalid soap response to `GetCapabilities` :
```
HTTP/1.1 200 OK
Content-Type: application/soap+xml; charset=utf-8
Connection: close
Content-Length: 4846
http://192.168.100.94/onvif/analytics_servicetruetruehttp://192.168.100.94/onvif/device_servicefalsefalsefalsefalsefalsefalsetruefalsefalsetruetrue20021022023024024216121806181219061912truefalsefalsefalse00falsefalsefalsefalsefalsefalsefalsefalsefalsefalsefalse0falsehttp://192.168.100.94/onvif/event_servicetruetruefalsehttp://192.168.100.94/onvif/imaging_servicehttp://192.168.100.94/onvif/media_servicetruetruetrue6http://192.168.100.94/onvif/deviceIO_service10100http://192.168.100.94/onvif/recording_servicefalsetruetruefalse129http://192.168.100.94/onvif/search_servicefalsehttp://192.168.100.94/onvif/replay_service
```
This soap XML structure is entirely missing leaving only the raw data.
I honestly have no clue why this is happening.

I will setup a Linux Mint VM and try to reproduce this.

Quedale, 2024-05-18:
Until I find the time to setup a new environment, could you try running `autogen.sh` again and rebuilt it?
I noticed that you are still using Gstreamer Version : 1.24.1.0, which has been updated to 1.24.3.

This also hints that you may have missing updates in the OnvifSoapLib dependency.

The `autogen.sh` script is design to run incrementally and will look for updated packages.

onyx82, 2024-05-19:
I deleted everything and tried again:

alexandru@alexandru-desktop:~/Documente/imouonvif$ cd OnvifDeviceManager
alexandru@alexandru-desktop:~/Documente/imouonvif/OnvifDeviceManager$ ONVIF_DEBUG=1 ./onvifmgr
INFO  24-05-19 20:54:24.140190 139928463294336 onvif-mgr.c:39: Onvif Device Manager Version : 0.0
INFO  24-05-19 20:54:24.141802 139928463294336 onvif-mgr.c:40: Using Gstreamer Version : 1.24.3.0
INFO  24-05-19 20:54:24.141841 139928463294336 gst_plugin_utils.c:224: Initializing Gstreamer plugins...
INFO  24-05-19 20:54:24.141854 139928463294336 gst_plugin_utils.c:229: Loading static plugins...
DEBUG 24-05-19 20:54:24.176118 139928463294336 gst_plugin_utils.c:436: Gstreamer plugins initialized...
INFO  24-05-19 20:54:24.176160 139928463294336 onvif-mgr.c:45: **** Video decoders *******
INFO  24-05-19 20:54:24.176182 139928463294336 gst_plugin_utils.c:485: * video/x-h264 decoders ***
INFO  24-05-19 20:54:24.176401 139928463294336 gst_plugin_utils.c:494: *    openh264dec[64]
INFO  24-05-19 20:54:24.176419 139928463294336 gst_plugin_utils.c:485: * video/x-h265 decoders ***
INFO  24-05-19 20:54:24.176488 139928463294336 gst_plugin_utils.c:494: *    libde265dec[128]
INFO  24-05-19 20:54:24.176507 139928463294336 gst_plugin_utils.c:485: * image/jpeg decoders ***
INFO  24-05-19 20:54:24.176575 139928463294336 gst_plugin_utils.c:494: *    jpegdec[256]
INFO  24-05-19 20:54:24.176591 139928463294336 gst_plugin_utils.c:485: * video/x-av1 decoders ***
INFO  24-05-19 20:54:24.176660 139928463294336 gst_plugin_utils.c:499: *    None
INFO  24-05-19 20:54:24.176675 139928463294336 onvif-mgr.c:50: ****************************
INFO  24-05-19 20:54:24.176690 139928463294336 onvif-mgr.c:52: **** Audio decoders *******
INFO  24-05-19 20:54:24.176701 139928463294336 gst_plugin_utils.c:485: * audio/x-mulaw decoders ***
INFO  24-05-19 20:54:24.176770 139928463294336 gst_plugin_utils.c:494: *    mulawdec[256]
INFO  24-05-19 20:54:24.176785 139928463294336 gst_plugin_utils.c:485: * audio/x-alaw decoders ***
INFO  24-05-19 20:54:24.176853 139928463294336 gst_plugin_utils.c:494: *    alawdec[256]
INFO  24-05-19 20:54:24.176869 139928463294336 gst_plugin_utils.c:485: * audio/mpeg decoders ***
INFO  24-05-19 20:54:24.176937 139928463294336 gst_plugin_utils.c:494: *    fdkaacdec[64]
INFO  24-05-19 20:54:24.176954 139928463294336 onvif-mgr.c:56: ****************************
TRACE 24-05-19 20:54:24.177123 139928463294336 event_queue.c:95: create...
TRACE 24-05-19 20:54:24.220338 139928463294336 app_settings.c:96: Generating default config path from HOME directory : /home/alexandru
INFO  24-05-19 20:54:24.220383 139928463294336 app_settings.c:259: Reading Settings file /home/alexandru/.config/onvifmgr_settings.ini
WARN  24-05-19 20:54:24.220419 139928463294336 app_settings.c:332: No config file found. Using default configs. 1
DEBUG 24-05-19 20:54:24.223869 139928463294336 src_retriever.c:12: Test Audio Src [pulsesrc] device [(null)]
INFO  24-05-19 20:54:24.230854 139928463294336 backchannel.c:111: Creating backchannel using source element pulsesrc
TRACE 24-05-19 20:54:24.231421 139928463294336 profiles_dialog.c:119: create
TRACE 24-05-19 20:54:24.231448 139928463294336 app_dialog.c:75: init
TRACE 24-05-19 20:54:24.232182 139928463294336 profiles_dialog.c:25: priv_ProfilesDialog__create_ui
TRACE 24-05-19 20:54:24.232540 139928463294336 app_dialog.c:75: init
TRACE 24-05-19 20:54:24.235529 139928463294336 credentials_input.c:28: create
TRACE 24-05-19 20:54:24.235581 139928463294336 app_dialog.c:75: init
TRACE 24-05-19 20:54:24.237363 139928463294336 msg_dialog.c:60: create
TRACE 24-05-19 20:54:24.237407 139928463294336 app_dialog.c:75: init
DEBUG 24-05-19 20:54:24.326134 139928373720640 queue_thread.c:74: Started...
DEBUG 24-05-19 20:54:24.326405 139928392840768 queue_thread.c:74: Started...
DEBUG 24-05-19 20:54:24.326815 139927986570816 queue_thread.c:74: Started...
DEBUG 24-05-19 20:54:24.327065 139927978178112 queue_thread.c:74: Started...
DEBUG 24-05-19 20:54:24.327143 139927994963520 queue_thread.c:74: Started...
DEBUG 24-05-19 20:54:24.327210 139928003356224 queue_thread.c:74: Started...
DEBUG 24-05-19 20:54:24.327276 139928011748928 queue_thread.c:74: Started...
DEBUG 24-05-19 20:54:24.328207 139928020141632 queue_thread.c:74: Started...
TRACE 24-05-19 20:54:24.329228 139928373720640 onvif_app.c:458: EventQueue Started : 0/0/8
TRACE 24-05-19 20:54:24.329347 139928373720640 onvif_app.c:458: EventQueue Started : 0/0/8

INFO  24-05-19 20:54:38.169185 139928463294336 onvif_app.c:495: Starting ONVIF Devices Network Discovery...
INFO  24-05-19 20:54:38.170365 139928463294336 onvif_app.c:528: Starting ONVIF Devices Network Discovery... done
TRACE 24-05-19 20:54:38.170437 139928392840768 onvif_app.c:458: EventQueue Dispatching : 1/0/8
TRACE 24-05-19 20:54:38.170467 139928392840768 onvif_app.c:129: _start_onvif_discovery
DEBUG 24-05-19 20:54:38.174506 139928392840768 onvif_discovery.c:210: Sending NVT probe...
DEBUG 24-05-19 20:54:38.410961 139928392840768 onvif_discovery.c:65: ProbeMatches found...
TRACE 24-05-19 20:54:38.410997 139928392840768 probmatch.c:15: ProbMatches__create...
TRACE 24-05-19 20:54:38.411009 139928392840768 probmatch.c:84: ProbMatch__create...
TRACE 24-05-19 20:54:38.411026 139928392840768 onvif_app.c:119: _dicovery_found_server_cb
TRACE 24-05-19 20:54:38.411139 139928463294336 onvif_app.c:368: OnvifApp__found_server_cb
[... cut ...]
INFO  24-05-19 20:55:41.950320 139927994963520 onvif_base_service.c:209: ONVIF_DEBUG variable set. '1'
WARN  24-05-19 20:55:41.950361 139927994963520 onvif_media_service.c:101: Profile token not found.
ERROR 24-05-19 20:55:41.950376 139927994963520 onvif_device.c:106: [http://192.168.100.94/onvif/media_service] No stream uri returned...
TRACE 24-05-19 20:55:41.950581 139927994963520 onvif_app.c:458: EventQueue Dispatched : 0/0/8
^C
alexandru@alexandru-desktop:~/Documente/imouonvif/OnvifDeviceManager$

Quedale, 2024-10-30:
I actually encountered a very similar issue (getting 400 Bad Request) on my own Imou camera yesterday.
It turns out that the camera's web service had crashed/corrupted and it needed a factory reset.

Although it was a camera failure, I wouldn't give it the entire blame since it was caused by a stress test I was running on my application. (A couple hundreds connections within a few minutes)

Considering your comment on Issue #30, I assume you managed to work around this failure by adding a new user.
I'm going to close this issue and carry on with the new one.

Feel free to reopen it if I'm mistaken.

Cheers!

closed 2024-10-30 by Quedale