Quedale/OnvifDeviceManager #25: Deb File Share

AnonymousWebHacker, 2024-05-18:
any can share the deb ?

Quedale, 2024-05-18:
Good day!

What do you mean by share?
The code is published under GPL3, so there isn't any issues sharing it as long as the license is honoured.

That being said, this tool isn't ready for its first release yet.
I still know about a few issues that needs to be resolved before I create an initial release.
For instance, the current deb configurations doesn't have any dependencies defined yet.

AnonymousWebHacker, 2024-05-18:
I was referring to someone sharing a built deb releases. Just so as not to be contaminating the compiler system...etc.

Quedale, 2024-05-18:
Okay, I see! You are asking for someone to share it.

Unfortunately, without package dependencies configuration in the deb metadata, there's a high chance that it won't work out-of-the-box for another computer.

This project was designed to not contaminate the host system.
Except the mandatory packages `make bison, flex, libtool, pkg-config, libgtk-3-dev, g++`, the `autogen.sh` will build every missing dependencies from source isolated under "./subprojects" folder.
**Nothing is installed anywhere outside this project's folder**
**No sudo privileged is require to complete the build.**

Once you successfully compiled the project, you can run the following to get an installable package for your system.
```
make deb
```

Working on a distributable deb/rpm packages is on top of my TODO list.

AnonymousWebHacker, 2024-05-18:
Thanks. Waiting for release packages .

closed 2024-05-18 by AnonymousWebHacker

Quedale, 2024-05-18:
I suggest that you follow Issue #9.
It is intentionally kept open as a feature tracking request.

Cheers!