Quedale/OnvifDeviceManager #26: fatal error: stdlib.h: No such file or directory

Adityashaw, 2024-06-20:
Any leads on how to debug this error? log file attached.

g++ works fine. I wrote sample porgrams importing `stdlib.h`. They are working fine.

```
locate stdlib.h
/usr/include/stdlib.h
/usr/include/bash/include/ansi_stdlib.h
/usr/include/bits/stdlib.h
/usr/include/bsd/stdlib.h
/usr/include/c++/14.1.1/stdlib.h
/usr/include/c++/14.1.1/tr1/stdlib.h
/usr/include/freetype2/freetype/config/ftstdlib.h
/usr/lib/clang/17/include/__clang_hip_stdlib.h
/usr/lib/clang/17/include/llvm_libc_wrappers/stdlib.h
/usr/share/man/man0p/stdlib.h.0p.gz
```
[log.txt](https://github.com/user-attachments/files/15917705/log.txt)

Quedale, 2024-06-21:
That seems to be an environmental problem where the system's include path got modified.

I can't know for sure how it happened on your system, but doing the following should resolve it:
```
export CPLUS_INCLUDE_PATH=/usr/include/c++/14.1.1:$CPLUS_INCLUDE_PATH
```
It simply adds your c++ include folder in the environment path.
See [this stackoverflow response](https://stackoverflow.com/questions/66304331/usr-include-c-8-cstdlib7515-fatal-error-stdlib-h-no-such-file-or-directo) for additional info.

</br>
</br>

I tested this build on a few vanilla distro, but I could test more if you want to share what you are using.

Adityashaw, 2024-06-21:
> I can't know for sure how it happened on your system, but doing the following should resolve it:
> 
> ```
> export CPLUS_INCLUDE_PATH=/usr/include/c++/14.1.1:$CPLUS_INCLUDE_PATH
> ```
> 

I tried that but that didn't help. Same error.

> 
> I tested this build on a few vanilla distro, but I could test more if you want to share what you are using.

```
uname -a
Linux hp 6.9.3-arch1-1 #1 SMP PREEMPT_DYNAMIC Fri, 31 May 2024 15:14:45 +0000 x86_64 GNU/Linux
```

It's arch linux. Let me know if you need any other details for debugging.

Quedale, 2024-06-22:
I actually never tried Archlinux, but I was able to reproduce the same issue on a fresh install.

The problem is related to the fact that I inject a CPLUS_INCLUDE_PATH variable to build gsoap with a custom openssl or zlib path.
It seems that either gcc14 or archlinux doesn't like having "/usr/include" added to "CPLUS_INCLUDE_PATH", which breaks "include_next".

I modified 'autogen.sh' to handle include path slightly better to workaround this issue.
Pull the latest change and give it another go.

Thanks for this feedback!

Adityashaw, 2024-06-22:
Thanks for explaining the issue and fixing it up so fast!

Build succeeds now.

closed 2024-06-22 by Adityashaw

Quedale, 2024-06-22:
I'm glad that you figured all the package necessary!
I should be able to update the documentation to include archlinux and add a rpm package.

Cheers!