Quedale/OnvifDeviceManager #27: compilation issue with OnvifSoapLib / Linking disco-test/client-test

starapo7348, 2024-09-01:
I have a problem compiling and linking OnvifSoapLib.

Building on:
OS: Debian GNU/Linux trixie/sid (trixie) x86_64 
Kernel: 6.10.6-amd64

./autogen.sh --prefix=$(pwd)/dist --enable-latest                                                          
cmake already installed.
Meson 1.5.1 already installed.
Ninja 1.12.1 already installed.
openssl already found.
zlib already found.
gsoap already found.
glib already found.
CUtils is already up-to-date. Do nothing...
CUtils already found.
libntlm already found.
OnvifSoapLib is already up-to-date. Do nothing...
Package onvifsoap was not found in the pkg-config search path.
Perhaps you should add the directory containing `onvifsoap.pc'
to the PKG_CONFIG_PATH environment variable
Package 'onvifsoap', required by 'virtual:world', not found
-- Bootstrap OnvifSoapLib  --

```[ 93%] Building C object CMakeFiles/onvifsoap.dir/home/user/coding/OnvifDeviceManager/subprojects/gsoap-2.8/gsoap/custom/struct_timeval.c.o
[ 94%] Linking C static library /home/user/coding/OnvifDeviceManager/subprojects/OnvifSoapLib/release/libonvifsoap.a
[ 94%] Built target onvifsoap
[ 97%] Building C object example/CMakeFiles/client-test.dir/client-test.c.o
[ 97%] Building C object example/CMakeFiles/disco-test.dir/disco-test.c.o
[ 98%] Linking C executable /home/user/coding/OnvifDeviceManager/subprojects/OnvifSoapLib/release/disco-test
[100%] Linking C executable /home/user/coding/OnvifDeviceManager/subprojects/OnvifSoapLib/release/client-test
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libcrypto.a(libcrypto-lib-c_zlib.o): in function `zlib_oneshot_expand_block':
(.text+0xbf4): undefined reference to `uncompress'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libcrypto.a(libcrypto-lib-c_zlib.o): in function `zlib_oneshot_compress_block':
(.text+0xc74): undefined reference to `compress'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libcrypto.a(libcrypto-lib-c_zstd.o): in function `zstd_stateful_expand_block':
(.text+0xaf): undefined reference to `ZSTD_decompressStream'
/usr/bin/ld: (.text+0xba): undefined reference to `ZSTD_isError'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libcrypto.a(libcrypto-lib-c_zstd.o): in function `zstd_stateful_compress_block':
(.text+0x188): undefined reference to `ZSTD_compressStream2'
/usr/bin/ld: (.text+0x190): undefined reference to `ZSTD_isError'
/usr/bin/ld: (.text+0x1cf): undefined reference to `ZSTD_endStream'
/usr/bin/ld: (.text+0x1d7): undefined reference to `ZSTD_isError'
/usr/bin/ld: (.text+0x1ff): undefined reference to `ZSTD_flushStream'
/usr/bin/ld: (.text+0x207): undefined reference to `ZSTD_isError'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libcrypto.a(libcrypto-lib-c_zstd.o): in function `zstd_stateful_finish':
(.text+0x23b): undefined reference to `ZSTD_freeCStream'
/usr/bin/ld: (.text+0x244): undefined reference to `ZSTD_freeDStream'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libcrypto.a(libcrypto-lib-c_zstd.o): in function `zstd_stateful_init':
(.text+0x313): undefined reference to `ZSTD_createCStream_advanced'
/usr/bin/ld: (.text+0x32c): undefined reference to `ZSTD_initCStream'
/usr/bin/ld: (.text+0x346): undefined reference to `ZSTD_createDStream_advanced'
/usr/bin/ld: (.text+0x35b): undefined reference to `ZSTD_initDStream'
/usr/bin/ld: (.text+0x384): undefined reference to `ZSTD_freeCStream'
/usr/bin/ld: (.text+0x38d): undefined reference to `ZSTD_freeDStream'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libcrypto.a(libcrypto-lib-c_zstd.o): in function `bio_zstd_new':
(.text+0x443): undefined reference to `ZSTD_createDStream_advanced'
/usr/bin/ld: (.text+0x457): undefined reference to `ZSTD_initDStream'
/usr/bin/ld: (.text+0x45c): undefined reference to `ZSTD_DStreamInSize'
/usr/bin/ld: (.text+0x47a): undefined reference to `ZSTD_createCStream_advanced'
/usr/bin/ld: (.text+0x494): undefined reference to `ZSTD_initCStream'
/usr/bin/ld: (.text+0x499): undefined reference to `ZSTD_CStreamInSize'
/usr/bin/ld: (.text+0x500): undefined reference to `ZSTD_freeDStream'
/usr/bin/ld: (.text+0x509): undefined reference to `ZSTD_freeCStream'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libcrypto.a(libcrypto-lib-c_zstd.o): in function `bio_zstd_ctrl':
(.text+0x6d4): undefined reference to `ZSTD_getErrorName'
/usr/bin/ld: (.text+0x780): undefined reference to `ZSTD_flushStream'
/usr/bin/ld: (.text+0x78b): undefined reference to `ZSTD_isError'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libcrypto.a(libcrypto-lib-c_zstd.o): in function `bio_zstd_write':
(.text+0x982): undefined reference to `ZSTD_compressStream2'
[... 59 lines cut ...]
/usr/bin/ld: (.text+0xbe8): undefined reference to `ZSTD_getErrorName'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libcrypto.a(libcrypto-lib-c_zstd.o): in function `zstd_oneshot_expand_block':
(.text+0xd2e): undefined reference to `ZSTD_decompress'
/usr/bin/ld: (.text+0xd39): undefined reference to `ZSTD_isError'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libcrypto.a(libcrypto-lib-c_zstd.o): in function `zstd_oneshot_compress_block':
(.text+0xda4): undefined reference to `ZSTD_compress'
/usr/bin/ld: (.text+0xdaf): undefined reference to `ZSTD_isError'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libcrypto.a(libcrypto-lib-c_zstd.o): in function `bio_zstd_free':
(.text+0xe1c): undefined reference to `ZSTD_freeDStream'
/usr/bin/ld: (.text+0xe36): undefined reference to `ZSTD_freeCStream'
collect2: error: ld returned 1 exit status
make[2]: *** [example/CMakeFiles/disco-test.dir/build.make:101: /home/user/coding/OnvifDeviceManager/subprojects/OnvifSoapLib/release/disco-test] Fehler 1
make[1]: *** [CMakeFiles/Makefile2:912: example/CMakeFiles/disco-test.dir/all] Fehler 2
make[1]: *** Es wird auf noch nicht beendete Prozesse gewartet....
collect2: error: ld returned 1 exit status
make[2]: *** [example/CMakeFiles/client-test.dir/build.make:101: /home/user/coding/OnvifDeviceManager/subprojects/OnvifSoapLib/release/client-test] Fehler 1
make[1]: *** [CMakeFiles/Makefile2:886: example/CMakeFiles/client-test.dir/all] Fehler 2
make: *** [Makefile:166: all] Fehler 2
*****************************
*** Make failed OnvifSoapLib ***
*****************************
```

What about these "undefined references"? Are there any additonal libs missing?
Any ideas what's wrong here?

Quedale, 2024-09-18:
Good day!

Thank you for this feedback and sorry for the late reply!

I honestly don't know what happened with the initial failure.
Both Openssl and Zlib are automatically built when missing.

You can actually see in the logs:
```
[...]
openssl already found.
zlib already found.
[...]
```

I'm currently working out of kubuntu 24.04, so there might be issues when working with an older version of zlib.
Can you provide the version of zlib installed?

*Reply regarding updates:*
Did you managed to run `autogen.sh` entirely without error?   
If not, I doubt the build will work since `autogen.sh` takes care of setting up missing libraries and only one missing library will lead to cascading errors.

Quedale, 2024-09-19:
I successfully build the project under a vanila DebianTrexie installation.
I did have to make some minor change to be successful tho.

I can tell from the logs that this wasn't its first run.
I'm not sure what happened on the first run, but it's possible that I didn't properly cleanup after a specific crash.

Can you try to pull the latest changes, delete the entire "subprojects" folder and run `autogen.sh` again?

If you encounter any failure, please share the entire log file please.

Thanks again for this help. I hadn't tested the latest debian yet.

starapo7348, 2024-09-21:
Hello, thank you for your reply. 

I have started fresh. I have completely removed the directory and checked out the code again.

Compiled with:

`[build.log](https://github.com/user-attachments/files/17084113/build.log)`

I stil get errors during linking. I have created a logfile for further  debugging. (see attached file).

I have no clue about these "undefined reference " errors at the moment.
Maybe you could shed some light on this.

My system:
 OS: Debian GNU/Linux trixie/sid (trixie) x86_6 
 Kernel: 6.10.9-amd64 
 Linux debian 6.10.9-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.10.9-1 (2024-09-08) x86_64 GNU/Linux

```
ii  zlib1g:amd64                            1:1.3.dfsg+really1.3.1-1             amd64        compression library - runtime
ii  zlib1g:i386                             1:1.3.dfsg+really1.3.1-1             i386         compression library - runtime
ii  zlib1g-dev:amd64                        1:1.3.dfsg+really1.3.1-1             amd64        compression library - development

ii  openssl                                 3.3.2-1                              amd64        Secure Sockets Layer toolkit - cryptographic utility
ii  openssl-provider-legacy                 3.3.2-1                              amd64        Secure Sockets Layer toolkit - cryptographic utility

```

Logfile:[build.log](https://github.com/user-attachments/files/17084116/build.log)

Thank you for your help.

Quedale, 2024-09-21:
Okay, I think I know why and I will try to replicate it later.

The big difference that I suspect is the root cause is that I didn't install any system library and let autogen.sh take care of it.
It actually compiled openssl 1.1.0 from source.
You are using the system's 3.3.2, which seems to depend on [ZSTD](https://github.com/facebook/zstd).
I'm not sure why the ZSTD dependency isn't picked up from pkg-conf, but I should be able to replicate this issue.

Thank you for this valuable feedback!

Quedale, 2024-09-22:
Good day!

So I was correct. OpenSSL introduced a new dependency to zstd and it wasn't configured inside the pkg-config configuration file. So my cmake script didn't pick it up dynamically.
To resolve this I only needed to add the missing ZSTD_LDFLAGS and change the linking order of OpenSSL and zlib.

The changes were applied to [OnvifSoapLib](https://github.com/Quedale/OnvifSoapLib).
Simply run `autogen.sh` again and it will automatically pick up the changes.

Let me know how that goes!

starapo7348, 2024-09-22:
Sorry to say... Did not work. Maybe I am doing something wrong.
I have removed the subprojects again and compiled OnvifDeviceManger with
`
./autogen.sh --prefix=$(pwd)/dist --enable-latest`

Stil compile errors. :( 
[odm-build.log](https://github.com/user-attachments/files/17090249/odm-build.log)

I also tried to compile OnvifSoapLib- -> no luck.

`./bootstrap.sh`

[OSL-build.log](https://github.com/user-attachments/files/17090253/OSL-build.log)

Quedale, 2024-09-22:
Okay that was a little odd.

I assumed that FindOpenSSL was taking care of setting up the ZSTD_LDFLAGS variable, but I was wrong.
It was apparently set by a previous build run from a different cmake file. (The variable stayed in the cache even after a clean build)

I nuked the entire project and I encountered the issue again.
I simply restored the line I thought wasn't necessary and it should work now.

Sorry for the trouble!

starapo7348, 2024-09-22:
Now it works. It compiled successfully.
Thank you for your help.

closed 2024-09-22 by starapo7348

Quedale, 2024-09-23:
Out of curiosity, I dug to find which version of OpenSSL introduced this dependency.

I found it here on [this commit](https://github.com/openssl/openssl/commit/caf9317d7d75213990014e07048384be15688889), for 3.2.0-alpha1.

Turns out that this dependency is only required if it is compiled with --enable-zstd (which it is for debian package).
Therefor I find the best implementation is to make the libzstd-dev package optional if it exists on the system.

I could link only when 3.2.0+ is used, but that sounds like unnecessary complexity.
The ideal resolution would be for openssl to add this dependency to the libssl.pc and/or libcrypto.pc configuration files.

Cheers!