Quedale/OnvifDeviceManager #29: compilation issue with OnvifSoapLib / Linking disco-test

stingdau1206, 2024-10-07:
when i run this cmd ./autogen.sh --prefix=$(pwd)/dist --enable-latest
it ok until 
```
[ 98%] Linking C executable /home/khoatn9/Documents/OnvifDeviceManager/subprojects/OnvifSoapLib/release/disco-test
/home/khoatn9/Documents/OnvifDeviceManager/subprojects/OnvifSoapLib/example/client-test.c: In function ‘main’:
/home/khoatn9/Documents/OnvifDeviceManager/subprojects/OnvifSoapLib/example/client-test.c:235:4: error: a label can only be part of a statement and a declaration is not a statement
  235 |    char * name = OnvifScopes__extract_scope(scopes,"name");
      |    ^~~~
/home/khoatn9/Documents/OnvifDeviceManager/subprojects/OnvifSoapLib/example/client-test.c:236:4: error: expected expression before ‘char’
  236 |    char * hardware = OnvifScopes__extract_scope(scopes,"hardware");
      |    ^~~~
In file included from /home/khoatn9/Documents/OnvifDeviceManager/subprojects/OnvifSoapLib/example/client-test.c:5:
/home/khoatn9/Documents/OnvifDeviceManager/subprojects/OnvifSoapLib/example/client-test.c:239:39: error: ‘hardware’ undeclared (first use in this function)
  239 |    C_DEBUG("OnvifScope hardware : %s",hardware);
      |                                       ^~~~~~~~
/home/khoatn9/Documents/OnvifDeviceManager/subprojects/CUtils/build/dist/include/clogger.h:14:75: note: in definition of macro ‘C_DEBUG’
   14 | #define C_DEBUG(fmt, ...) c_log(C_DEBUG_E, __FILENAME__, __LINE__, fmt, ##__VA_ARGS__)
      |                                                                           ^~~~~~~~~~~
/home/khoatn9/Documents/OnvifDeviceManager/subprojects/OnvifSoapLib/example/client-test.c:239:39: note: each undeclared identifier is reported only once for each function it appears in
  239 |    C_DEBUG("OnvifScope hardware : %s",hardware);
      |                                       ^~~~~~~~
/home/khoatn9/Documents/OnvifDeviceManager/subprojects/CUtils/build/dist/include/clogger.h:14:75: note: in definition of macro ‘C_DEBUG’
   14 | #define C_DEBUG(fmt, ...) c_log(C_DEBUG_E, __FILENAME__, __LINE__, fmt, ##__VA_ARGS__)
      |                                                                           ^~~~~~~~~~~
/home/khoatn9/Documents/OnvifDeviceManager/subprojects/OnvifSoapLib/example/client-test.c:300:13: error: a label can only be part of a statement and a declaration is not a statement
  300 |             OnvifDeviceCapabilities * device_caps = OnvifCapabilities__get_device(caps);
      |             ^~~~~~~~~~~~~~~~~~~~~~~
make[2]: *** [example/CMakeFiles/client-test.dir/build.make:76: example/CMakeFiles/client-test.dir/client-test.c.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:886: example/CMakeFiles/client-test.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libcrypto.a(dso_dlfcn.o): in function `dlfcn_globallookup':
(.text+0x17): undefined reference to `dlopen'
/usr/bin/ld: (.text+0x2a): undefined reference to `dlsym'
/usr/bin/ld: (.text+0x35): undefined reference to `dlclose'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libcrypto.a(dso_dlfcn.o): in function `dlfcn_bind_func':
(.text+0x1b7): undefined reference to `dlsym'
/usr/bin/ld: (.text+0x282): undefined reference to `dlerror'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libcrypto.a(dso_dlfcn.o): in function `dlfcn_load':
(.text+0x2f5): undefined reference to `dlopen'
/usr/bin/ld: (.text+0x369): undefined reference to `dlclose'
/usr/bin/ld: (.text+0x3a5): undefined reference to `dlerror'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libcrypto.a(dso_dlfcn.o): in function `dlfcn_pathbyaddr':
(.text+0x466): undefined reference to `dladdr'
/usr/bin/ld: (.text+0x4d7): undefined reference to `dlerror'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libcrypto.a(dso_dlfcn.o): in function `dlfcn_unload':
(.text+0x6b8): undefined reference to `dlclose'
collect2: error: ld returned 1 exit status
make[2]: *** [example/CMakeFiles/disco-test.dir/build.make:101: /home/khoatn9/Documents/OnvifDeviceManager/subprojects/OnvifSoapLib/release/disco-test] Error 1
make[1]: *** [CMakeFiles/Makefile2:912: example/CMakeFiles/disco-test.dir/all] Error 2
make: *** [Makefile:166: all] Error 2
*****************************
*** Make failed OnvifSoapLib ***
*****************************
```

Quedale, 2024-10-08:
Good day and thanks for this feedback.

The first error to focus on is:
```
error: a label can only be part of a statement and a declaration is not a statement
```

This looks like a really simple fix. (Declare variables outside select-case statement)
I'm just not sure why I didn't encounter it myself.

I'll try to commit the fix later.

Quedale, 2024-10-09:
Good news! 

I managed to fix the build, but additional issues had to be dealt with before it was fully successful.
I had to include additional gnu tools from source, since libx11 requires autoconf version 2.70 and ubuntu 20.04 comes with 2.69.

I was able to reproduce the issue and get a functional build in a VM.
I just have to finish cleaning up the changes before I can commit it.

It should be ready later tomorrow.

Quedale, 2024-10-10:
I actually committed the changes last night.
The changes were extensive but mostly related to readability.

Simply pull the latest changes and run autogen.sh again.

Let me know how that goes!