Quedale/OnvifDeviceManager #30: Linux Mint 22 and imou bulet lite 4mp-black screen

onyx82, 2024-10-21:
After authentication, the screen is black, but on the left, next to the camera name and IP, you can see the screenshot taken by the camera.
The one that is written in the terminal after authentication
[image]
```

INFO  24-10-21 19:47:25.552504 138050328158336 onvif_app.c:432: OnvifAuthentication attempt...
TRACE 24-10-21 19:47:25.561453 138049553958592 onvif_app.c:486: EventQueue Dispatching : 1/0/8
DEBUG 24-10-21 19:47:25.561870 138049553958592 onvif_app.c:280: _onvif_authentication_reload
INFO  24-10-21 19:47:25.562013 138049553958592 onvif_device.c:80: [http://192.168.100.94/onvif/device_service] OnvifDevice__authenticate
TRACE 24-10-21 19:47:25.562231 138049553958592 onvif_device.c:42: [http://192.168.100.94/onvif/device_service] OnvifDevice__createMediaService
DEBUG 24-10-21 19:47:25.562270 138049553958592 onvif_device_service.c:54: [http://192.168.100.94/onvif/device_service] Invoking GetCapabilities
TRACE 24-10-21 19:47:25.653301 138049553958592 onvif_base_service.c:196: [http://192.168.100.94/onvif/device_service] No time adjustment required. Camera in sync with client.
DEBUG 24-10-21 19:47:25.677173 138049553958592 onvif_device.c:127: [http://192.168.100.94/onvif/media_service] Successfully created Media soap
TRACE 24-10-21 19:47:25.677226 138049553958592 onvif_media_service.c:430: [http://192.168.100.94/onvif/media_service] OnvifMediaService__get_profile_token 0
DEBUG 24-10-21 19:47:25.677244 138049553958592 onvif_media_service.c:90: [http://192.168.100.94/onvif/media_service] Invoking GetProfiles
TRACE 24-10-21 19:47:25.738244 138049553958592 onvif_base_service.c:196: [http://192.168.100.94/onvif/media_service] No time adjustment required. Camera in sync with client.
TRACE 24-10-21 19:47:25.784528 138049553958592 onvif_media_service.c:444: [http://192.168.100.94/onvif/media_service] OnvifMediaService__get_profile_token : Found profile [MediaProfile00000]
DEBUG 24-10-21 19:47:25.784613 138049553958592 onvif_media_service.c:130: [http://192.168.100.94/onvif/media_service] Invoking GetStreamUri
TRACE 24-10-21 19:47:25.845068 138049553958592 onvif_base_service.c:196: [http://192.168.100.94/onvif/media_service] No time adjustment required. Camera in sync with client.
ERROR 24-10-21 19:47:25.873444 138049553958592 onvif_device.c:150: [http://192.168.100.94/onvif/media_service] StreamURI : rtsp://192.168.100.94:554/cam/realmonitor?channel=1&subtype=0&unicast=true&proto=Onvif
TRACE 24-10-21 19:47:25.873599 138049532987072 onvif_app.c:486: EventQueue Dispatching : 2/0/8
TRACE 24-10-21 19:47:25.873621 138049553958592 onvif_app.c:207: _play_onvif_stream http://192.168.100.94/onvif/device_service
INFO  24-10-21 19:47:25.873656 138049553958592 onvif_device.c:80: [http://192.168.100.94/onvif/device_service] OnvifDevice__authenticate
DEBUG 24-10-21 19:47:25.873687 138049553958592 omgr_device_row.c:418: http://192.168.100.94/onvif/device_service OnvifMgrDeviceRow__get_profile
WARN  24-10-21 19:47:25.873709 138049553958592 onvif_media_profile.c:181: Failed to get profile index. NULL pointer. Defaulting to 0.
TRACE 24-10-21 19:47:25.873734 138049553958592 onvif_media_service.c:430: [http://192.168.100.94/onvif/media_service] OnvifMediaService__get_profile_token 0
TRACE 24-10-21 19:47:25.873767 138049553958592 onvif_media_service.c:444: [http://192.168.100.94/onvif/media_service] OnvifMediaService__get_profile_token : Found profile [MediaProfile00000]
DEBUG 24-10-21 19:47:25.873786 138049553958592 onvif_media_service.c:130: [http://192.168.100.94/onvif/media_service] Invoking GetStreamUri
TRACE 24-10-21 19:47:25.873836 138049532987072 onvif_app.c:145: _display_onvif_device
INFO  24-10-21 19:47:25.873857 138049532987072 onvif_device.c:80: [http://192.168.100.94/onvif/device_service] OnvifDevice__authenticate
DEBUG 24-10-21 19:47:25.873877 138049532987072 omgr_device_row.c:418: http://192.168.100.94/onvif/device_service OnvifMgrDeviceRow__get_profile
TRACE 24-10-21 19:47:25.873914 138049532987072 omgr_device_row.c:432: http://192.168.100.94/onvif/device_service OnvifMgrDeviceRow__load_thumbnail
DEBUG 24-10-21 19:47:25.873934 138049532987072 onvif_media_service.c:385: [http://192.168.100.94/onvif/media_service] OnvifMediaService__getSnapshot
TRACE 24-10-21 19:47:25.873955 138049532987072 onvif_media_service.c:430: [http://192.168.100.94/onvif/media_service] OnvifMediaService__get_profile_token 0
TRACE 24-10-21 19:47:25.873969 138049532987072 onvif_media_service.c:444: [http://192.168.100.94/onvif/media_service] OnvifMediaService__get_profile_token : Found profile [MediaProfile00000]
TRACE 24-10-21 19:47:25.928383 138049553958592 onvif_base_service.c:196: [http://192.168.100.94/onvif/media_service] No time adjustment required. Camera in sync with client.
INFO  24-10-21 19:47:25.953821 138049553958592 gstrtspplayer.c:1057: set location : rtsp://192.168.100.94:554/cam/realmonitor?channel=1&subtype=0&unicast=true&proto=Onvif
DEBUG 24-10-21 19:47:25.953886 138049532987072 onvif_media_service.c:162: [http://192.168.100.94/onvif/media_service] Invoking GetSnapshotUri
DEBUG 24-10-21 19:47:25.953934 138049553958592 gstrtspplayer.c:224: RtspPlayer__play retry[0] - playing[0] rtsp://192.168.100.94:554/cam/realmonitor?channel=1&subtype=0&unicast=true&proto=Onvif
TRACE 24-10-21 19:47:25.954418 138049553958592 onvif_app.c:486: EventQueue Dispatched : 1/0/8
DEBUG 24-10-21 19:47:25.958814 138050328158336 app_dialog.c:157: hide
TRACE 24-10-21 19:47:26.010139 138049532987072 onvif_base_service.c:196: [http://192.168.100.94/onvif/media_service] No time adjustment required. Camera in sync with client.
INFO  24-10-21 19:47:26.051113 138049532987072 onvif_media_service.c:401: [http://192.168.100.94/onvif/media_service] Snapshot URI : http://192.168.100.94/onvifsnapshot/media_service/snapshot?channel=1&subtype=0
TRACE 24-10-21 19:47:26.119468 138049532987072 onvif_media_service.c:177: [http://192.168.100.94/onvifsnapshot/media_service/snapshot?channel=1&subtype=0] Failed begin recv getting snapshot
DEBUG 24-10-21 19:47:26.119512 138049532987072 onvif_media_service.c:249: [http://192.168.100.94/onvifsnapshot/media_service/snapshot?channel=1&subtype=0] Snapshot WWW-Authorization challenge 'Login to 0ccf58c7410332c7caa363406561b351'
TRACE 24-10-21 19:47:26.553088 138049532987072 gtkbinaryimage.c:116: GtkBinaryImage__new - size 442793
TRACE 24-10-21 19:47:26.649731 138049532987072 omgr_device_row.c:464: http://192.168.100.94/onvif/device_service OnvifMgrDeviceRow__load_thumbnail - img created
TRACE 24-10-21 19:47:26.649796 138049532987072 omgr_device_row.c:527: http://192.168.100.94/onvif/device_service OnvifMgrDeviceRow__load_thumbnail done
TRACE 24-10-21 19:47:26.649829 138049532987072 onvif_app.c:486: EventQueue Dispatched : 0/0/8
ERROR 24-10-21 19:47:27.125792 138050328158336 gstrtspplayer.c:741: Error received from element [9] rtspsrc: Nu se poate citi din resursă.
ERROR 24-10-21 19:47:27.125852 138050328158336 gstrtspplayer.c:742: Debugging information: ../subprojects/gst-plugins-good/gst/rtsp/gstrtspsrc.c(6822): gst_rtsp_src_receive_response (): /GstPipeline:onvif-pipeline/GstRTSPSrc:rtspsrc:
ERROR 24-10-21 19:47:27.125852 138050328158336 gstrtspplayer.c:742: Could not receive message. (Timeout while waiting for server response)
WARN  24-10-21 19:47:27.126893 138050328158336 gstrtspplayer.c:776: ****************************************************
WARN  24-10-21 19:47:27.126948 138050328158336 gstrtspplayer.c:777: * Retry attempt #0 - rtsp://192.168.100.94:554/cam/realmonitor?channel=1&subtype=0&unicast=true&proto=Onvif
WARN  24-10-21 19:47:27.126984 138050328158336 gstrtspplayer.c:778: ****************************************************
TRACE 24-10-21 19:47:27.127024 138050328158336 onvif_app.c:446: OnvifApp__player_retry_cb
TRACE 24-10-21 19:47:27.127208 138049744799424 onvif_app.c:486: EventQueue Dispatching : 1/0/8
TRACE 24-10-21 19:47:27.127266 138049744799424 onvif_app.c:103: _player_retry_stream
DEBUG 24-10-21 19:47:29.127554 138049744799424 gstrtspplayer.c:224: RtspPlayer__play retry[1] - playing[1] rtsp://192.168.100.94:554/cam/realmonitor?channel=1&subtype=0&unicast=true&proto=Onvif
TRACE 24-10-21 19:47:29.127961 138049744799424 onvif_app.c:486: EventQueue Dispatched : 0/0/8
ERROR 24-10-21 19:47:30.618069 138050328158336 gstrtspplayer.c:741: Error received from element [9] rtspsrc: Nu se poate citi din resursă.
[... cut ...]
ERROR 24-10-21 19:47:36.884922 138050328158336 gstrtspplayer.c:742: Debugging information: ../subprojects/gst-plugins-good/gst/rtsp/gstrtspsrc.c(6822): gst_rtsp_src_receive_response (): /GstPipeline:onvif-pipeline/GstRTSPSrc:rtspsrc:
ERROR 24-10-21 19:47:36.884922 138050328158336 gstrtspplayer.c:742: Could not receive message. (Timeout while waiting for server response)
TRACE 24-10-21 19:47:36.884951 138050328158336 gstrtspplayer.c:785: Player giving up. Too many retries...
ERROR 24-10-21 19:47:36.885445 138050328158336 onvif_app.c:453: Stream encountered an error

```

onyx82, 2024-10-22:
The Imou camera has the default user "admin", and I created a second user "admin2" with Administrator rights.
Both users have the same password.
What seems strange to me:
When I try to log into Onvif Device Manager with the "admin" user, it doesn't work, but if I use the user I created, "admin2", it logs into Onvif Device Manager successfully.
I tried the command:
gst-launch-1.0 rtspsrc location="rtsp://192.168.100.94:554/cam/realmonitor?channel=1&subtype=0&unicast=true&proto=Onvif" user-id="admin" user-pw="mypass" ! fakesink
If I use the "admin" user, it gives me an error:

Unauthorized (401)
ERROR: The pipeline does not want to preload.
Setting pipeline to NULL...
Releasing pipeline...

If I use the "admin2" user that I created, no error appears; it says:
Redistributing latency...
(and a timer runs)

```
alex:~$ gst-launch-1.0 rtspsrc location="rtsp://192.168.100.94:554/cam/realmonitor?channel=1&subtype=0&unicast=true&proto=Onvif" user-id="admmin2" user-pw="mypass" ! fakesink
Setting the pipeline to PAUSE ...
The pipeline is "live" and does not require PRELOADING ...
Progress: (open) Opening Stream
The pipeline is PRELOADING ...
Pre-rolled, waiting for progress completion...
Progress: (connect) Connecting to rtsp://192.168.100.94:554/cam/realmonitor?channel=1&subtype=0&unicast=true&proto=Onvif
Progress: (open) Retrieving server options
Progress: (open) Retrieving media info
Progress: (request) SETUP stream 0
Progress: (request) SETUP stream 1
Progress:   
 (request) SETUP stream 2
Progress: (open) Opened Stream
Setting   
 the pipeline to PLAYING ...
New clock: GstSystemClock
Progress: (request) Sending PLAY request
Redistributing latency...
Redistributing latency...
Redistributing latency...
Redistributing latency...
Redistributing latency...
Progress: (request) Sending PLAY request
Redistributing latency...
Progress: (request) Sent PLAY request
Redistributing latency...
Redistributing latency...
Redistributing latency...
Redistributing latency...
Redistributing latency...
Redistributing latency...
^Chandling interrupt.
Interrupt: Stopping the pipeline ...
Execution ended after 0:00:07.972021118
Setting the pipeline to NULL ...
Releasing the pipeline...
alex:~$
```

In VLC, I use the address rtsp://admin:password@192.168.100.94:554/cam/realmonitor?channel=1&subtype=0, and it works perfectly, regardless of whether I use "admin" or "admin2".
However, if I use the address:
rtsp://admin:password@192.168.100.94:554/cam/realmonitor?channel=1&subtype=0&unicast=true&proto=Onvif
with the "admin" user, it doesn't work, but with the user I created, "admin2", the live stream runs perfectly.

I am using an Imou Bullet Lite 4MP camera. When accessing its web interface, I've attempted to locate a setting for UNICAST transmission but was only able to find MULTICAST, which is currently activated. Both the default 'admin' user and the custom 'admin2' user have ONVIF enabled.All settings are at their default values.

Quedale, 2024-10-22:
Okay, the first issue at hand is that onvifmgr is unable to connect, but I suspect it might just be a timeout setting a little bit too low.
I could add an option to adjust it in the application settings.

Could you try the following?
```
gst-launch-1.0 rtspsrc location="rtsp://192.168.100.94:554/cam/realmonitor?channel=1&subtype=0&unicast=true&proto=Onvif" user-id="INSERT-YOUR-USERID" user-pw="INSERT-YOUR-PASSWORD" latency=0 teardown-timeout=0 do-retransmission=true is-live=true tcp-timeout=1000000 protocols=4 ! fakesink
```
This should simulate the rtspsrc configuration within [gstrtspplayer.c](https://github.com/Quedale/OnvifDeviceManager/blob/main/src/gst/gstrtspplayer.c).
If it fails, we now have a good hint about a faulty configuration.

If it is successful, you can try to run the application with Gstreamer debug enabled:
```
GST_DEBUG=4 onvifmgr > onvfmgr.log 2>&1
```
This will generate a large amount of logs from gstreamer, giving additional insight about the failure.

</br>
</br>
</br>

In regards to user permissions, that would heavily depends on the device's implementation.
I can only imagine that ONIVF related features are considered for "administrator" while some regular user can connect to other RTSP streams.

In this case, it has different RTSP stream parameters (proto=Onvif) which I assume leads to different permissions validations.
(e.g. Administrators could have 4K with regular users at 720p)

I find it a little odd, since I do own a Imou Ranger 2C and I don't experience those permission issues.

onyx82, 2024-10-24:
I ran the first command, and no error appeared; a timer was running.
I also ran the second command, and it created the file onvfmgr.log, which I uploaded to GitHub.  []( https://github.com/onyx82/onvfmgr.log/blob/main/onvfmgr.log)

Quedale, 2024-10-27:
I really don't see anything other than the camera not responding, but that's just gstreamer's message.
It is possible that some exchange occurs, but the handshake fails to complete.

We can try to increase log level a big to see actual rtsp message exchanged:
```
GST_DEBUG=*:4,rtsp*:7 onvifmgr > onvfmgr.log 2>&1
```
The new section `rtsp*:7` should show RTSP message log (OPTIONS, DESCRIBE, SETUP, PLAY) and a lot more.

<br/>
<br/>
<br/>

You can always try to use your system's gstreamer instead of the static linked latest version.
You can simply rebuild the tool without "enable-latest" argument:
```
make distclean
./autogen.sh
make -j$(nproc)
./onvifmgr
```