Quedale/OnvifDeviceManager #31: issue installing on Almalinux 9

uncovery, 2024-10-28:
I hit a roadblock installing it:

`x11 git clone libX11-1.8.8@https://gitlab.freedesktop.org/xorg/lib/libx11.git
x11 git From https://gitlab.freedesktop.org/xorg/lib/libx11
x11 git  * tag                 libX11-1.8.8 -> FETCH_HEAD
x11 git Already up to date.
x11 git copy repo from cache/home/uncovery/OnvifDeviceManager/subprojects/.cache/libx11-libX11-1.8.8
x11 build src: 'libx11'
x11 build prefix: '/home/uncovery/OnvifDeviceManager/subprojects/libx11/dist'
x11 autogen.sh src: libx11
x11 autogen.sh autoreconf: export WARNINGS=
x11 autogen.sh autoreconf: Entering directory '.'
x11 autogen.sh autoreconf: configure.ac: not using Gettext
x11 autogen.sh autoreconf: running: aclocal -I m4
x11 autogen.sh autoreconf: configure.ac: tracing
x11 autogen.sh autoreconf: running: libtoolize --copy
x11 autogen.sh libtoolize: putting auxiliary files in '.'.
x11 autogen.sh libtoolize: copying file './ltmain.sh'
x11 autogen.sh libtoolize: putting macros in AC_CONFIG_MACRO_DIRS, 'm4'.
x11 autogen.sh libtoolize: copying file 'm4/libtool.m4'
x11 autogen.sh libtoolize: copying file 'm4/ltoptions.m4'
x11 autogen.sh libtoolize: copying file 'm4/ltsugar.m4'
x11 autogen.sh libtoolize: copying file 'm4/ltversion.m4'
x11 autogen.sh libtoolize: copying file 'm4/lt~obsolete.m4'
x11 autogen.sh autoreconf: configure.ac: not using Intltool
x11 autogen.sh autoreconf: configure.ac: not using Gtkdoc
x11 autogen.sh autoreconf: running: aclocal -I m4
x11 autogen.sh autoreconf: running: /home/uncovery/OnvifDeviceManager/subprojects/autoconf-2.72/build/dist/bin/autoconf
x11 autogen.sh configure.ac:95: error: possibly undefined macro: XTRANS_CONNECTION_FLAGS
x11 autogen.sh       If this token and others are legitimate, please use m4_pattern_allow.
x11 autogen.sh       See the Autoconf documentation.
x11 autogen.sh autoreconf: error: /home/uncovery/OnvifDeviceManager/subprojects/autoconf-2.72/build/dist/bin/autoconf failed with exit status: 1
x11 autogen.sh *****************************
x11 autogen.sh *  Autogen failed libx11
x11 autogen.sh *     Kernel: 5.14.0-427.40.1.el9_4.x86_64
x11 autogen.sh *     Kernel: x86_64
`

any adice?

Quedale, 2024-10-28:
Good day!

I never actually tried and Redhat based distro yet, so I'm not too surprised that it failed.
I'll update the `autogen.sh` script to support Almalinux 9 minimal soon.

I was already able to reproduce it on a VM.

Thanks for the feedback!

Quedale, 2024-10-29:
I just committed ([6221ed0](https://github.com/Quedale/OnvifDeviceManager/commit/6221ed0338358188720561160103eabd7a44438c)) the changed to support Almalinux 9 minimal installation.

I did end up covering more ground than what was identified here, but the only thing missing to resolve this error was [xtrans](https://gitlab.freedesktop.org/xorg/lib/libxtrans).
`autogen.sh` will now build it automatically if not found on the system.

Let me know how that goes!

uncovery, 2024-10-29:
Thanks, that was really quick, I'll have a chance to try it out end of the week again, will let you know!

uncovery, 2024-11-01:
ok, just tested it. ./autogen works now, but make returns an error:

 make -j$(nproc)
  CC       src/app/c_ownable_interface.o
  CC       src/app/gtkbinaryimage.o
  CC       src/app/gtkstyledimage.o
  CCLD     images/locked-icon.o
  CCLD     images/microphone.o
  CCLD     images/warning.o
  CCLD     images/save.o
  CCLD     images/tower.o
  CC       src/onvif-mgr.o
  CC       src/alsa/alsa_devices.o
  CC       src/alsa/alsa_utils.o
  CC       src/animations/gtk/custom_gtk_revealer.o
  CC       src/animations/gtk/custom_gtk_progress_tracker.o
  CC       src/animations/gtk/gtk_dotted_slider_widget.o
  CC       src/app/gui_utils.o
  CC       src/app/omgr_device_row.o
  CC       src/app/onvif_app_shutdown.o
  CC       src/app/onvif_app.o
  CC       src/app/details/onvif_details.o
  CC       src/app/details/onvif_info.o
  CC       src/app/details/onvif_network.o
  CC       src/app/onvif_nvt.o
  CC       src/app/task_manager.o
  CC       src/app/dialog/add_device.o
  CC       src/app/dialog/app_dialog.o
  CC       src/app/dialog/credentials_input.o
  CC       src/app/dialog/gtkprofilepanel.o
  CC       src/app/dialog/msg_dialog.o
  CC       src/app/dialog/profiles_dialog.o
  CC       src/app/settings/app_settings_discovery.o
  CC       src/app/settings/app_settings_stream.o
  CC       src/app/settings/app_settings.o
  CC       src/gst/gst_plugin_utils.o
  CC       src/gst/overlay.o
  CC       src/gst/src_retriever.o
  CC       src/gst/backchannel.o
  CC       src/gst/gstrtspplayer.o
  CC       src/queue/event_queue.o
  CC       src/queue/queue_event.o
  CC       src/queue/queue_thread.o
  CCLD     onvifmgr
/bin/ld: /folder/OnvifDeviceManager/subprojects/OnvifSoapLib/build/dist/lib/libonvifsoap.a(stdsoap2.c.o): in function `ssl_auth_init':
/folder/OnvifDeviceManager/subprojects/gsoap-2.8/gsoap/stdsoap2.c:4573: undefined reference to `SSL_CTX_set0_tmp_dh_pkey'
/bin/ld: /folder/OnvifDeviceManager/subprojects/OnvifSoapLib/build/dist/lib/libonvifsoap.a(stdsoap2.c.o): in function `soap_ssl_accept':
/folder/OnvifDeviceManager/subprojects/gsoap-2.8/gsoap/stdsoap2.c:5202: undefined reference to `SSL_get1_peer_certificate'
/bin/ld: /folder/OnvifDeviceManager/subprojects/OnvifSoapLib/build/dist/lib/libonvifsoap.a(stdsoap2.c.o): in function `tcp_connect':
/folder/OnvifDeviceManager/subprojects/gsoap-2.8/gsoap/stdsoap2.c:6441: undefined reference to `SSL_get1_peer_certificate'
/bin/ld: /folder/OnvifDeviceManager/subprojects/OnvifSoapLib/build/dist/lib/libonvifsoap.a(smdevp.c.o): in function `soap_smd_size':
/folder/OnvifDeviceManager/subprojects/gsoap-2.8/gsoap/plugin/smdevp.c:374: undefined reference to `EVP_PKEY_get_size'
/bin/ld: /folder/OnvifDeviceManager/subprojects/OnvifSoapLib/build/dist/lib/libonvifsoap.a(smdevp.c.o): in function `soap_smd_init':
/folder/OnvifDeviceManager/subprojects/gsoap-2.8/gsoap/plugin/smdevp.c:544: undefined reference to `EVP_MAC_fetch'
/bin/ld: /folder/OnvifDeviceManager/subprojects/gsoap-2.8/gsoap/plugin/smdevp.c:547: undefined reference to `EVP_MAC_CTX_new'
/bin/ld: /folder/OnvifDeviceManager/subprojects/gsoap-2.8/gsoap/plugin/smdevp.c:573: undefined reference to `EVP_MD_get0_name'
/bin/ld: /folder/OnvifDeviceManager/subprojects/gsoap-2.8/gsoap/plugin/smdevp.c:573: undefined reference to `OSSL_PARAM_construct_utf8_string'
/bin/ld: /folder/OnvifDeviceManager/subprojects/gsoap-2.8/gsoap/plugin/smdevp.c:574: undefined reference to `OSSL_PARAM_construct_end'
/bin/ld: /folder/OnvifDeviceManager/subprojects/gsoap-2.8/gsoap/plugin/smdevp.c:575: undefined reference to `EVP_MAC_init'
/bin/ld: /folder/OnvifDeviceManager/subprojects/OnvifSoapLib/build/dist/lib/libonvifsoap.a(smdevp.c.o): in function `soap_smd_update':
/folder/OnvifDeviceManager/subprojects/gsoap-2.8/gsoap/plugin/smdevp.c:622: undefined reference to `EVP_MAC_update'
/bin/ld: /folder/OnvifDeviceManager/subprojects/OnvifSoapLib/build/dist/lib/libonvifsoap.a(smdevp.c.o): in function `soap_smd_final':
/folder/OnvifDeviceManager/subprojects/gsoap-2.8/gsoap/plugin/smdevp.c:675: undefined reference to `EVP_MAC_final'
/bin/ld: /folder/OnvifDeviceManager/subprojects/OnvifSoapLib/build/dist/lib/libonvifsoap.a(smdevp.c.o): in function `soap_smd_cleanup':
/folder/OnvifDeviceManager/subprojects/gsoap-2.8/gsoap/plugin/smdevp.c:727: undefined reference to `EVP_MAC_CTX_free'
/bin/ld: /folder/OnvifDeviceManager/subprojects/gsoap-2.8/gsoap/plugin/smdevp.c:748: undefined reference to `EVP_MAC_free'
/bin/ld: /folder/OnvifDeviceManager/subprojects/OnvifSoapLib/build/dist/lib/libonvifsoap.a(mecevp.c.o): in function `soap_mec_start_alg':
/folder/OnvifDeviceManager/subprojects/gsoap-2.8/gsoap/plugin/mecevp.c:633: undefined reference to `EVP_CIPHER_get_iv_length'
/bin/ld: /folder/OnvifDeviceManager/subprojects/gsoap-2.8/gsoap/plugin/mecevp.c:700: undefined reference to `EVP_CIPHER_get_block_size'
/bin/ld: /folder/OnvifDeviceManager/subprojects/OnvifSoapLib/build/dist/lib/libonvifsoap.a(mecevp.c.o): in function `soap_mec_size':
/folder/OnvifDeviceManager/subprojects/gsoap-2.8/gsoap/plugin/mecevp.c:819: undefined reference to `EVP_PKEY_get_size'
/bin/ld: /folder/OnvifDeviceManager/subprojects/OnvifSoapLib/build/dist/lib/libonvifsoap.a(mecevp.c.o): in function `soap_mec_upd_enc':
/folder/OnvifDeviceManager/subprojects/gsoap-2.8/gsoap/plugin/mecevp.c:898: undefined reference to `EVP_CIPHER_get_block_size'
/bin/ld: /folder/OnvifDeviceManager/subprojects/OnvifSoapLib/build/dist/lib/libonvifsoap.a(mecevp.c.o): in function `soap_mec_upd_dec':
[... cut ...] _size'
/bin/ld: /folder/OnvifDeviceManager/subprojects/gsoap-2.8/gsoap/plugin/mecevp.c:1129: undefined reference to `EVP_CIPHER_get_iv_length'
/bin/ld: /folder/OnvifDeviceManager/subprojects/gsoap-2.8/gsoap/plugin/mecevp.c:1149: undefined reference to `EVP_CIPHER_get_iv_length'
/bin/ld: /folder/OnvifDeviceManager/subprojects/gsoap-2.8/gsoap/plugin/mecevp.c:1150: undefined reference to `EVP_CIPHER_get_iv_length'
collect2: error: ld returned 1 exit status
make: *** [Makefile:947: onvifmgr] Error 1

closed 2024-11-01 by Quedale

Quedale, 2024-11-16:
I just finished implementing the Encrypted Credentials Storage.
The application will now retain cameras and their credentials.

Cheers!