Quedale/OnvifDeviceManager #32: event queue

a2082761, 2024-10-30:
I want to receive fire detection events from the camera through ODM and handle these events on my server using an event handler. I see that there is an event queue, but it doesn’t appear in the GUI. Could you let me know how to set this up?

Quedale, 2024-10-30:
Good day!

Unfortunately this isn't the same kind of events.
This EventQueue is actually a thread queue design to manage background tasks.
This is actually part of what I call the application foundation or skeleton.
Once a strong foundation is in place, adding ONVIF capabilities will be a breeze.
I still have a few bugs to resolve and features to add before I consider the foundation stable.

What you are looking for is actually [ONVIF Events](https://www.onvif.org/ver10/events/wsdl/event.wsdl), which isn't implemented yet.
I only implemented bare-bone ONVIF binding at the moment.

That being said, this is not a server, but a GUI application designed to run in a user session.
It's not really meant to be used as a service.
My focus is on a desktop application given the apparent lack on Linux, but there is a decent selection of ONVIF NVR services.

For your use-case, I would recommend looking into a [HomeAssistant](https://www.home-assistant.io/) server with the [ONVIF Integration](https://www.home-assistant.io/integrations/onvif/).
It's a considerable amount of work to setup, but I think it would meet your requirements. (And all free)

If you still would like me to prioritize this over other features in the future, feel free to start a [Discussion](https://github.com/Quedale/OnvifDeviceManager/discussions).
Once I finished everything I wanted, I will start to pick from feature requests.

Cheers!

closed 2024-10-30 by Quedale

Quedale, 2024-12-08:
Maybe check this out too.
[onvif2mqtt](https://github.com/dmitrif/onvif2mqtt)