Quedale/OnvifDeviceManager #33: virtualenv Problem with Debian 11

4920441, 2024-12-03:
Hi,

don't know what's the root cause of the 

virtualenv setup AttributeError: module 'virtualenv.create.via_global_ref.builtin.cpython.mac_os' has no attribute 'CPython2macOsFramework'

... any Ideas? 

`m4 check found
autoconf check found
automake check found
libtool check found
flex check found
bison check found
virtualenv check not found
virtualenv setup Traceback (most recent call last):
virtualenv setup   File "/usr/lib/python3.9/runpy.py", line 197, in _run_module_as_main
virtualenv setup     return _run_code(code, main_globals, None,
virtualenv setup   File "/usr/lib/python3.9/runpy.py", line 87, in _run_code
virtualenv setup     exec(code, run_globals)
virtualenv setup   File "/home/ug/Downloads/onvif-device-manager/OnvifDeviceManager/subprojects/virtualenv.pyz/__main__.py", line 178, in <module>
virtualenv setup   File "/home/ug/Downloads/onvif-device-manager/OnvifDeviceManager/subprojects/virtualenv.pyz/__main__.py", line 174, in run
virtualenv setup   File "/home/ug/Downloads/onvif-device-manager/OnvifDeviceManager/subprojects/virtualenv.pyz/virtualenv/__main__.py", line 20, in run
virtualenv setup   File "/home/ug/Downloads/onvif-device-manager/OnvifDeviceManager/subprojects/virtualenv.pyz/virtualenv/run/__init__.py", line 31, in cli_run
virtualenv setup   File "/home/ug/Downloads/onvif-device-manager/OnvifDeviceManager/subprojects/virtualenv.pyz/virtualenv/run/__init__.py", line 49, in session_via_cli
virtualenv setup   File "/home/ug/Downloads/onvif-device-manager/OnvifDeviceManager/subprojects/virtualenv.pyz/virtualenv/run/__init__.py", line 82, in build_parser
virtualenv setup   File "/home/ug/Downloads/onvif-device-manager/OnvifDeviceManager/subprojects/virtualenv.pyz/virtualenv/run/plugin/creators.py", line 24, in __init__
virtualenv setup   File "/home/ug/Downloads/onvif-device-manager/OnvifDeviceManager/subprojects/virtualenv.pyz/virtualenv/run/plugin/creators.py", line 31, in for_interpreter
virtualenv setup   File "/home/ug/Downloads/onvif-device-manager/OnvifDeviceManager/subprojects/virtualenv.pyz/virtualenv/run/plugin/base.py", line 39, in options
virtualenv setup   File "/home/ug/Downloads/onvif-device-manager/OnvifDeviceManager/subprojects/virtualenv.pyz/virtualenv/run/plugin/base.py", line 18, in entry_points_for
virtualenv setup   File "/home/ug/Downloads/onvif-device-manager/OnvifDeviceManager/subprojects/virtualenv.pyz/virtualenv/run/plugin/base.py", line 18, in <genexpr>
virtualenv setup   File "/usr/lib/python3.9/importlib/metadata.py", line 79, in load
virtualenv setup     return functools.reduce(getattr, attrs, module)
virtualenv setup AttributeError: module 'virtualenv.create.via_global_ref.builtin.cpython.mac_os' has no attribute 'CPython2macOsFramework'
virtualenv setup *****************************
virtualenv setup *  failed to run python3 virtualenv.pyz
virtualenv setup *  Kernel: 5.15.6
virtualenv setup *  Kernel: unknown
virtualenv setup *****************************
`

Quedale, 2024-12-03:
It looks like virtualenv simply fails to setup.

```
AttributeError: module 'virtualenv.create.via_global_ref.builtin.cpython.mac_os' has no attribute 'CPython2macOsFramework'
```

I honestly have no idea why it would setup a MacOs framework. I just use this tool to leverage a python isolated environment to build dependencies.

After a quick search I found [this](https://github.com/pypa/virtualenv/issues/2565). 
It looks like you may have a different python virtual environment installed causing conflict.

I'll start by trying to reproduce this problem and see what can be done.

Quedale, 2024-12-04:
I just performed a complete build on a vanilla Debian 11.11 without issues.
I did manage to reproduce the issue after installing virtualenv from the apt repository.

I should be able to simply check if virtualenv is already installed before downloading it.

The fix should be available soon.

Quedale, 2024-12-04:
You can pull the latest changes and give it a shot.

4920441, 2024-12-04:
cool, thank you very much. 
Just wanted to add the information, that I double checked it on another installation of the same os and the same happens:-)
But yes, virtualenv is also installed there as a debian package.

I'll try it tonight!

Thanks a lot!