Quedale/OnvifDeviceManager #34: Autogen fails with error about unknown option "--show-progress" [solved]

DougLung2000, 2024-12-17:
Compiling on updated Fedora 40, the autogen.sh script failed when downloading gsoap with an error that "--show-progress" was an unknown option. 

I edited the autogen.sh file removing all references to --show-progress and the script ran without any errors and "make" also ran without errors, giving me a working onvifmgr. 

Thanks for the work on this! 

       --Doug

Quedale, 2024-12-17:
Good day!
Thanks for this feedback!

My first guess would be that you have an older version of wget, which doesn't support the "show-progress" parameter.
It was apparently introduced in version [1.16](http://savannah.gnu.org/forum/forum.php?forum_id=8133), which was released quite a while back (Oct 2014)...
I find it odd since Fedora 40 should be using the latest version 1.21. ([see Fedora packge](https://packages.fedoraproject.org/pkgs/wget/wget/index.html))
Can you provide the wget version used? (You possibly have a conflicting version in your PATH)

I should be able to check the version of wget available and ignore this parameter if it's not supported.
I didn't anticipate this since wget was patched over 10 years ago.

I should have some spare time tonight to setup a testing environment.

DougLung2000, 2024-12-17:
Thanks for the quick response!

Here's what I get for the version:

```
bash-5.2$ wget --version
GNU Wget2 2.2.0 - multithreaded metalink/file/website downloader

+digest +https +ssl/gnutls +ipv6 +iri +large-file +nls -ntlm -opie +psl -hsts
+iconv +idn2 +zlib -lzma +brotlidec +zstd -bzip2 -lzip +http2 +gpgme

Copyright (C) 2012-2015 Tim Ruehsen
Copyright (C) 2015-2024 Free Software Foundation, Inc.

License GPLv3+: GNU GPL version 3 or later
<http://www.gnu.org/licenses/gpl.html>.
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
```
I checked my path and found only one "wget". I wonder if this was a change with Version 2.2.0?  Googling for similar cases I'm not seeing any mention of that, but version 2.2.0 was released less than a month ago. 

Doug

Quedale, 2024-12-17:
Okay, I see!

In your case, "wget" is an alias for wget2.
I doubt wget2 would cause conflict out of the box.

Regardless, I will update autogen.sh so that it handles it gracefully.

Thanks again!

Quedale, 2024-12-21:
It is done!

Fixing it turned out to be more complex than expected to avoid any regression.

I should thank you, because turns out that wget2 uses a two-row status indicator, which uses a VT code I wasn't handling.
This made me revisit my logging function in autogen.sh, which resulted in a considerable performance increase.

I will close this issue, considering you managed to build the project initialy anyway, but feel free to reopen it if you are still experiencing this problem.

Cheers and thanks again for this valuable feedback!

closed 2024-12-21 by Quedale