Quedale/OnvifDeviceManager #36: Compilation issue on latest Debian 12.8

rems28, 2025-01-03:
Hello, I'm trying to build your project, and I have compilation issue with GStreamer that complains about Meson. He now want a minimum meson version to 1.1 and Debian 12 has 1.0.1. I'm trying to compile without --enable-latest, but it's continues to compile against gstreamer-1.24.8.
```gstreamer meson src: 'gstreamer'
gstreamer meson prefix: '/home/onvif/onvif/subprojects/gstreamer/build/dist'
gstreamer meson download subprojects gstreamer
gstreamer meson setup gstreamer
gstreamer meson The Meson build system
gstreamer meson gstreamer meson Version: 1.0.1
gstreamer meson gstreamer meson Source dir: /home/onvif/onvif/subprojects/gstreamer
gstreamer meson gstreamer meson Build dir: /home/onvif/onvif/subprojects/gstreamer/build
gstreamer meson gstreamer meson Build type: native build
gstreamer meson gstreamer meson 
gstreamer meson gstreamer meson meson.build:1:0: ERROR: Meson version is 1.0.1 but project requires >= 1.1
gstreamer meson gstreamer meson 
gstreamer meson gstreamer meson A full log can be found at /home/onvif/onvif/subprojects/gstreamer/build/meson-logs/meson-log.txt```
Thanks for your project

Quedale, 2025-01-03:
Good day!

This should be an easy fix. I never realized that gstreamer increased the minimum meson version and I only check for version 0.63.2.
Autogen.sh should fallback to a pip package instead of using the distro package.

Replace line 955 of autogen.sh from:
```{.bash .numberLines .lineAnchors}
if [ ! -z "$(progVersionCheck project="meson" program=meson linenumber=1 lineindex=0 major=0 minor=63 micro=2)" ]; then
``` 

to the following:
```{.bash .numberLines .lineAnchors}
if [ ! -z "$(progVersionCheck project="meson" program=meson linenumber=1 lineindex=0 major=1 minor=1)" ]; then
``` 
The difference is simply changing `major=0 minor=63 micro=2` to `major=1 minor=1`.

I should have some spare time this evening to try to reproduce the issue.

</br>
</br>

Thanks for this feedback!

Quedale, 2025-01-03:
I forgot to clarify, `autogen.sh` will automatically build missing dependencies, including gstreamer.
Using "--enable-latest" only ensures that the latest supported gstreamer package will be used.

You can try to install your distro's gstreamer package and run `autogen.sh` without '--enable-latest'.
This will make onvifmgr use your distro's shared gstreamer ibraries.
`apt install -y libgstreamer1.0-dev libgstreamer-plugins-base1.0-dev libgstreamer-plugins-bad1.0-dev gstreamer1.0-plugins-base gstreamer1.0-plugins-good gstreamer1.0-plugins-bad gstreamer1.0-plugins-ugly gstreamer1.0-libav gstreamer1.0-tools gstreamer1.0-x gstreamer1.0-alsa gstreamer1.0-gl gstreamer1.0-gtk3 gstreamer1.0-qt5 gstreamer1.0-pulseaudio`

Some of these package may not be necessary since I never bothered filtering them out.
This list is also for Ubuntu 22, so it might be slightly different with debian 12.

Quedale, 2025-01-04:
I just committed a fix to properly handle the minimum version of meson.
You can pull the latest changes and give it a go.

Let me know how that goes!