Quedale/OnvifDeviceManager #40: How do I use this?

mices, 2025-03-10:
I have no idea how to use this I'm trying to discover the stream url of my new POE camera My LAN address is 192.168.15.*

[image]

Quedale, 2025-03-10:
Good day!

There's a couple of thing to consider first.
1. Is the camera ONVIF compatible? (PoE cameras aren't usually dirt cheap, and I would expect it does)
2. Is ONVIF enabled? (e.g. Reolink ship their cameras with ONVIF disabled by default)
3. Is it discoverable? (After enabling ONVIF, it usually comes enabled)

What's the brand/model of you camera?

Quedale, 2025-03-10:
I have a hard time finding official documentation about this brand, although I did find some user feedback claiming that ONVIF does actually work.

I did find the following unofficial documentation for the model from the link you shared "I706-POE-FHW".
https://wiki.zoneminder.com/Revotech_I706-P
It refer to a proprietary configuration page available at "http://yourcameraip/" (All cameras have some form of proprietary GUI on top of ONVIF)

Can you access the default Revotech interface dashboard and see if there's any ONVIF related setting? 
I can't even find a screenshot of what that would look like online.

Quedale, 2025-03-10:
If it's anything like the device in [this video](https://www.youtube.com/watch?v=xO6wv5YBeNo), you may be able to manually add the device using "6688" as the ONVIF port.
The discovery may have other problems specific to your network setup.

Something like this: (with appropriate password if you configured one)
[image]

mices, 2025-03-25:
I have to put my computer on the 192.168.1 subnet to access the camera at 192.168.1.105 I tried that but didn't know what value to use for gateway since my gateway is still 192.168.15.1 so I just put 192.168.1.1 but that didn't work and leaving gateway blank causes me to be placed on the 192.168.15 subnet What to do?

Quedale, 2025-03-25:
Okay, this really sounds like a networking issue.

For instance, your camera probably won't work within a network without a DHCP server. (Home routers all include a DHCP service)
Without a DHCP server on the network, a static IP has to be configured on every endpoints (including the camera).
That's why connecting directly to the camera won't work without the use of a router.

Now, there's the issue of routing, subnets and default gateway.
When a DHCP server issues an IP address to a device, it also sends gateway configurations. (Most likely itself)
The gateway isn't something that you use when connecting to a remote endpoint. It's only used by the operating system to determine where to send the traffic for a given interface.
The usual  subnets for home network are 192.168.0.0/24 or 192.168.1.0/24.
You have to make sure that both of your devices are on the same subnet or that the router has a route between those two subnets.

Each subnet will have their own default gateways leading outside, but no route between each other.
When your laptop sends traffic to the camera, the router will search its own routing table for the camera's subnet.
If no route exist in the routing table, the default "0.0.0.0" kicks in, which is usually WAN, not the alternate subnet.

Using seperate subnets is considerably advanced, which I doubt is what you are trying to achieve here...

Lets start from the beginning by setting up a simple flat network.
1. Connect both device to your router (Do not use WAN port for your devices).
2. Power the router and wait for it to finish starting
3. Power your laptop and camera (Make sure that camera is finished starting)
4. Run `arp -a` to find your IP neighbors (the list should include your camera)
5. Find your camera's IP and perform a ping test (`ping -c 4 IPADDRESS`)

If the ping is successful and it is actually your camera's IP, it should work on the application.
if you are unsuccessful for whatever reason, can you share the output of `ifconfig` and `arp -a` ?

Cheers!

mices, 2025-03-26:
The router has an ip address Why are you assuming that the router and camera are both on the same subnet?

mices, 2025-06-15:
From what I read I need a POE switch to be able to use a POE camera

mices, 2025-06-24:
I bought an NVR and a PoE switch I read that  the default ONVIF port for Revotech cameras was 8088 and I know the ip address of the camera is 192.168.1.105 The NVR control panel was new to me so I connected the laptop So far I can't detect the camera at those address/port values when using "ONVIF Add Device" with either 6688 or 8088 with the camera connected to either router or switch

mices, 2025-07-25:
I'm back after adding an NVR and PoE switch to my setup I have the nvr and switch both connected to the router and the camera connected to the switch What do I do now?

Quedale, 2025-07-26:
Are you able to ping the camera's IP now?