Quedale/OnvifDeviceManager #41: Recording or stream export?

erjoalgo, 2025-03-18:
Thanks for this nice project. How difficult would it be to either support recording, or perhaps more generally to export an RTSP or MJPG stream to feed into other applications like ffmpeg or vlc?

Quedale, 2025-03-18:
Good day!

Thanks for trying my project!
Unfortunately. "Record video" is still in the TODO list.
It would be relatively simple to record the stream as-is, but a considerable amount of work would be required to support choosing your own output format.

To answer your general question, ONVIF MediaService has a GetStreamUri soap method to retrieve this information from the camera.
This information isn't displayed anywhere in the GUI yet, but could be added relatively easily. (Exposing ONVIF MediaProfile details) 

Currently, you can find the RTSP URL from the application output logs after playing the stream.
Look for any output from `gstrtspplayer.c`, which should have the RTSP URL as prefix format like below:
```
...
DEBUG 25-03-18 16:06:28.011975 133497865897664 gstrtspplayer.c:474: rtsp://192.168.1.20:554/stream1 Received new pad 'recv_rtp_src_0_1406423677_96' from 'rtspsrc'
TRACE 25-03-18 16:06:28.012028 133497865897664 gstrtspplayer.c:504: rtsp://192.168.1.20:554/stream1 dispatch idle_attach_video_pad
TRACE 25-03-18 16:06:28.012047 133497865897664 gstrtspplayer.c:508: rtsp://192.168.1.20:554/stream1 wait idle_attach_video_pad
TRACE 25-03-18 16:06:28.012183 133498147388096 gstrtspplayer.c:453: rtsp://192.168.1.20:554/stream1 GstRtspPlayerSession__idle_attach_video_pad
DEBUG 25-03-18 16:06:28.110614 133496697784000 gstrtspplayer.c:474: rtsp://192.168.1.20:554/stream1 Received new pad 'recv_rtp_src_1_1650594731_8' from 'rtspsrc'
TRACE 25-03-18 16:06:28.118602 133498039961280 gstrtspplayer.c:945: rtsp://192.168.1.20:554/stream1 State set to READY for audiobin
...
```

I'll look into adding MediaProfile information soon. (Stream URL, Encoding, FPS, etc...)

erjoalgo, 2025-03-18:
Thanks, I was able to find the RTSP stream url from the logs.

Quedale, 2025-03-18:
Great!

I will keep this open as a feature tracking to add this profile information on the GUI and basic live  record.

Cheers!