Quedale/OnvifDeviceManager #46: Eroor while making binaries

smopro, 2025-07-25:
```
onvifmgr build *****************************
onvifmgr build *  OnvifDeviceManager is ready to be built.
onvifmgr build *  	Simply run "make -j$(nproc)"
onvifmgr build *  
onvifmgr build *  Script runtime: 16 minutes and 34 seconds
onvifmgr build *****************************

```

```
src/app/dialog/omgr_encrypted_store.c: В функции «OnvifMgrEncryptedStore__capture_passphrase»:
src/app/dialog/omgr_encrypted_store.c:307:20: ошибка: too many arguments to function «OnvifMgrTrustStoreDialog__new»; expected 0, have 1
  307 |     priv->dialog = OnvifMgrTrustStoreDialog__new(priv->queue);
      |                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~ ~~~~~~~~~~~
In file included from src/app/dialog/omgr_encrypted_store.h:5,
                 from src/app/dialog/omgr_encrypted_store.c:1:
src/app/dialog/omgr_encryoted_store_dialog.h:24:28: замечание: объявлено здесь
   24 | OnvifMgrTrustStoreDialog * OnvifMgrTrustStoreDialog__new();
      |                            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
make: *** [Makefile:1164: src/app/dialog/omgr_encrypted_store.o] Ошибка 1
```

Pierluigi2497, 2025-07-26:
I had the same problem today and i fixed the problem, modifying the line 307 of file src/app/dialog/omgr_encrypted_store.c
From
priv->dialog = OnvifMgrTrustStoreDialog__new(priv->queue);
To
priv->dialog = OnvifMgrTrustStoreDialog__new();

smopro, 2025-07-26:
> I had the same problem today and i fixed the problem

Work, but look like cheat)

Quedale, 2025-07-26:
Thanks @Pierluigi2497 for providing the fix.
I havent looked into it yet, but that's most likely it.

Two of my computers died and I had to build a new one out of spare parts since last time I worked on this, sorry if it has been dormant for a while.

I never realised I left it in a broken state.
I'll get it fixed soon.

Quedale, 2025-08-04:
The problem has been fixed in commit https://github.com/Quedale/OnvifDeviceManager/commit/0f56ffc07a5c5d8f7a1ed55cf1e4c02515d8b4d3

I'm going to close this it since it was pretty straight forward, but feel free to open new issues if you encounter any.

Cheers!